MarlinFirmware/Marlin #12973: Multiple Probing

zafff, 2019-01-21:
Hello,

### Description
I am using Marling bugfix version 1.1.x.
In Configuration.h, enabling `#define MULTIPLE_PROBING 2` does not change the way that the Z-axis is homed. If this setting is disabled, then with a fixed mounted probe (induction), Z-homing does a fast and slow movement. If I enable `#define MULTIPLE_PROBING 2`, then I still get the same homing movements (fast first followed by a slow one).
### Steps to Reproduce
1. Disable  MULTIPLE_PROBING, flash
2. G28
3. Enable MULTIPLE_PROBING and flash
4. G28

**Expected behavior:** [What you expect to happen]
I would expect that with MULTIPLE_PROBING commented, the homing would be only a fast one based on the code. However, this is not the case.

**Actual behavior:** [What actually happens]
With MULTIPLE_PROBING disabled, there is both a fast followed by a slow movement of my steppers to home the Z-axis.

zafff, 2019-01-21:
Sorry I was not explicit. I'll edit to clarify that.

AnHardt, 2019-01-21:
No reason to say sorry.
Just wanted to say:
Altering probing adjustments will not make a difference for homing.

zafff, 2019-01-21:
Yes, I agree that it should not make a difference for homing. I am referring more to the action of the `#define MULTIPLE_PROBING` line. By default (uncommented) it should probe a single time. With it disabled, the 3D printer probes twice still which is counter-intuitive. That is the reason why I wanted to check whether this is a bug or a feature.

AnHardt, 2019-01-21:
As long as you are using G28 Z (Home Z) you are not probing but homing - even when the Z-probe is used for homing.

MULTIPLE_`**PROBING** will not change anything while **homing**.

boelle, 2019-02-21:
@zafff problem solved?

buitre-buitaker, 2019-02-25:
Hi, I have two questions related to MULTIPE_PROBING: I don't why it G28 command is issued, the Z homming movement is made twice, fast and slow, as if it was probing. Also, why a fast and slow movement? why not two slow movements? Thank you

AnHardt, 2019-02-25:
When **homing** the first way is potentially very long. We don't know where we are, that's why we are homing.  If we can be anywhere between min and max the average way is half the axis usable length. Long ways do last long when the speed is low. So we want a high speed to go to the endstop.
When we hit the endstop we immediately stop stepping. There is no well defined deacceleration. If the speed is above 'jerk-speed' the stepper will lose some steps in that. That's about the definition of 'jerk-speed'. 
To get a correct result the second approach should be much slower - at or below 'jerk-speed'. 

When **probing** the bed we already know where it should be. We can go down at high speed to a point assumed to be near the bed, break down to a slow enough speed to make a good probe with the first approach. If we do not trust the process we can repeat that, but in theory there is no second approach needed - contrary to homing. (or at least there was no reason before software denoising mutated to software switch debouching with much higher delays. (1 step vs. N (7) ms). But that's an other discussion.  )

closed 2019-03-03 by thinkyhead

skorokithakis, 2020-04-04:
@AnHardt Is there a solution for people like me, who get inconsistent homing heights and want to try homing multiple times to improve accuracy? Having a BLTouch is worse than useless in my case, because it's so inconsistent that I have to change the offset every time, and multiple attempts would help.