MarlinFirmware/Marlin #12976: cool down after print stop

Thowra, 2019-01-21:
hi,
My colleague uses your printer like this:

Starts printing from the LCD menu, head and bed heating starts.
He immediately Stop print from LCD menu.
He remove SD card.
He copy new files to the card.
He insert the SD card.
He start new print.

The essence is that as long as you copy the files it will heat up the machine.
It is easier and faster than find the preheat option as a menu.
It works with older versions, but not works with Marlin 1.1.8.
The Marlin 1.1.8 cool down head and bed after he select "stop print" option.
How to solve this problem?

FanDjango, 2019-01-21:
<ironic mode on>
I used to use my cars handbrake to initiate a nice drifting rear wheel powerslide. My new car now has an electro-hydraulic handbrake, so no more powerslides. How to solve this problem?
<ironic mode off>

I wonder, if the controller has some pins left over, one could think about a [FR] to be able to assign a user defined list of GCODES to a pushbutton?

I would also perhaps use that for preheating or for extrude 10mm or somethings like that and I have enough places near the display, where I could fit a few buttons.

Thowra, 2019-01-22:
hi,
Good analogy
The push button is a good idea. 
I'll see where to connect.
The machine use Ramps 1.4 and Arduino (some free pins).
Is there a similar function in Marlin (external button)?
How to define external button in the Marlin?

boelle, 2019-01-26:
or does marlin support setting a cool down temp?

M109 R30 to cool down to 30

boelle, 2019-01-26:
if you read this page: https://github.com/MarlinFirmware/Marlin/wiki/M109

then M109 R30 would make things wait until it reaches that temp

Thowra, 2019-01-28:
No external push button installation yet (I was no time to read how to can insert external button option in the Marlin.).
I turned on "custom commands", I resolved the issue temporarily.

boelle, 2019-03-13:
@Thowra still working on this?

Thowra, 2019-03-15:
Yes, but it doesn't work yet.

boelle, 2019-03-15:
what version of marlin do you use?

Thowra, 2019-03-15:
I use V2.0

boelle, 2019-03-15:
ohh, my mistake

i read your first post and now i understand

this should be seen as a feature reqest

maybe close this issue and open a new one with title: [FR] Assign button to function

then in text of new issue write:

I would like to be able to attach a push button and assign functions to it, like preheat and cooldown of bed/nozzle

closed 2019-05-16 by thinkyhead