MarlinFirmware/Marlin #12979: [bugfix-2.0.x] FR to improve ABL report

MasterPIC, 2019-01-21:
I think the user should be able to get:

bilinear_start[X_AXIS]
bilinear_start[Y_AXIS]
bilinear_grid_spacing[X_AXIS]
bilinear_grid_spacing[Y_AXIS]

I may be wrong but, on my Anet A8 with ABL enabled, I'm not able to get them. So I cant' check if the ABL grid loaded from EEPROM is to be replaced by a new one...

gloomyandy, 2019-01-21:
This may be the best solution, but you never know there may be a better one, so could you please describe the problem you are trying to solve here? Why do you need to check if the grid can be replaced? What is it that needs to check (a program running on a PC? A user making use of an attached display?)?

MasterPIC, 2019-01-21:
I prefer to use an ABL grid limited to the area required by the print. Then I store the grid to EEPROM to be able to reuse it.
But if a bigger print is required I need a new probing.
So I need to know the area covered by current ABL grid to figure out if it's good for the next print...

I could obviously store a grid that covers the entire printing area but I prefer to use a small grid spacing and to limit probing points to speed up probing.

MasterPIC, 2019-01-21:
I just added the following code 
```
SERIAL_ECHOLNPGM("xmin:");
SERIAL_ECHOLN(bilinear_start[X_AXIS]);
SERIAL_ECHOLNPGM("ymin:");
SERIAL_ECHOLN(bilinear_start[Y_AXIS]);
SERIAL_ECHOLNPGM("x_spacing:");
SERIAL_ECHOLN(bilinear_grid_spacing[X_AXIS]);
SERIAL_ECHOLNPGM("y_spacing:");
SERIAL_ECHOLN(bilinear_grid_spacing[Y_AXIS]);
```
after

`print_bilinear_leveling_grid();`

in M420.ccp

and it works.
Surely a better solution is advisable.

boelle, 2019-03-13:
maybe submit a PR to bugfix 2.0?

MasterPIC, 2019-03-13:
I'm not sure it's the best solution.

boelle, 2019-03-13:
i read the first post again, and i think that its a bit "overkill"

there is not reason to limit probing to just the area you print, in fact the bigger area the better

boelle, 2019-06-22:
missing label

closed 2019-06-25 by MasterPIC