MarlinFirmware/Marlin #12980: [2.0] missing acceleration on Z axis?

hamster65, 2019-01-21:
I just realized something odd: When I move the Z axis by 100mm, I can clearly hear the sound of the steppers when they accelerate or decelerate. But when I move the Z axis by 10 or 1 mm, there is no acceleration sound. It apperars as if the motor goes to it's target speed instantaneous.
Can anyone reproduce this?

Steps per unit: 1200 (spindle driven lift with belt)
Max Feedrade: 10
Max Acceleration:  60
S-Curve and Junction deviation: off
Step smoothing: on

Firmware: pulled and compiled just an hour ago.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2779954/Configuration.zip)

boelle, 2019-02-21:
@hamster65 problem solved?

thinkyhead, 2019-03-03:
A start and a stop within a short interval is bound to cause more juddering than a start, long move, and gradual stop.

closed 2019-03-03 by thinkyhead