MarlinFirmware/Marlin #12981: [1.1.x] some excess & bugggy code in temperature.cpp

Theoristos, 2019-01-21:
[1.1.9] lines 611-629 in temperature.cpp , Temperature::get_pid_output() routine
----------------------------------------------------------------

   #if HEATER_IDLE_HANDLER
         **if (heater_idle_timeout_exceeded[HOTEND_INDEX]) {
           pid_output = 0;
           pid_reset[HOTEND_INDEX] = true;
         }
         else**
       #endif
       if (pid_error[HOTEND_INDEX] > PID_FUNCTIONAL_RANGE) {
         pid_output = BANG_MAX;
         pid_reset[HOTEND_INDEX] = true;
       }
       else if (pid_error[HOTEND_INDEX] < -(PID_FUNCTIONAL_RANGE) || 
       target_temperature[HOTEND_INDEX] == 0
         #if HEATER_IDLE_HANDLER
           **|| heater_idle_timeout_exceeded[HOTEND_INDEX]**
         #endif
        ) {
        pid_output = 0;
        pid_reset[HOTEND_INDEX] = true;
      }

----------------------------------------------------------------

1. The last one bold line are unused at all bacause of if heater_idle_timeout_exceeded[HOTEND_INDEX] == True , the evaluation is goes through the first bold branch only.

2. The last one bold line are bugged because of if hotend is constantly underheated (pid_error[HOTEND_INDEX] > PID_FUNCTIONAL_RANGE), the evaluation takes the second branch and the last heater timeout check remains unchecked and heater remains constantly maxed.

Better change it to

----------------------------------------------------------------

   #if HEATER_IDLE_HANDLER
         **if (heater_idle_timeout_exceeded[HOTEND_INDEX]) {
           pid_output = 0;
           pid_reset[HOTEND_INDEX] = true;
         }
         else**
       #endif
       if (pid_error[HOTEND_INDEX] > PID_FUNCTIONAL_RANGE) {
         pid_output = BANG_MAX;
         pid_reset[HOTEND_INDEX] = true;
       }
       else if (pid_error[HOTEND_INDEX] < -(PID_FUNCTIONAL_RANGE) || 
       target_temperature[HOTEND_INDEX] == 0
        ) {
        pid_output = 0;
        pid_reset[HOTEND_INDEX] = true;
      }

----------------------------------------------------------------

or maybe to

----------------------------------------------------------------

   if (pid_error[HOTEND_INDEX] < -(PID_FUNCTIONAL_RANGE) || 
       target_temperature[HOTEND_INDEX] == 0
         #if HEATER_IDLE_HANDLER
          **|| heater_idle_timeout_exceeded[HOTEND_INDEX]**
         #endif
        ) {
        pid_output = 0;
        pid_reset[HOTEND_INDEX] = true;
      }
      else if (pid_error[HOTEND_INDEX] > PID_FUNCTIONAL_RANGE) {
         pid_output = BANG_MAX;
         pid_reset[HOTEND_INDEX] = true;
       }

----------------------------------------------------------------

boelle, 2019-02-21:
@Theoristos problem solved?

Theoristos, 2019-02-24:
I have tried to commit fix today. But not sure - am I succeeded or not. I'm still new in Git.

closed 2019-03-03 by thinkyhead