MarlinFirmware/Marlin #12985: Glitches on SD card SPI CLK line

jeffstaley, 2019-01-22:
I'm bringing Marlin up on a RemRam board (STM32F7) and I'm noticing glitches on the SD card's SPI clock line.  In the scope capture you can see three glitches on the clock (purple trace) line while the the chip_select (yellow) line is held low.  You can also see glitches on the MOSI line but those aren't so important. These glitches occur because Marlin is (re)initializing the SPI bus.  The STM32 core releases any existing resources (e.g. the CLK line) before (re)initializing the SPI bus, hence you see the CLK (and MOSI) line start to float high. A not unreasonable behavior.  But glitches on the CLK line is very bad while CS is held low.

My first thought is that Marlin shouldn't be (re)initializing the SPI bus while it's in the middle of using it (CS line held low) since -- I assume -- you can't expect frameworks to always (re)initialize the SPI bus glitch free.

Any thoughts?  Something that I'm missing?

[image]

hasenbanck, 2019-01-22:
Referencing the issue on the effected boards issue tracker:
https://github.com/hasenbanck/remram/issues/12

The glitch will produce consistent errors when an interrupt is firing while the device in initialized. Then it WILL be pulled to HIGH. This behavior of the CLK line and SPI device initialization while asserting the device (SD card) to active low is problematic for all SD card operations.

I don't know why this behavior is only manifesting on this STM32F7 board, but non the less, we shouldn't call ```spiInit()``` while the device is asserted to active low. I will try different fixes in software and will report back, once I found some solutions.

gloomyandy, 2019-01-22:
Hmm the code in the Marlin SD2 interface is a little sloppy. It basically has a function (cardCommand) used to send a command, this will always call a function (chipSelect) before sending data. This chopSelect call will call spiInit and then set chip select low. The problem is that there can be multiple calls to cardCommand and each one will be calling chipSelect and hence spiInit. Which will cause the glitch with CS low.  I guess there are a few solutions...
1. In the spiInit code check to see if anything has changed and only actually do the init if needed (probably not a good idea, as it may be hard to be sure).
2. Refactor the SD2 layer so that chipSelect is no longer called from cardCommand, but is called from the outer functions just once.
3. Modify (or bodge!) the chipSelect (and chipDeselect) to track the state of the select line and only do anything if needed.

hasenbanck, 2019-01-22:
Catching special cases is always a good idea.

jeffstaley, 2019-01-24:
The major problem here is that the SD code is reinitializing the SPI bus numerous times once it has taken the bus (by pulling the CS line low). Good SPI practice is to 

	1. initialize the SPI HW
	2. take the SPI bus by pulling CS line low
	3. do the work (without reinitializing the SPI HW)
	4. release the SPI bus by pulling CS line high

I took a more in-depth look at just *one* of the SD card routines `Sd2Card::init`.  It's has multiple places where the SPI HW is reinitialized once the bus is taken.

	1. spiInit() is called directly after the bus is taken.
	2. spiSend() initializes the SPI HW
	3. cardCommand() initializes the SPI HW
	4. spiRec() initializes the SPI HW
	5. waitNotBusy() initializes the SPI HW

Some of these initialize the SPI HW via `spiInit()` and others via `spi_init()` but I suspect that this an STM32 Core issue.

I don't really know the SPI framework used here but you really want to use one that allows you to send and receive SPI data without initializing the SPI HW.  

Alternatively, you can consider not taking the SPI bus for the entire time. Perhaps you can release the SPI bus between each data packet. This might fit the framework usage pattern better since it wants to initialize the bus before each packet.  But you'd need to know whether this is acceptable for the SD card protocol since it might require the CS line held low for multiple packets.

gloomyandy, 2019-01-24:
Not a good idea to release the CS during SD card operations. 

During initialisation of the SD card you need to change the speed used to communicate and it is spiInit that lets you do that, so there will need to be at least one call made while CS is low. But that is a special case and there are steps to ensure things are synchronised correctly after that call. In all of the other cases it should be possible to call the init before chip select is taken low.

@jeffstaley I'm a bit confused about your comments that spiRec, spiSend and waitNotBusy initialise the hardware. Is that STM32 specific, because I'm pretty sure that is not the case when running on the LPC1768X.

jeffstaley, 2019-01-24:
BTW the LPC176x method makes a lot of sense.  You really need a framework that'll allow you to send and receive bytes with initializing the SPI HW.

gloomyandy, 2019-01-24:
The spiInit call cardCommand is wrong really and the code should be refactored so that spiInit is not called every time. As to releasing CS during the initialisation I'm really not sure to be honest. What I do know is that the SD card interface is not a typical SPI device and there are various special cases. The best reference I've found is this http://elm-chan.org/docs/mmc/mmc_e.html.

Yes the way that the different HALs handle SPI is very confusing, actually MArlin SPI in general is very confusing. I'm not at all convinced that the Due code is correct as as you have seen it messes around with the CS line, which does not seem correct to me.

boelle, 2019-02-21:
@jeffstaley problem solved?

jeffstaley, 2019-02-21:
No, it hasn’t been solved.

boelle, 2019-02-21:
and this is a marlin issue not hardware?

jeffstaley, 2019-02-21:
Definitely a Marlin issue.  Looks to me like there needs to be some 1) significant refactoring of SPI code  with regard to SD `cardCommand` since it's initializing the SPI bus and 2) a serious looking at the various SPI HAL code implementations.  I'm seeing wildly different different implementations of the SPI HAL such that I can't figure who's right.  For example, some SPI HAL's on a SPI Write initialize the SPI bus, others don't. One even toggles the CS line which seems totally weird.

boelle, 2019-03-13:
@thinkyhead any thoughts on this one?

boelle, 2019-06-22:
missing label

closed 2019-07-11 by thinkyhead