MarlinFirmware/Marlin #12989: Dual Extruder Offset problem

avengre, 2019-01-23:
### Description

Marlin 2.0 does not seem to be compensating for dual extruder tool change / Offset.

I have a dual extruder setup, with a X 35.00 offset, When T0 or T1 the firmware faithfully shifts the active tool head over the work space,  I have set the HOTEND_OFFSET_X {0.0, 35.00}  in the firmware..

However, when a T1 is issued, it does not respect the X_MIN position of either the hardware endstop, or the X_MIN software endstop, as it tries to push the carriage to the left to center the T1 toolhead; ramming into the physical constraint of the printer, it does the same in X_MAX if T0 is issue while T1 is at the end of the rail

### Steps to Reproduce

1. G28 command to home
2. While printer is in the park 0,0,0 position issue a T1, this causes the print head to try to move < X=0;, reproduceable with T1 being at X = X_MAX and issuing a T0

**Expected behavior:

If X position can not accomodate the tool change, it wouldn't attempt it, or the park position would have  HOTEND_OFFSET (and bed_min) added to it, and BED_MAX reduced by the same offset to avoid this

**Actual behavior:** 

Crunching noises

#### Additional Information
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2785458/config.zip)

Maxwellfire, 2019-01-24:
This appears to be the same as described in #11845 . I believe that the issue is not updating the software endstops ever on a toolchange, though this may be the symptom of more complex logic gone awry.

~Max

marcio-ao, 2019-02-01:
Yes, we have seen this too. We've carefully had to craft our GCODE so it moves the nozzle away from endstops prior to do tool changes.

boelle, 2019-02-21:
@avengre problem solved?

thinkyhead, 2019-03-03:
> I believe that the issue is Marlin not updating the software endstops ever on a toolchange

Marlin simply doesn't bounds-check the tool-change move. That still needs to be added.

Bidulglloq, 2019-04-04:
Probably connected issue here: 
- Dual Extruder: T0 on the right side, T1 is 33mm on the left: HOTEND_OFFSET_X {0.0, -33.00}
- activated the BLTouch options in marlin as recommended
- T0 active: G28 works well, but everything is then shifted 33mm to the right
(READ: **X:107.00** Y:93.00 Z:5.00 E:0.00 Count X:10700 Y:9300 Z:2000)
- G29 (3x3) has an issue on the first 2 points (shifted 33 right), 3rd probe point is on the right (at physical endstop) and X is correct. Following 2 lines (3 + 3 probe points are correct)
(READ: **X:140.00** Y:93.00 Z:5.00 E:0.00 Count X:10700 Y:9300 Z:2000)
-  From now, everything is correctly located. Following G29 as well.

As a conclusion, G29 V4 gives same correct sampling coordinates on both attempts, but the 1st G29 after G28 is probing the wrong location

More on: [(https://github.com/MarlinFirmware/Marlin/issues/13543)]

boelle, 2019-06-22:
missing label

closed 2019-07-11 by thinkyhead