MarlinFirmware/Marlin #12992: Save Bed Leveling

gmdimitriz, 2019-01-23:
Hi, 
don't know if its a issue and if its the right place to ask, sorry if its not.

I'm trying to save bed leveling data to dont need G29 on every print. so i'm doing this:

G28
G29 - level 
M420 V - check leveling data
M500 - save leveling data
G1 Z0 - Z at right place

restart printer
G28
M420 S1 - enable leveling data
M420 V - check leveling data still same
G1 Z0 - Z not at right place, Z is 1cm above

i'm using AUTO_BED_LEVELING_LINEAR

i'm doing something wrong?

thanks alot

reloxx13, 2019-01-23:
did u set the z offset?

gmdimitriz, 2019-01-23:
yes, i'm using z-offset -1.35 and softlevel limit on z as -2

gloomyandy, 2019-01-23:
That is a pretty old version of Marlin. There have been a lot of changes (and bug fixes) since then. Does the problem still exist in the current release and bug fix?

gmdimitriz, 2019-01-24:
Hi @gloomyandy , thanks you all for the answers.

Yes i updated yesterday and the problem is solved. 

thanks alot

closed 2019-01-24 by gmdimitriz