MarlinFirmware/Marlin #12993: 'lcd_tmpfan_speed' was not declared in this scope after 082f6a27

comps, 2019-01-23:
### Description

Commit 082f6a27 (ADAPTIVE_FAN_SLOWING extension to hotend thermal protection) has caused my configuration to fail on compile (`pio run`) with:
```
Compiling .pioenvs/sanguino_atmega1284p/src/src/module/stepper.cpp.o
Compiling .pioenvs/sanguino_atmega1284p/src/src/module/stepper_indirection.cpp.o
Compiling .pioenvs/sanguino_atmega1284p/src/src/module/temperature.cpp.o
Compiling .pioenvs/sanguino_atmega1284p/src/src/module/tool_change.cpp.o
Compiling .pioenvs/sanguino_atmega1284p/src/src/sd/Sd2Card.cpp.o
Compiling .pioenvs/sanguino_atmega1284p/src/src/sd/SdBaseFile.cpp.o
Compiling .pioenvs/sanguino_atmega1284p/src/src/sd/SdFatUtil.cpp.o
Marlin/src/module/temperature.cpp: In static member function 'static void Temperature::set_fan_speed(uint8_t, uint16_t)':
Marlin/src/module/temperature.cpp:168:7: error: 'lcd_tmpfan_speed' was not declared in this scope
lcd_tmpfan_speed[target] = speed;
^
```

I run a Sanguinololu with an i2c LCD, git diff of my `Configuration.h` / `Configuration_adv.h`: https://gist.github.com/comps/82bcae0de8ffb64175b610d040afac4f
Zip file provided as well, if you prefer that format.

### Steps to Reproduce
1. Apply the configuration diff on 082f6a27
2. `pio run`

### Additional
[082f6a27-lcd_tmpfan_speed-configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2787510/082f6a27-lcd_tmpfan_speed-configuration.zip)

AnHardt, 2019-01-23:
Try replacing
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L167
with
```
#if HAS_LCD_MENU
```

comps, 2019-01-23:
> Try replacing
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L167
> with
> 
> ```
> #if HAS_LCD_MENU
> ```

That seems to fix the issue, thanks!

```diff
diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
index 6e1c93d75..b8d13571c 100644
--- a/Marlin/src/module/temperature.cpp
+++ b/Marlin/src/module/temperature.cpp
@@ -164,7 +164,7 @@ int16_t Temperature::current_temperature_raw[HOTENDS], // = { 0 }
     if (target >= FAN_COUNT) return;
     
     fan_speed[target] = speed;
-    #if ENABLED(ULTRA_LCD)
+    #if HAS_LCD_MENU && ENABLED(ULTRA_LCD)
       lcd_tmpfan_speed[target] = speed;
     #endif
   }
```
Feel free to push it upstream, otherwise I'll submit a pull request tomorrow.

AnHardt, 2019-01-23:
That opens the door for the error from the other side.
The use of the variable
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L167
has to have the same condition as its declaration / definition at 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L140

comps, 2019-01-24:
> That opens the door for the error from the other side.
> The use of the variable
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L167
> has to have the same [...]

How come? ... The diff I posted uses `&&`, which means the assignment will never get through cpp without the declaration. I didn't want to drop the `ULTRA_LCD` as I imagine there can be an LCD menu other than `ULTRA_LCD`.

On the other hand, I see that
```c++
#if ENABLED(ULTIPANEL)
  #define NEWPANEL  // Disable this if you actually have no click-encoder panel
  #define ULTRA_LCD
#endif
...
#define HAS_LCD_MENU        (ENABLED(ULTIPANEL) && DISABLED(NO_LCD_MENUS))
```
so `ENABLED(ULTRA_LCD)` is always true when `HAS_LCD_MENU` is also true.

Therefore would it make more sense to just use
```c++
    #if HAS_LCD_MENU
      lcd_tmpfan_speed[target] = speed;
    #endif

```
? ... I don't know what was the original reasoning behind `ENABLED(ULTRA_LCD)`.

InsanityAutomation, 2019-01-28:
If this is working, can you update and close the issue? Otherwise let us know and we can look into this further.

PR #13006 (2019-01-24): Prevent undefined lcd_tmpfan_speed on NO_LCD_MENUS
This fixes a build error introduced by 082f6a2 when NO_LCD_MENUS
is uncommented:
```
 Marlin/src/module/temperature.cpp: In static member function
 'static void Temperature::set_fan_speed(uint8_t, uint16_t)':
 Marlin/src/module/temperature.cpp:168:7: error: 'lcd_tmpfan_speed'
 was not declared in this scope
```
Given that `ENABLED(ULTRA_LCD)` is always true when `HAS_LCD_MENU`
is also true,
```cpp
 #if ENABLED(ULTIPANEL)
   ...
   #define ULTRA_LCD
 #endif
 ...
 #define HAS_LCD_MENU  (ENABLED(ULTIPANEL) && DISABLED(NO_LCD_MENUS))
```
this shou [... cut]
closed 2019-01-29 by comps