MarlinFirmware/Marlin #12995: [Marlin 2.0@MKS Sbase]Printer reboots on Delta Calibration attempt

cs50AdverD, 2019-01-23:
### Description

Printer reboots when you attempt to calibrate Delta with 3-points calibration

### Steps to Reproduce

1. Using LCD go to Configuration->Delta Calibration
2. Press Auto Home
3. In the same menu choose Calibrate X/Y/Z/Center

**Expected behavior:** [What you expect to happen]
Carriage moves to a chosen point to probe it.

**Actual behavior:** [What actually happens]
Printer reboots.
#### Additional Information
Today's Marlin-bugfix-2.0.x compiled with Platform.io, issue is persistent for at least a month.

Printer:
    Antet A4 with board swapped to MKS Sbase 1.3.
Config:
   [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2788449/Configuration.zip)
Video link:
   https://www.youtube.com/watch?v=HTKyA4j7_mU

boelle, 2019-02-21:
@cs50AdverD problem solved?

thinkyhead, 2019-03-03:
A reboot indicates all the processor time is being used up. Reduce your maximum feedrates and/or your probing / homing feedrates, or reduce your micro-steps to 16x or less.

closed 2019-03-03 by thinkyhead