MarlinFirmware/Marlin #12996: PID_FUNCTIONAL_RANGE , the silent killer of PID. - need clear warning

AndKe, 2019-01-23:
### Description

no amount of PID autotune, and manuaøl massaging of PID's changed my temperature oscillations

### Steps to Reproduce

use PID's that apparently are not perfect for given heater, 
  #define  DEFAULT_Kp 25
  #define  DEFAULT_Ki 1.5 
  #define  DEFAULT_Kd 60 
PID_FUNCTIONAL_RANGE 10

try to autotune or tune PID's - nothing will improve/change.

#### Additional Information

n/a

th33xitus, 2019-01-23:
The heater will heat with full power until the target temperature minus pid_functional_range is reached.
In this case, the difference is 230°C, target temperature is 250°C so the heater will heat with full power until 240°C (`PID_FUNCTIONAL_RANGE 10`) are reached. Only then PID will kick in...

AndKe, 2019-01-23:
Ah , sorry, now I understand.
Should changed PID' s (M301 ...) kick in right away ? - or do they need M500 or anything else to be applied ?  - I am asking because I had long sessions of tweaking with no change in results.

th33xitus, 2019-01-23:
They don't kick in right away after the M303.
You first have to do the M303 for either Hotend or Hotbed (or both).
Then you apply those values with M301 / M304.
And then you do a M500 to save all those values into the EEPROM.

Personally i like to "hardcode" those values when i found the right ones, so i set them into the `configuration.h` and flash them to my controller.

AndKe, 2019-01-23:
Thank you.
I now found my main reason for trouble, I tried to tune in good PID's for 250°C , and I had HEATER_0_MAXTEMP 250 from before printing that hot... that limit messed up any regulation it seems, once I raised it to default 275, everything is good.

closed 2019-01-23 by AndKe

AnHardt, 2019-01-23:
If you are not trying to tune the PIDs manually, you never need M301/M304.
Just use the U parameter of M303 to use the new found parameters. If working save them with M500.

One time you should experiment with PID_FUNCTIONAL_RANGE. I do use it to tune the first overshoot when heating from room temperature to print temperature.