MarlinFirmware/Marlin #12999: [1.1.8] Weird axis movements with TMC2130

dmitry-kutergin, 2019-01-23:
Hi.
Recently I started to see some weird behavior of the steppers with TMC2130 drivers.
I am using 1.1.8 Marlin version and TM2130 library of 2.0.1 version. So the problem is sometimes I see that in the middle of the print steppers out of the sudden start to move really slowly (like 1 mm/s or something), it still moves to the correct position but real slow, after the move finishes it continues with the normal pace.
Furthermore sometimes rarely there is a case when out of a sudden print head moves outside of the printing area, and extruder start erratically moving retracting the filament completely, and this never ends, until you reboot the printer. Some times I notice that when I stop the print and then try to move print head by octoprint commands or try to home, movements seem to be inverted, for example X axis moves to right instead of the left. When this happens, after I do home again, eventually it starts to move into the right direction but, each time it homes into different X position, shifting to the right several cm.
Does anybody experience any of those issues?

teemuatlut, 2019-01-23:
See the first two steps from http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

dmitry-kutergin, 2019-01-24:
Unfortunately I can not upgrade Marlin and a TMC2130 lib, since current versions I have give me the least amount of layer shifts.

boelle, 2019-02-21:
@dmitry-kutergin problem solved?

dmitry-kutergin, 2019-02-23:
Yes, it seems to be solved at least. It just disappeared by itself, not sure what was the problem.

boelle, 2019-02-23:
if you are sure then maybe click close below

closed 2019-02-24 by dmitry-kutergin