MarlinFirmware/Marlin #13000: Error copiling with JUNCTION_DEVIATION enabled

dzieva, 2019-01-24:
Hello,
I am having error in configuration_store.cpp when compiling with JUNCTION_DEVIATION enabled.

### Description

configuration.h

//
// Use Junction Deviation instead of traditional Jerk Limiting
//
#define JUNCTION_DEVIATION
#if ENABLED(JUNCTION_DEVIATION)
  #define JUNCTION_DEVIATION_MM 0.08  // (mm) Distance from real junction edge
#endif

/**
 * Default Jerk (mm/s)
 * Override with M205 X Y Z E
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#if DISABLED(JUNCTION_DEVIATION)
  #define DEFAULT_XJERK 10.0
  #define DEFAULT_YJERK DEFAULT_XJERK
  #define DEFAULT_ZJERK DEFAULT_XJERK // Must be same as XY for delta
#endif

#define DEFAULT_EJERK    5.0  // May be used by Linear Advance

**Actual behavior:** error: DEFAULT_XJERK ...YJERK ...ZJERK was not declared

[image]

dzieva, 2019-01-24:
I'm using the latest version 2.0.x-bugfix

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2793592/Configuration.zip)

dzieva, 2019-01-24:
My Workaround in configuration_store.cpp

[image]

cacrippen, 2019-02-19:
I also had this problem, the fix worked great!! thank you for the help. 

shouldn't this be added to the official code base? I also was using the latest and greatest version and when I tried to enable junction deviation setting build failed with same error as described above.

closed 2019-02-20 by thinkyhead