MarlinFirmware/Marlin #13002: Dual Z motors

Codyisin, 2019-01-24:
Looking for some help Noobie here.... I have a Wanhao I3 that blew its factory Melzi board. I order a MSK 1.4 with tm2208 drivers and have been fighting the upgrade all day. I have of it working however the Z motors are giving me trouble as one motor turns in the same direction no mater what direction its supposed to turn this is the motor plugged into e1 the motor plugged into z is working. Any help would be appreciated

.

gloomyandy, 2019-01-24:
What version of Marlin are you using? Where are your config files? What steps have you taken so far to work out what is going wrong? 

Have you tried swapping the motors around So switch the cables so that E1 drives the other Z motor, what happens?

Have you tried swapping the driver modules around between the two Z and E1? What happens?

Oh and there are almost certainly better ways of getting help then reporting an issue here (issues are supposed to be used to report actual bugs with Marlin, not for general support). You might consider asking for help on the Marlin Facebook page or the Marlin discord channel.

RudyBenner, 2019-01-24:
You could switch the rotation of one motor by swapping 2 pins. Either the 2 on the right, or on the left.

Codyisin, 2019-01-24:
Ended up figuring out it was a bad stepper motor.

gloomyandy, 2019-01-24:
Glad you have found the problem. If things are fixed, please close the issue.

Codyisin, 2019-01-24:
Sorry about using the forum incorrectly i literally googled Marlin help forum and this is what i found didn't realize it was intended for something else.

closed 2019-01-24 by Codyisin