MarlinFirmware/Marlin #13003: Web documentation link issue on the main marlin firmware github project homepage

JPL001au, 2019-01-24:
### Description

The Marlin Firmware github home page "https://github.com/MarlinFirmware/Marlin" has an incorrectly linked hyperlink in the README.md section under heading "Marlin 1.1.9" (direct url "https://github.com/MarlinFirmware/Marlin#marlin-11x").

The hyperlink "**bugfix-1.1.x**"on this web page  in the text starting:
     "However **bugfix-1.1.x** will continue  ...." 
to access the "bugfix-1.1.x" branch of Marlin firmware on github incorrectly links to the "bugfix-2.0.x" branch url instead.

### Steps to Reproduce

1. Use a web browser to access the Marlin Firmware github home page "https://github.com/MarlinFirmware/Marlin"

2. scroll to the "README.md" text section heading "Marlin 1.1.9":
 The direct url for this text section is "https://github.com/MarlinFirmware/Marlin#marlin-11x"

3. click the hyperlink "**bugfix-1.1.x**" in the text  "However **bugfix-1.1.x** will continue  ...."

4. web browser opens the bugfix-2.0.x branch webpage on github instead of the expected bugfix-1.1.x branch.

**Expected behavior:** 

Selecting the affected hyperlink "bugfix-1.1.x" takes the client's web browser to the correct url "https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x"

**Actual behavior:** 

Selecting the affected hyperlink "bugfix-1.1.x" currently takes the client's web browser to the incorrect url "https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x"

#### Additional Information

None

boelle, 2019-02-21:
@JPL001au problem solved?

JPL001au, 2019-02-22:
Hello.

 

No, the problem still exists on the marlin firmware Github home page where the link to Bugfix 1.1.x takes you to the Bugfix 2.0.x branch on github.

closed 2019-03-03 by thinkyhead