MarlinFirmware/Marlin #13007: Bed Heating Fails when attempting PID Autotune

Deneteus, 2019-01-24:
Is this issue still resolved. [BR] Bed PID Autotune kills Heater Immediately #9005 & M303 Bed PID Autotune fails #12468

I am having an issue with bed heating where it fails not too long after attempting to heat when running PID Autotune in Bugfix 1.1.x from 1/22/19. 

I used the stock CR-10 configs for the PID initially then tried to changing the hysteresis settings and its still not working correctly. It was working in a previous build from September 2018. These are 2017 CR-10 Stock boards and I have 2 printers exhibiting the same behavior.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2793607/Marlin.zip)

Deneteus, 2019-02-06:
I abandoned 1.1.9 and compiled for Bugfix 2.0.x (2,5,2019) and had the same issue using PID Autotune at 240C so I re-ran it again and got 2 printers to pass but the temp difference is +_/- 10C.

 
[image]

[image]

AnHardt, 2019-02-06:
PID auto-tuning oscillates the temperature around the target temperature by purpose to find the PID parameters.
I guess that's what your picture shows.This does not tell anything about the success of tuning.
Better show us:
```
M117 Waiting for heat up
M109 R240
M117 Holding temperature
M0 S300 
M117 Wating for cooling
M109 R150
M117 Holding temperature
M0 S300 
M117 Waiting for heat up
M109 R240
M117 Holding temperature
M0 S300 
M104 S0
M117 Ready
```

AnHardt, 2019-02-06:
> I am having an issue with bed heating where it fails not too long after attempting to heat when running PID Autotune

How did it fail? What error message? At about what part of the temperature curve?

Deneteus, 2019-02-06:
> PID auto-tuning oscillates the temperature around the target temperature by purpose to find the PID parameters.
> I guess that's what your picture shows.This does not tell anything about the success of tuning.
> Better show us:
> 
> ```
> M117 Waiting for heat up
> M109 R240
> M117 Holding temperature
> M0 S300 
> M117 Wating for cooling
> M109 R150
> M117 Holding temperature
> M0 S300 
> M117 Waiting for heat up
> M109 R240
> M117 Holding temperature
> M0 S300 
> M104 S0
> M117 Ready
> ```
> Without M303s U parameter the new values are not used after the auto-tuning.
> Without a M500 or setting them in the config (if EEPROM is off), the new values are lost after a reboot.

The error I received when the PID Autotune failed.

Recv: Error:Heating failed, system stopped! Heater_ID: 0
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!"
Changing monitoring state from "Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!" to "Offline (Error: Heating failed, system stopped! Heater_ID: 0 - Printer halted. kill() called!)"
Connection closed, closing down monitor

**I initially used the firmware menu PID Autotune after flashing which is when all 3 printers failed. Since Octoprint disconnected. I don't have the errors in the logs anymore since I re-ran it manually using** 

**M303 E0 S240 C8**

**Then it passed .**

Recv:  T:243.75 /0.00 @:38
Recv:  T:242.50 /0.00 @:38
Recv:  T:241.56 /0.00 @:38
Recv:  T:240.00 /0.00 @:38
Recv:  bias: 164 d: 90 min: 237.00 max: 243.75 Ku: 33.95 Tu: 26.05
Recv:  Classic PID
Recv:  Kp: 20.37 Ki: 1.56 Kd: 66.34
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
Recv: #define DEFAULT_Kp 20.37
Recv: #define DEFAULT_Ki 1.56
Recv: #define DEFAULT_Kd 66.34

**Then I saved it using: M301 P20.37 I1.56 **D66.34 and received the following message.***

Recv:  T:63.15 /0.00 @:0
Recv:  T:62.45 /0.00 @:0
Recv:  T:61.85 /0.00 @:0
Recv:  T:61.18 /0.00 @:0
Send: M500
Recv: echo:Settings Stored (583 bytes; crc 53123)
Recv: ok
Recv:  T:60.56 /0.00 @:0
Recv:  T:59.93 /0.00 @:0
Recv:  T:59.24 /0.00 @:0

Then I went to 

[image]

Deneteus, 2019-02-07:
I had to go into Octoprint and change the behavior setting so that it didn't disconnect from the printer, also enabled the serial.log since I had nothing to look at once it wiped. 

1. M301 P19.60 I1.42 D19.60
2. Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
Recv: #define DEFAULT_Kp 20.04
Recv: #define DEFAULT_Ki 1.42
Recv: #define DEFAULT_Kd 70.60

3.  Send: M301 P20.04 I1.42 D70.60
Recv: echo: p:20.04 i:1.42 d:70.60
Recv: ok
Recv:  T:149.98 /0.00 @:0
Recv:  T:147.84 /0.00 @:0
Recv:  T:145.48 /0.00 @:0
Recv:  T:143.38 /0.00 @:0
Recv:  T:141.32 /0.00 @:0
Send: M500
Recv: echo:Settings Stored (583 bytes; crc 20768)

4. Waited til the hot end temp lowers back to 25.4c 

[image]

5. Preheated the printer to Hotend: 240C Bed: :75C for PETG

6.  What's crazy is that it doesn't show the bed temp.  It's not even heating it.

[image]

[image]

Config Files:  [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2838826/Marlin.zip)
Octoprint Log: [octoprint log.txt](https://github.com/MarlinFirmware/Marlin/files/2838833/octoprint.log.txt)

Deneteus, 2019-02-07:
> > I am having an issue with bed heating where it fails not too long after attempting to heat when running PID Autotune
> 
> How did it fail? What error message? At about what part of the temperature curve?

Deneteus, 2019-02-07:
Nevermind the closed. Issue is unresolved. Browser was having an issue.

Deneteus, 2019-02-07:
I figured out what the problem was with the bed temp. The temp sensor was disabled. Let me try this again.

Deneteus, 2019-02-07:
So after it flashed I attempted the heat the bed up to temp and this is the error.

[octoprint bed heat failed log.txt](https://github.com/MarlinFirmware/Marlin/files/2839015/octoprint.bed.heat.failed.log.txt)

Deneteus, 2019-02-07:
Alright. Here is the failure. Heating Failed

[image]

_Send: M303 E-1 C5 S80
Recv: PID Autotune start
Recv:  T:24.37 /0.00 B:24.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:24.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:24.50 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:24.87 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:25.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:25.50 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:25.94 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:26.06 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:26.50 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:27.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:27.50 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:28.00 /0.00 @:0 B@:127
Recv:  T:24.26 /0.00 B:28.50 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:29.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:29.50 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:30.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:30.34 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:30.69 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:31.03 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:31.38 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:31.72 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:32.07 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:32.41 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:32.76 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:33.10 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:33.53 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:34.09 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:34.48 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:34.83 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:35.17 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:35.52 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:35.86 /0.00 @:0 B@:127
Recv:  T:24.30 /0.00 B:36.21 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:36.62 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:37.24 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:37.59 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:37.93 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:38.28 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:38.62 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:39.03 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:39.59 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:40.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:40.25 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:40.50 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:40.75 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:41.00 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:41.48 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:41.75 /0.00 @:0 B@:127
Recv:  T:24.30 /0.00 B:42.00 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:42.25 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:42.50 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:42.75 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:43.25 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:43.50 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:43.75 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:44.00 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:44.25 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:44.55 /0.00 @:0 B@:127_
Recv:  T:24.34 /0.00 B:44.97 /0.00 @:0 B@:127
Recv:  T:24.30 /0.00 B:45.25 /0.00 @:0 B@:127
Recv:  T:24.26 /0.00 B:45.50 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:45.75 /0.00 @:0 B@:127
Recv:  T:24.34 /0.00 B:46.00 /0.00 @:0 B@:127
Recv:  T:24.37 /0.00 B:46.31 /0.00 @:0 B@:127
[... 169 lines cut ...]
Recv:  T:25.00 /0.00 B:72.54 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:72.54 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:72.68 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:72.68 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:72.82 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:72.82 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:72.96 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:72.96 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.07 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.10 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.11 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.24 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.24 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.38 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.38 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.52 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.52 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.60 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.66 /0.00 @:0 B@:127
Recv:  T:25.00 /0.00 B:73.66 /0.00 @:0 B@:127
Recv: Error:Heating failed, system stopped! Heater_ID: bed
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!"
Changing monitoring state from "Error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!" to "Offline (Error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!)"
Connection closed, closing down monitor

AnHardt, 2019-02-07:
Your bed is simply slow.
Relax
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L104-109

Deneteus, 2019-02-07:
> Your bed is simply slow.
> Relax
> https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L104-109
> Or improve the bed heater. "More Power" ;-)

No, Because in previous builds it wasn't this slow on 3 separate printers. Besides the stock PID being inadequate there is something else going on here. In documentation it says after you run PID Autotune you can save the settings via M500. That's not happening either. I'll have some logs to post in a few.

[image]

[image]

Deneteus, 2019-02-07:
So it failed again and I noticed a few things. 

1. * The LCD display did not show the temp it was being told to heat to but it showed the thermistor temp.
2. * It was disconnected but no errors happened and the bed was cold.
3. * When I restarted Octoprint to reconnect and get the info it showed there were no bed PID settings which means they were erase after it failed.

[octoprint bed heat failed log #3.txt](https://github.com/MarlinFirmware/Marlin/files/2839232/octoprint.bed.heat.failed.log.3.txt)

Deneteus, 2019-02-07:
> Your bed is simply slow.
> Relax
> https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L104-109
> Or improve the bed heater. "More Power" ;-)

Seems like I am not the only one that has run into this problem. #12265

When I was running 1.1.5, 1.1.6, 1.1.7 and 1.1.8 I wasn't having the same issue and 75C took not time to get to. It wasn't till 1.1.9 that these issues started popping up. Only one of my printers had issues getting near 100C and it was a board issue.

Deneteus, 2019-02-07:
So since Creality posted the source a month ago  I checked and the original firmware was using Bang-Bang for the CR-10 S3.  I am able to get to 75 fairly quick though with PidTempBed enabled though. 

https://github.com/Creality3DPrinting/CR10-Melzi-1.1.2/blob/master/Firmware/Marlin/Configuration.h

[image]

[image]

Deneteus, 2019-02-07:
Failed to complete with no errors.

[octoprint bed heat failed log #5.txt](https://github.com/MarlinFirmware/Marlin/files/2840227/octoprint.bed.heat.failed.log.5.txt)

[image]

Deneteus, 2019-02-07:
I kept wondering why it seemed off with the ramping. I checked the original source and the bed thermistor value  was set to 5 instead of 1.

[image]

[image]

AnHardt, 2019-02-07:
From your logs i can see you are now able to store the PID settings.
Better - i can see how autotune fails.
For 80°C the heater is not able to increase WATCH_BED_TEMP_INCREASE °C in WATCH_BED_TEMP_PERIOD seconds - so "Heating failed".
This was not checked for in versions before 1.1.9., so can not occur in earlier versions of Marlin.

For 75°C the case is a bit more complicated. Let me try to explain.
The autotune algorithm tries to center the target temperature between max. overshoot and min. undershoot. Because at 75°C the bed heater is already very slow the overshoot is very small when heating at full power and undershoot is huge at 0 power, the algorithm tries to gradually increase the power during the cooling phase. Because the bed heater can't heat more than 100% the overshoot stays very small (< 1°C) so in the next iteration the power at cooling is increased again - until a too high heater power (for cooling) is piked and the temperature of the bed is not falling below the target temperature anymore. Autotune fails with "PID Autotune failed! timeout" (Log 5 Line 186). This is not alerted by Octoprint because the "Error:" keyword is not in the message (and it shouldn't because this is no threatening problem - just an info).

If using a wrong thermistor setting for the bed, measuring a much lower temperature then you are used to, would explain the behavior. You are at a much higher temperature than supposed. That will look like a slow/weak heater.
If your board has a bed-power-led check, if it is constantly on during full power heat up. Else there may be a problem with the heater PWM. (rare)

The missing target temperature during autotune is normal. The display directly shows the value the 'normal' heater algorithm does use for its work. That is completely bypassed during autotune.

Deneteus, 2019-02-07:
The bed doesn't have a power led but the SSR that Creality uses does. It turns Blue when there is heating. 

[image]

Now based on what you were saying doesn't that mean I should be adjusting the Max Power setting?

[image]

AnHardt, 2019-02-07:
No! Every change here will decrease the maximum power - never increase.
As long as the LED is constantly on during (far) heatup all is well with the PWM.

AnHardt, 2019-02-07:
Table | ADC | Temperature
--- | --- | ---
  5 | 600 |  high 80s  
  5  | 700 | low 80s 
 5 | 800 | ~65   
--- | --- | ---
 1 | 600 | high 90s
1 | 700 | upper 80s
1 | 800 | ~70

That's not much of a difference but at least the expected direction.

Deneteus, 2019-02-07:
Why would an Autotune for the bed report nothing? 

[image]
****

AnHardt, 2019-02-08:
M303 called with C <= 2?

AnHardt, 2019-02-09:
No answer is an answer. See #13113

Deneteus, 2019-02-16:
> M303 called with C <= 2?

No it was called with 5. And it's supposed to default to a specific amount of tests.

boelle, 2019-02-21:
@Deneteus problem solved?

Deneteus, 2019-02-26:
> @Deneteus problem solved?

No problem still exists.

boelle, 2019-03-13:
tried to insulate the underside of the bed so heat loss is keept to minimum?

Deneteus, 2019-05-27:
> tried to insulate the underside of the bed so heat loss is kept to minimum?

I am testing this some more because in the current release I can get the bed to 70c without it failing on two out of 3 printers. No changes to the hardware were necessary.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-11 by thinkyhead

Deneteus, 2019-11-09:
As an update. I added bed insulation to the heat beds and had the exact same issue.