MarlinFirmware/Marlin #13014: Panucatt Re-Arm Not Connecting With USB After Compiling

trnmcd, 2019-01-25:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

kAdonis, 2019-01-26:
Things have changed since the video was recorded
The EEPROM is no longer saved to the SD card, so you won't find it there
The Firmware.bin file wich is uploaded to the SD card gets renamed to Firmware.cur when marlin is  successfully installed on the Re-Arm
You find the correct Windows-USB driver in Marlin/src/HAL/HAL_LPC1768/win_usb_driver
It could take a few minutes til Windows shows the SD Card in Explorer

trnmcd, 2019-01-26:
Thanks kAdonis for the quick reply. Haha, I wish I knew that before I went and botched everything up. I was able to connect to the board and see it in the beginning, I'm not sure what changed. I let the board stay plugged into the computer for 5 or so minutes and it still didn't recognize it. I'm thinking most likely the board is bricked. It's just weird that it still turns on but won't connect.

trnmcd, 2019-01-26:
Is there any way I can just write to the sd card without connecting the board to my computer?

RudyBenner, 2019-01-26:
Yes, best thing to do is try it. No more than 32gig. FAT32 format.

gloomyandy, 2019-01-26:
If you have an SD card reader/adaptor you can just insert the SD card into your computer and copy the firmware over. Then insert the card into your board and restart it. If you do not see the Marlin device and the SD card appear on your computer, reboot again. If that does not work check the sd card to see if the firmware.bin has been renamed firmware.cur. You might also want to try a different SD card, smaller, slower ones are often better.

kAdonis, 2019-01-26:
You can find the firmware.bin file in your marlin folder in .pioenvs/LPC1768  after you compiled
http://marlinfw.org/docs/basics/install_rearm.htm

trnmcd, 2019-01-26:
@Ketchu13 Thanks for the suggestion. I'm just working on the Re-ARM without the RAMPS. Though it does sound like a similar situation.

trnmcd, 2019-01-26:
> If you have an SD card reader/adaptor you can just insert the SD card into your computer and copy the firmware over. Then insert the card into your board and restart it. If you do not see the Marlin [...]

> You can find the firmware.bin file in your marlin folder in .pioenvs/LPC1768 after you compiled
http://marlinfw.org/docs/basics/install_rearm.htm

@gloomyandy @kAdonis I moved the firmware.bin to the sd card and then put the sd card back into the Re-ARM and my computer recognized it! Thanks for the help, I have a little hope now. When it opens the Re-ARM it has an all caps FIRMWARE Cursor file. Does that sound correct? Thanks for all of the help.

boelle, 2019-01-26:
slighty related q, is it possible to expose the onboard sd card so it's seen as a drive on debian?

might even be the default but i dont know this

gloomyandy, 2019-01-26:
@trnmcd Yes that is correct it actually calls it FIRMWARE.CUR which on Windows may show up as a cursor file.

@boelle Yes you need to mount it. There is probably a way to make it automount, but exactly how you do that may vary from one version of linux to another.

gloomyandy, 2019-01-26:
By default the drive is exposed via USB from Marlin. But Octoprint sends an M21 as part of its connect sequence. But I'm not sure this has anything to do with the original question or with Marlin really.

boelle, 2019-01-26:
no its starts to substitute hijacking

right now eyeballing the posts before mine to see if i can see why and what

boelle, 2019-01-26:
@trnmcd the others have answered most questions, but i would connect a thermistor, and thou i have not done it i think you can do that correctly, just to keep marlin from locking you out

another great help/tool would be a sd card reader so you dont need to connect directly

and depending on how your board is mounted the sd card could get dificult to get to, so an extender might also be worth considering

or if you read my post mount the drive that is exposed through a pi and share it on your private network

lots of ways to skin a cat :-D

trnmcd, 2019-01-26:
Thanks to everyone who helped me with my issue, I was able to edit my config and upload it to the Re-ARM directly. I haven't tested it on my machine yet, but the issue that I opened here is resolved. 
@boelle That's an interesting idea. I think I should be good now, going to hook it up later and see if most things work (I figure it'll take awhile to get it fully functioning). I'll look into your post about the pi, I'm currently using octoprint on the printer building my new machine. Thanks again to everyone who helped!!

boelle, 2019-01-26:
i have not made a post, its just an idea i got when i saw your post

trnmcd, 2019-01-26:
Oh my bad, I misread your comment. It is a cool idea. I figure I should probably try to get my machine working before making it more confusing.

closed 2019-01-26 by trnmcd