MarlinFirmware/Marlin #13016: Problem using new extruder that has two heaters

neoxm3, 2019-01-26:
Hi, i have a problem using a new design extruder, that uses two heaters (and possibility of a third) that is impossible in the firmware two assign two diferents pins in rumba.h for example, to the same extruder.

I have: 
#define HEATER_0_PIN        2
#define HEATER_1_PIN        3

and the idea was:
#define HEATER_0_PIN        2 , 3
#define HEATER_1_PIN        -1

but i have a compile error that: "i cant assign more than 1 pin"

for not connect all the cables to one mosfet and split the connection into the mosfets of E0 and E1 and when i say "heat e0", it heats at the same time e0 and e1...

any idea?

thanks in advance

7eggert, 2019-01-28:
My ¢¢: Each extruder should have one pin driving one or more MOSFET. You can connect as many heaters to one MOSFET as its max. current and your cooling allows, and about any sane number of MOSFET per pin. You can only have one temperature sensor anyway so by using a second pin, you gain nothing.

That's talking about normal extruders for one filament. I don't know about mixing extruders.

Ketchu13, 2019-01-29:
maybe by enable this:
```cpp

// @section temperature

// Control heater 0 and heater 1 in parallel.
#define HEATERS_PARALLEL
```

boelle, 2019-02-21:
@neoxm3 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

neoxm3, 2019-03-06:
sorry, i was sick!
yes, it is resolved! very thanks

boelle, 2019-03-06:
@neoxm3 can happen to all of us

hit the close button below

closed 2019-03-06 by neoxm3