MarlinFirmware/Marlin #13019: Q: how to add an extra extruder fan to Re-Arm pins file?

boelle, 2019-01-26:
A bit related to https://github.com/MarlinFirmware/Marlin/issues/12961

but how would i add an extra fan so i could control it with M106 P1?

the pins file do not look quite as in the post i found

Ketchu13, 2019-01-29:
Just about the pins, I use P1_18, P1_19, P1_20, P1_21 and P2_06 with IRF520 for 12v fans with  Re-Arm powered with 24v
```cpp
#define FAN_PIN              P1_21 // (6)  IO pin. Buffer needed - Vcc to IRF520 3.3V - PWM
#define FAN1_PIN             P1_18 // (4)  IO pin. Buffer needed - Vcc to IRF520 5V   - PWM
#define E0_AUTO_FAN_PIN      P1_19 // (5)  IO pin. Buffer needed - Vcc to IRF520 3.3V 
#define E1_AUTO_FAN_PIN      P2_06 // (59) IO pin. Buffer needed - Vcc to IRF520 3.3V
#define CONTROLLER_FAN_PIN   P1_20 // (11) IO pin. Buffer needed - Vcc to IRF520 3.3V - PWM
```
The fans are working properly but .. I do not know if the problem is related to this configuration, but I lose the serial connection (USB between RE-ARM and Host) every time I use the M43 command.
So I'll start from 0 .. and test before opening a new issue topic.

boelle, 2019-01-29:
that is not in the pin file

i think that is from the configuration_adv.h file

boelle, 2019-01-29:
the way i tried to do it (and it did not work)

#ifndef FAN_PIN
  #if ENABLED(IS_RAMPS_EFB) || ENABLED(IS_RAMPS_EFF)   // Hotend, Fan, Bed or Hotend, Fan, Fan
    #define FAN_PIN        RAMPS_D9_PIN
    #define FAN1_PIN       P1_20   // (11)

Ketchu13, 2019-01-29:
I use a custom pins file for the Re-Arm but like I said I have some serial com issues so I'm going to restart with defaults config and pins files.
[pins_RAMPS_S3DP_24.zip](https://github.com/MarlinFirmware/Marlin/files/2807361/pins_RAMPS_S3DP_24.zip)

boelle, 2019-01-29:
well i hope @thinkyhead will shine some of his magic light on my Q  posted at the top

boelle, 2019-02-03:
@thinkyhead what would be the "correct" way of adding a 2nd part cooling fan to re-arm/ramps EFB setup?

EsserPrototyping, 2019-02-04:
I cant´t help with the fans, but i can confirm my Re-Arm is also resetting pretty often when using M43.. It comes back then as Smoothie in the device manager on Win7.

boelle, 2019-02-04:
mine is not resetting at all, but then again i have never had it attached to anything other than octoprint

boelle, 2019-02-05:
so far i have tried this:

#ifndef FAN_PIN
  #if ENABLED(IS_RAMPS_EFB) || ENABLED(IS_RAMPS_EFF)   // Hotend, Fan, Bed or Hotend, Fan, Fan
    #define FAN_PIN        RAMPS_D9_PIN
    #define FAN1_PIN       P1_20   // (11)

did not work :-(

boelle, 2019-02-11:
@thinkyhead i also asked on facebook how to add the 2nd part cooling fan and i did not get a clear answer i could understand

what is the correct way of adding a 2nd part cooling fan to re-arm ramps pin file? 
ie what should be inserted where?

i did try the lines i posted above but it did not work and others have suggested to add 

could it be anywhere outside an if statement like this:

  #define RAMPS_D10_PIN    P2_05   // (10)
#endif

#define HEATER_0_PIN       RAMPS_D10_PIN

#define FAN1_PIN       P1_20   // (11)

ie just after line 165

closed 2019-02-18 by boelle