MarlinFirmware/Marlin #13020: Printrboard G2 SAM3XC

drphil3d, 2019-01-26:
I have painstakingly extracted the pin configuration from g2Core for the Printrboard G2 and created a Marlin 2.0 pin file, I don't know what to do next. I understand its a DUE processor and is from the same family ATSAM3X but has less pins 100 vs 144.

I was able to get Klipper firmware running on, moved the axis around but the heaters would not turn on and I was controlling the DRV8825 current with software pwm sense hardware pwm support has not been implemented for the ATSAM3XC.

I tried compiling marlin 2.0 for DUE and got a ton of errors. I'm not sure what's needed next it's a little above my understanding but I assume there are some other files that need to be modified in the fastio_Due.h file

I would love some guidance on where to go from here, I have a lot of these g2 boards that are difficult to use with the g2Core firmware. Myself and many Printrbot Simple Pro owners would be forever thankful.

I added this line to boards.h
#define BOARD_PRINTRBOARD_G2   1811   // G2 Printrboard SAM3XC Arduino DUE

Pinout and board files from g2core
https://github.com/synthetos/g2/blob/master/g2core/board/printrboardg2/printrboardG2v3-pinout.h

Page 15 has the pinout of this board /micro controller pin name
http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf

Klipper - https://github.com/KevinOConnor/klipper/issues/1121

Pin file and board file for  G2
[pins_PRINTRBOARDg2.zip](https://github.com/MarlinFirmware/Marlin/files/2799552/pins_PRINTRBOARDg2.zip)

vmorrisPolar, 2019-01-30:
I can help test and work through this as I too have a PB Simple Pro which feels a bit stymied. Exactly the post I wanted to find.

Bob-the-Kuhn, 2019-01-31:
Did you men "ATSAM3X**8**C" when you mentioned "ATSAM3XC"?  

I'm pretty sure that none of the items missing in the "8C" chip are used by Marlin.  If that's the case then the easiest way to get Marlin to compile for this board is to lie and say it uses the "8X" chip.

Using that method I got a sucessful compile by:
- Copying the guts of  **pins_PRINTRBOARDg2.h** into **pins_RAMPS_FD_V1.h**
- Creating a minimal **configuration.h** to use the modified RAMPS_FD_V1 file
- Adding the pin **SDSS**.  Marlin insists on having it set to something. I set it to a random number.
- Making the motor PWM array and pin defins match Marlin's requirements.  Right now Marlin assumes that the **DEFAULT_PWM_MOTOR_CURRENT** array has three entries and that the X & Y steppers share a single PWM pin.  That means we'll need to do some custom coding 
- Corrected a Marlin bug in the XYZ defines in the **motion** modules.

Here are the modified files.
[modified files.zip](https://github.com/MarlinFirmware/Marlin/files/2815785/modified.files.zip)

If you can compile & download this into your board then start using M42 and the M43 utilities to verify the pin numbers in **pins_PRINTRBOARDg2.h**.

Bob-the-Kuhn, 2019-01-31:
Happily, I was wrong about needing custom coding.  Marlin currently handles the PWM currents as written in **pins_PRINTRBOARDg2.h.**

drphil3d, 2019-01-31:
yeah its a sam3x8c, my mistake haha big difference there. wonderful I will try it out and also send out that board today

drphil3d, 2019-01-31:
I don't think it's a huge deal if X&Y share the same PWM value at least for most people.

drphil3d, 2019-01-31:
I'm able to compile and upload the firmware just fine. The stepper motors are going high by default when powered on but don't move. some pin assignments are mixed up x min should be y. but it's a start.

drphil3d, 2019-01-31:
SENT: M43
READ: PIN: 00        <unused/unknown>                       Output = 1
READ: PIN: 01        Y_MAX_PIN                              protected 
READ: PIN: 02        <unused/unknown>                       Input  = 1
READ: PIN: 03        Z_MIN_PIN                              protected 
READ: PIN: 04        X_MIN_PIN                              protected 
READ: PIN: 05        <unused/unknown>                       Input  = 1
READ: PIN: 06        <unused/unknown>                       Input  = 1
READ: PIN: 07        <unused/unknown>                       Input  = 1
READ: PIN: 08        <unused/unknown>                       Input  = 1
READ: PIN: 09        <unused/unknown>                       Input  = 1
READ: PIN: 10        <unused/unknown>                       Input  = 1
READ: PIN: 11        <unused/unknown>                       Input  = 0
READ: PIN: 12        <unused/unknown>                       Input  = 0
READ: PIN: 13        <unused/unknown>                       Input  = 1
READ: PIN: 14        <unused/unknown>                       Input  = 0
READ: PIN: 15        HEATER_0_PIN                           protected 
READ: PIN: 16        <unused/unknown>                       Input  = 1
READ: PIN: 17        <unused/unknown>                       Input  = 0
READ: PIN: 18        <unused/unknown>                       Input  = 0
READ: PIN: 19        <unused/unknown>                       Input  = 1
READ: PIN: 20        <unused/unknown>                       Output = 1
READ: PIN: 21        <unused/unknown>                       Output = 1
READ: PIN: 22        <unused/unknown>                       Input  = 1
READ: PIN: 23        <unused/unknown>                       Input  = 0
READ: PIN: 24        <unused/unknown>                       Input  = 0
READ: PIN: 25        <unused/unknown>                       Input  = 0
READ: PIN: 26        <unused/unknown>                       Input  = 0
READ: PIN: 27        <unused/unknown>                       Input  = 0
READ: PIN: 28        <unused/unknown>                       Input  = 0
READ: PIN: 29        <unused/unknown>                       Input  = 0
READ: PIN: 30        <unused/unknown>                       Input  = 0
READ: PIN: 31        <unused/unknown>                       Input  = 1
READ: PIN: 32        <unused/unknown>                       Input  = 0
READ: PIN: 33        <unused/unknown>                       Input  = 1
READ: PIN: 34        <unused/unknown>                       Input  = 1
READ: PIN: 35        <unused/unknown>                       Input  = 1
READ: PIN: 36        <unused/unknown>                       Input  = 1
READ: PIN: 37        <unused/unknown>                       Input  = 1
READ: PIN: 38        <unused/unknown>                       Input  = 1
READ: PIN: 39        <unused/unknown>                       Input  = 1
READ: PIN: 40        <unused/unknown>                       Input  = 1
READ: PIN: 41        <unused/unknown>                       Input  = 1
READ: PIN: 42        <unused/unknown>                       Input  = 0
READ: PIN: 43        <unused/unknown>                       Input  = 0
READ: PIN: 44        <unused/unknown>                       Input  = 1
READ: PIN: 45        <unused/unknown>                       Input  = 1
READ: PIN: 46        <unused/unknown>                       Input  = 1
READ: PIN: 47        <unused/unknown>                       Input  = 1
READ: PIN: 48        <unused/unknown>                       Input  = 1
READ: PIN: 49        Z_MS3_PIN                              protected 
READ: PIN: 50        Z_MS1_PIN                              protected 
READ: .              Z_MS2_PIN                              protected 
READ: PIN: 51        <unused/unknown>                       Input  = 1
READ: PIN: 52        <unused/unknown>                       Input  = 0
READ: PIN: 53        <unused/unknown>                       Input  = 1
READ: PIN: 54 (A 0)  Z_STEP_PIN                             protected 
READ: PIN: 55 (A 1)  Z_ENABLE_PIN                           protected 
READ: PIN: 56 (A 2)  Z_DIR_PIN                              protected 
READ: PIN: 57 (A 3)  <unused/unknown>   Analog in =   944   Input  = 1
READ: PIN: 58 (A 4)  <unused/unknown>   Analog in =   606   Input  = 1
READ: PIN: 59 (A 5)  <unused/unknown>   Analog in =  1023   Input  = 1
READ: PIN: 60 (A 6)  FAN1_PIN                               protected 
READ: PIN: 61 (A 7)  <unused/unknown>   Analog in =   466   Input  = 0
READ: PIN: 62 (A 8)  <unused/unknown>   Analog in =   158   Input  = 0
READ: PIN: 63 (A 9)  MOTOR_CURRENT_PWM_E_PIN                Output = 0
READ: PIN: 64 (A10)  <unused/unknown>   Analog in =   526   Input  = 1
READ: PIN: 65 (A11)  <unused/unknown>   Analog in =     0   Input  = 0
READ: PIN: 66        <unused/unknown>                       Input  = 0
READ: PIN: 67        MOTOR_CURRENT_PWM_Z_PIN                Output = 0
READ: PIN: 68        MOTOR_CURRENT_PWM_XY_PIN               Output = 0
READ: PIN: 69        X_MS3_PIN                              protected 
READ: PIN: 70        X_MS1_PIN                              protected 
READ: .              X_MS2_PIN                              protected 
READ: PIN: 71        <unused/unknown>                       Output = 0
READ: PIN: 72        <unused/unknown>                       Input  = 1
READ: PIN: 73        <unused/unknown>                       Input  = 0
READ: PIN: 74        MISO_PIN                               protected 
[... cut ...]
READ: PIN: 76        SCK_PIN                                protected 
READ: .              X_DIR_PIN                              protected 
READ: PIN: 77        FAN_PIN                                protected 
READ: .              SDSS                                   protected 
READ: .              SS_PIN                                 protected 
READ: .              Y_MS3_PIN                              protected 
READ: PIN: 78        Y_MS1_PIN                              protected 
READ: .              Y_MS2_PIN                              protected

Bob-the-Kuhn, 2019-01-31:
FANTASTIC!!!

> The stepper motors are going high by default when powered on but don't move.

Sounds like:
- The steppers are powered up after a reset (can't turn them by hand)
- No movement when a commanded.

Are the steppers power up when a movement command is issued?  If yes then try doubling the current.  If not try flipping the X_ENABLE_ON flags.

drphil3d, 2019-02-01:
A friend of mine who knows a lot more about micro controllers then I told me that the registers many need to be set so the pin muxing is handled properly. I ran into an issue with Klipper where I was unable to get any of the fans or heaters to turn on, the developer suggested there may be a global pin that needs to be set to enable those pins. 

These are the three files in order that reference each other that seem to have very important information in them.
Here is the pinout for this board
https://github.com/synthetos/g2/blob/master/g2core/board/printrboardg2/printrboardG2v3-pinout.h
https://github.com/synthetos/g2/blob/master/g2core/board/printrboardg2/motate_pin_assignments.h
https://github.com/synthetos/Motate/blob/master/MotateProject/motate/Atmel_sam3x/motate_chip_pin_functions.h

Bob-the-Kuhn, 2019-02-01:
The first file was gold.  It's almost as good as a schematic.

The last file indicates that some of the pins are inverted.  I'll have to see which ones those are.

I don't see anything that implies a mux function.

I have a nagging suspicion that there is a filter and/or voltage divider between the DRV8825's Vref pin and the signal from the CPU.  Any idea what the motor PWM frequency and duty cycle range are?

I'll look at it and generate configuration files that match tomorrow morning.

drphil3d, 2019-02-01:
I pulled this info out of https://github.com/synthetos/g2/blob/master/g2core/board/printrboardg2/hardware.h
[updated pin.zip](https://github.com/MarlinFirmware/Marlin/files/2820059/updated.pin.zip)

/**** Stepper DDA and dwell timer settings ****/

//#define FREQUENCY_DDA		200000UL		// Hz step frequency. Interrupts actually fire at 2x (400 KHz)
#define FREQUENCY_DDA		150000UL		// Hz step frequency. Interrupts actually fire at 2x (300 KHz)
#define FREQUENCY_DWELL		1000UL
#define FREQUENCY_SGI		200000UL		// 200,000 Hz means software interrupts will fire 5 uSec after being called

drphil3d, 2019-02-03:
Motor 1 = z
Motor 2 = x
Motor 3 = y
Motor 4 = e

The ribbon cable has motors y and e
We can also look at Klipper and see what was done to get things moving.
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

drphil3d, 2019-02-03:
Here is more info about how the whole system is designed to work. I also have the gerber files for the g2. http://www.appfruits.com/2016/11/printrbot-simple-2016-hardware-pcb-explained/

Bob-the-Kuhn, 2019-02-04:
Sounds like this board was fully functional using Klipper.  Please post your Klipper configuration file. That should contain a lot of the pin info I'm looking for.

drphil3d, 2019-02-04:
I was able to read the endstop status and move the motors just fine but was unable to control heaters. https://github.com/KevinOConnor/klipper/issues/1121#issuecomment-457800253

Bob-the-Kuhn, 2019-02-05:
I've been able to beep out enough of the circuitry to show that PA7 provides an AC coupled enable signal to the FETs.  

Now I need to find the frequency & duty cycle areas that make the system happy.

Bob-the-Kuhn, 2019-02-05:
2uS low pulse every 50mS makes it happy.

I can control the FAN FETs with no problems.

It'll take a new board definition in PlatformIO to control the extruder and bed FETs.  The port/pin combos used by the hardware are not assigned logical pin numbers in the Due system.

drphil3d, 2019-02-05:
Are you serious! That's awesome, I struggled with it for so long.

ps: Just finished that part I'll send it to you tomorrow.

drphil3d, 2019-02-05:
When I power on the Y and Z axis are energized and buzzing, G28 makes the tone change. I jogged the Z axis and I could hear it trying to move.

drphil3d, 2019-02-06:
interesting, If I'm reading this correctly It would seem to me that they are using a linkage file to modify the pin mapping for sam3xc for this specific board sense they originally wrote the code for the sam3xe. https://github.com/synthetos/g2/blob/1ff7f600923c622fbf2b546e7d15a24498d1d1c1/g2core/board/printrboardg2/motate_pin_assignments.h#L249

drphil3d, 2019-02-06:
Here we go this explains things.
https://github.com/synthetos/g2/issues/371#issuecomment-408630672

Bob-the-Kuhn, 2019-02-07:
I've gotten brave enough (desperate enough) to solder some wires to the board.  I can now look at all the Motor 1 signals using a logic analyzer.  Much easier to see what is going on.

I think the buzzing is coming from the 500Hz square wave my test code is feeding to the Vref pins.  What the chips are seeing is a 75% duty cycle wave.  Apparently the DRV8825 is fast enough to respond to the off time of the Vref signal.  I think I'll play with that next.

I think I have the configuration_adv.h file setup to implement the microstep selection feature.  I need to do more testing before declaring victory.

There are 8 pins used by the G2 that are not in the Arduino/DUE pin map.  I've modified fastio to use the G2 defs when that motherboard is selected.  The results in fastio WRITE and READ functions accessing different port/pin combos than the Arduino/DUE digitalRead and digitalWrite functions.  Functionally that's not a problem but it makes trouble shooting & verification a real bear because the M42 & M43 commands can't access/view/write the 8 pins.

drphil3d, 2019-02-08:
I may not have mapped the microstep pins out correctly which would
attribure to the buzzing sound instead of movement.

>

Bob-the-Kuhn, 2019-02-08:
The buzzing is due to the 1KHz square wave I was feeding it.  I've got an UGLY 10KHz current PWM working & the buzzing is gone.

[Marlin_G2 rev H.zip](https://github.com/MarlinFirmware/Marlin/files/2845851/Marlin_G2.rev.H.zip)

Here's a snapshot of the code:
- Motor 1 is the only one that works 100%
- Motor 2 moves but doesn't disable when it should.
- Motors 3 & 4 do nothing.
- The stepper currents can be changed via configuration_adv.h settings.
- Microsteps may be working.  I don't remember how far I got in testing them.
- Haven't looked at the FETs in a few days.  They should still be working.

The schematic agrees with the pins file you provided.  

I'm going to have to solder wires to the Motor 2, 3 & 4 drivers before I can make more progress.

I need to put this on the back burner for a bit.  The wife is having surgery Wednesday which means well intentioned people are coming which means I need to clean up all my messes.  I'll probably be spending half days on that for a bit.

drphil3d, 2019-02-08:
I will do some testing and see what I can do. I have a feeling the micro stepper pins may be reversed on motor 3, 4.
Where you able to get the hotend fet to turn on?
I greatly appreciate the work you have done. If this ends up working out I can give you a few more boards for other projects.

Bob-the-Kuhn, 2019-02-08:
Yes, I was able to get all four FETs to turn on.

I just tried motor 2 and it seems happy.  The disable worked as it should.  Maybe I typed the command in wrong earlier.

drphil3d, 2019-02-08:
Oh excellent!! That’s exciting
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

Bob-the-Kuhn, 2019-02-09:
I've got it working.  Ugly code by all the FETs and motors work.

Here's a snapshot of the code.  [Marlin_G2_2019-02-09.zip](https://github.com/MarlinFirmware/Marlin/files/2848340/Marlin_G2_2019-02-09.zip)

I'm going to make the code presentable and then start on the display.  Please post a snapshot of what it looks like with the G2 code.

FYI - here's what the board looks like now.
[image]

drphil3d, 2019-02-09:
WOW! I don't even have words thats pretty impressive. I'm going to try it out right now.

drphil3d, 2019-02-10:
It works! The hotend seems to be partially powered because I smelled burning plastic and the thermistor was reading 70c only a 30 seconds after connecting to it via usb.
the X and Z axis are swapped.

Bob-the-Kuhn, 2019-02-10:
Here's the corrected pins file.

[pins_PRINTRBOARD_G2.h.txt](https://github.com/MarlinFirmware/Marlin/files/2848438/pins_PRINTRBOARD_G2.h.txt)

drphil3d, 2019-02-10:
I somehow locked my processor
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba, stlink
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Auto-detected: COM10
Forcing reset using 1200bps open/close on port COM10
Waiting for the new upload port...
Uploading .pioenvs\DUE_USB\firmware.bin
Atmel SMART device 0x284e0a60 found

Flash page is locked
Erase flash
*** [upload] Error 1
 [ERROR] Took 19.79 seconds 
[image]

drphil3d, 2019-02-10:
I swapped out the boad, programmed it, then went to program it again and get the same result. yikes

drphil3d, 2019-02-10:
I’m not going to mess with it anymore until I have a better understanding
of what’s going on. I’d rather not ruin a bunch of boards. I may need to
get my hands on an atmel ice to fix these. I may have messed with something
in settings that cause it. Still shows up as a programming interface but
it’s locked,  I tried jumping the erase and reset no luck.

I had firmware on the board and was messing around with it, but after I
went to program it again I seem to have put it in some state where it won’t
exit programming mode but won’t accept new firmware.
I did enable EEPROM option in marlin maybe it’s not happy about that?

I had some really strange behavior, moving the Y axis away from the end
stop switch, as soon as the switch was in the open position the neo pixels
turn green. It’s repeatable beheavior.

 I must not have the microstepping working correctly, the z axis moves way
to far way too fast regardless the steps per mm being set correctly. (400
at 16x microstepping.)
Also the x and y axis end stops switch assignments are swapped. I’m able to
home the Z axis.
I can control the Fan just fine.
We are so close! I have more board files for you as well.
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

drphil3d, 2019-02-10:
Ok nevermind I solved that issue using the SAM-BA utility https://www.microchip.com/developmenttools/ProductDetails/atmel%20sam-ba%20in-system%20programmer#additional-summary

drphil3d, 2019-02-10:
I tried to compile using the new pin file and it fails:
Marlin\src\module\stepper.cpp: In static member function 'static void Stepper::digipot_current(uint8_t, int16_t)':
Marlin\src\module\stepper.cpp:2492:98: error: 'MOTOR_CURRENT_PWM_RANGE' was not declared in this scope
#define _WRITE_CURRENT_PWM(P) analogWrite(MOTOR_CURRENT_PWM_## P ##_PIN, 255L * current / (MOTOR_CURRENT_PWM_RANGE))
^
Marlin\src\module\stepper.cpp:2496:13: note: in expansion of macro '_WRITE_CURRENT_PWM'
_WRITE_CURRENT_PWM(X);

drphil3d, 2019-02-10:
Here are the schematics and source for the LCD
[Printrhub-master.zip](https://github.com/MarlinFirmware/Marlin/files/2848556/Printrhub-master.zip)
[Power Board V1.2 Docs.zip](https://github.com/MarlinFirmware/Marlin/files/2848557/Power.Board.V1.2.Docs.zip)
[Printrboard G2 V3c.zip](https://github.com/MarlinFirmware/Marlin/files/2848558/Printrboard.G2.V3c.zip)

drphil3d, 2019-02-10:
Good News, I can home Y and Z but for some reason can't move my X axis.
I have my hotend unplugged as it seems to be enabled by default and thats a bit dangerous lol.

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/2848564/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/2848565/Configuration.h.txt)

[pins_PRINTRBOARD_G2.h.txt](https://github.com/MarlinFirmware/Marlin/files/2848563/pins_PRINTRBOARD_G2.h.txt)

drphil3d, 2019-02-10:
I can kinda move x axis with this pin config but something isn't right. makes a lot of nose but doesn't move much. Also the neopixel turns green when the Y axis limit switch is triggered while homing. 

#define   X_STEP_PIN               75 //
#define   X_DIR_PIN                 74 //
#define   X_ENABLE_PIN          73 //

Bob-the-Kuhn, 2019-02-10:
Seems like I've spend most of the evening shooting myself in the foot.

Here's another snapshot.  I think the one I posted earlier may not have been setting the custom pins as outputs.

[Marlin_G2_2019-02-10.zip](https://github.com/MarlinFirmware/Marlin/files/2848704/Marlin_G2_2019-02-10.zip)

I've been doing most of my downloads via the USB port.  Sometimes I use the JTAG port.

I have not seen the "Flash page is locked" error before.  I'm surprised the erase pin didn't fix the problem.  Worst case you send me the tanked boards & I'll download them via the JTAG port.

Bob-the-Kuhn, 2019-02-10:
I've opened PR #13116 to get feedback.

Bob-the-Kuhn, 2019-02-10:
I finally understand how the power module, LCD & G2 board hook up.  

Looks like my little 12V supply doesn't have enough muscle to power the 5V supply on the power module.  Looks like I'll need to look for something beefier.

The LCD looks rather intimidating at first glance.  Can you point me to the 3X8C code that interfaces with it?  Hopefully it'll supply most of the needed code.  

I'm wondering who controls who in the system.  If all the menus reside on the LCD and it just queiries the SAM3X8C for data then we probably won't be able to interface it to Marlin in a meaningful way.

abdrumm, 2019-02-10:
https://github.com/Printrbot/Printrhub/blob/master/README.md

The informative link here is Phillip’s blog... he was the mastermind behind the firmware on the printrhub... http://www.appfruits.com/2016/11/printrbot-simple-2016-commstack-explained/

Bob-the-Kuhn, 2019-02-10:
I don't think I'm interested in the ESP/MK20 COMMSTACK communications.

What I want to know is the how and what of the MK20/G2 communications.  Example: When the user brings up a screen that has temperature on it, how does the MK20 ask for the data and how should the G2 respond?

drphil3d, 2019-02-10:
It's communicating over UART, there are gcode commands in the esp firmware for things like loading and unloading filament, setting temperature, etc. 
Esp3d is very similar in function 
https://github.com/luc-github/ESP3D

abdrumm, 2019-02-11:
The g2 emulates marlin, so assume the communication from the hub is talking marlin... while early on, we spoke the g2core language (Json)... but I thought we dumped that. Since my programmer left, I can’t soeak w authority, only point to the firmware for the hub on github: https://github.com/Printrbot/Printrhub/blob/master/README.md

Bob-the-Kuhn, 2019-02-11:
What does it take to power up the display?

I've plugged the 14 pin cable into the hub and the power module.
The 10 pin cable is plugged into the hub and the G2 board.
There's 12V on the power module past the fuse.

abdrumm, 2019-02-11:
The display should power on w the powerboard power switch... you might try reversing the cable going to the hub... if reversed, it remains black. It should signal the power board to give power - via the relays- to the g2

Brook

abdrumm, 2019-02-11:
The display should power on w the powerboard power switch... you might try reversing the cable going to the hub... if reversed, it remains black. It should signal the power board to give power - via the relays- to the g2

Bob-the-Kuhn, 2019-02-11:
I take it that the 1/4" spade terminals is where a switch goes?

abdrumm, 2019-02-11:
You can also jump two pins on the powerboard - the ribbon cable plug that normally goes out to the hub- to give power to the g2 when hub is not connected. Looking down on the powerboard with front of bot closest to you...Jump the bottom lower pin in the left row and the third pin up from bottom in the right row

Bob-the-Kuhn, 2019-02-11:
I  HAVE POWER !!!!

All it took was finding a connector that had a fat center pin to match my wall warts and realizing that pin one on the connector housing wasn't actually the pin one of the board.  Someone needs to take a basic technician course.

drphil3d, 2019-02-11:
[image]

abdrumm, 2019-02-11:
Woohoo! Power!

Bob-the-Kuhn, 2019-02-11:
How do you get it to send something to the G2?  I've tried tying CTS high & low but nothing happened.

Bob-the-Kuhn, 2019-02-11:
Where is the MK20 software?

Where is the G2 serial init routine?

drphil3d, 2019-02-11:
It’s all in the printrhub repo. I’ll see if I can dig some of it up
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

Bob-the-Kuhn, 2019-02-11:
I can see RS232  data being transmitted by G2 and on the FTDI port.

dmsmartinsUS, 2019-02-12:
Hello guys.. I have been digging the printrhub.. Maybe I can help. Where you should look is:
https://github.com/Printrbot/Printrhub/blob/master/mk20/src/Printr.cpp
I believe that the g codes are sent direct through the serial.
I'm looking now what is the SendLine to make sure.

dmsmartinsUS, 2019-02-12:
So I found the pin's used at the printrhub:
https://github.com/Printrbot/Printrhub/blob/master/mk20/src/framework/core/HAL.h
Has the serial and other pins..

[85 more replies]

closed 2023-01-13 by thisiskeithb