MarlinFirmware/Marlin #13021: [Question] TMC and dedge

GMagician, 2019-01-26:
I'm wondering if, for people using TMC (I don't), using `dedge` may help people with 8bit boards to get higher speed (or increase microsteps) since in this way same speed may be reached with half step freq

teemuatlut, 2019-01-27:
The steps are not driven in 50% "duty cycle" but rather the pulse is terminated right as soon as minimum pulse length is satisfied. I think this irregularity could cause issues.

What would be easier to determine is if using microstepping setting of 8 would have any discernible print quality difference. With TMCs everything can be interpolated to 256 anyway.

GMagician, 2019-01-27:
> The steps are not driven in 50% "duty cycle"

I agree with you. That may cause issues

> With TMCs everything can be interpolated to 256 anyway

that's true and of course, as state more than once, microstepping will not for sure help in printer quality since one thing is theoretical position (the one you get with microstepping) and the other is the real position (tha's depend on microsteps, inertia, friction and many other physical things)

closed 2019-01-27 by GMagician

r-a-i, 2019-05-27:
Hi, 
I'm adapting Marlin (working with a fork from 1.1.8x) for use in an application similar to 3DPrinting where I can drive the STEP signal with a 50% duty cycle and want to use the double edge feature of TMC2130. 

Looking through the code and probing the TMC2130 by adding functions to Marlin to report the state of the DEDGE bit, it seems like by default Marlin configures the TMC2130.  This is what I observe when I probe the TMC2130 using M122 and and extended status function that reports the state of DEDGE. So the bit seems to be set to 1, which according to the data sheet enables double edge pulses. 

Am I missing something? Are we using these drivers in double edge mode?

teemuatlut, 2019-05-27:
https://github.com/teemuatlut/Marlin/commit/b00a597d8d53e800c887b439e59737a3e1e86957

[image]

Already working on it but there are a few things that take priority. That commit is a mostly at proof of concept stage at this point.
Marlin does not currently use dedge in any way.

r-a-i, 2019-05-27:
Cool! What is the simulator/waveform viewer you are using? 

I checked the status of the driver again and saw dedge was sometimes true and other times false. This also happened when I enabled all 4 drivers. This happened while I was tuning the max current (906) and set it so high that the motors stopped and then restarted (instead of constant motion). My guess is that the high current created a glitch that messed up the state of the drivers.  Now that I have the current set I always see dedge on all drivers set to false, as expected. 

I too have other priorities but hope to check out your new branch soon.

teemuatlut, 2019-05-27:
PulseView. Free & Open Source and works with my 10€ logic analyzer.

The current code base should never set the dedge bit and I'm more inclined to suspect a bad read or a misinterpretation. Do you have a `M122 V` output that would show this?

r-a-i, 2019-05-27:
Thanks for the waveform pointer.  What is that 10€ logic analyzer you speak of? 

I searched the whole codebase and I agree that the code never sets the dedge bit. That is why I was so puzzled and motivated me to post here to confirm. 
I extended the M122 status/debug to read dedge, and I do have many M122 reports that showed dedge=1 for one driver and 4 driver setups. Even more, the motors would run in a jerky/choppy way, instead of the smooth motion, which suggests that indeed the bit was set. Resetting the system (I'm running on MEGA2560) brought back dedge=0 and smooth motion. I really think this is due to the system running way more current that it could hold and causing a voltage drop or some spike that corrupted the state. As long as we don't operate in that regime (which we shouldn't) I don't think we need to trace this issue further.

teemuatlut, 2019-05-27:
It's the HobbyComponents 8ch 24MHz one. Might very well be the same as you have.

Bob-the-Kuhn, 2019-05-27:
I switched from PulseView back to Saleae Logic.   I dumped the sigrok software because the 'scope driver it installed wouldn't allow me to see the logic analyzer portion of my Hantek 6022BL.

The only thing I don't like about Saleae Logic is that it displays only 8 of the 16 channels on my Hantek 6022BL.  But that's only caused me inconvenience a few times.

Never did find a useable software suite for both the 'scope & logic portions of the Hantek 6022BL.