MarlinFirmware/Marlin #13023: Wrong processor for recent STEVAL-3DP001v1 implementation.

wesk-ms, 2019-01-27:
Hello!

Over the past couple of weeks, I've been eagerly watching #12894, #12895, and #13011 and very excited to see support for ST's controller added to Marlin 2.0.x.  I've had one of these boards for a couple years now and never could get Marlin4ST to work correctly.  Nice work @Bob-the-Kuhn and @thinkyhead!

I wish I would have caught this a couple weeks ago, but while trying to figure out some compilation errors, it hit me that this board actually uses a STM32F103CBT6 processor, not the STM32F4.  I've validated this visually on my board and in the [BOM published by ST](https://www.st.com/content/ccc/resource/technical/document/bill_of_materials/group0/de/4b/d0/f1/40/64/45/57/STEVAL-3DP001V1%20BOM/files/steval-3dp001v1_bom.pdf/jcr:content/translations/en.steval-3dp001v1_bom.pdf).

I plan to go through the commits above to see if I can get it working, but probably won't get time to do this until later this week.

Thanks!
Wes

Bob-the-Kuhn, 2019-01-27:
I just pulled down the BOM & schematic from the ST site.  They show both a STM32F401VET6 and a STM32F103CBT6.  The 401 is the main CPU and is on page 6 of the schematic.  The 103 is on  page 7 which is labelled STLINK V2.

There's some bugs in the current 2.0.x.  See PRs #12 (for the library) and #13011 (for 2.0.x).

wesk-ms, 2019-01-28:
Thanks!  I'll close this issue.

closed 2019-01-28 by wesk-ms