MarlinFirmware/Marlin #13025: [bugfix-2.0.x] TUNE menu is missing

dahuafschmied, 2019-01-27:
since #12926 the TUNE menu is missing while printing with octoprint.
the first entry is PAUSE PRINT the next is CONFIGURATION

here is my repo:
https://github.com/dahuafschmied/Marlin-for-Anycubic-i3-Mega

GMagician, 2019-01-27:
Menu is still there in code, when printer is printing...

Source line 110 in Marlin/src/lcd/menu/menu_main.cpp

GMagician, 2019-01-27:
Right, missed it

J-charles-C, 2019-01-29:
I,

same problem for me. 
Parts of menu are in english and not traduce to french. I have choice fr in configuration.H

I have the last commit.

best

J-charles-C, 2019-01-30:
I,
Thanks @GMagician .

best

GMagician, 2019-01-30:
Yes but I suspect that who made that code forgot a menu item there but I don't know what was its original intent

InsanityAutomation, 2019-01-30:
Ill double check if anything is missing from what I had intended tonight. Been alot of hands in this area lately and more than a few overlapping merges in the past couple weeks.

InsanityAutomation, 2019-01-30:
It looks like the PR's that added host command support for the pause / resume from the LCD and the ones moving the sd file menu around collided and left some crap behind. Im 90% sure thats the original conditions around the abort menu before it was modified to support an action command. If so, it can come out with the surrounding preprocessor. Ill verify shortly, and if so throw up a new PR to supersede that one.

dahuafschmied, 2019-01-31:
Fixed. Thank You

PR #13049 (2019-01-31): Always show Tune sub-menu
This should fix #13025.
I just added a ';' because maybe something I don't know is missing in if statement
closed 2019-01-31 by dahuafschmied