MarlinFirmware/Marlin #13028: [2.0.x] m125 message

GMagician, 2019-01-27:
I'm wondering if message that appear when M125 and LCD pause/M25 (when `PARK_HEAD_ON_PAUSE`)/M125 is called shouldn't be changed.

Not message is displayed with:
`lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_INIT, ADVANCED_PAUSE_MODE_PAUSE_PRINT, active_extruder);`
that is the same used by M600 and write something about "filament change"...

I can fix it if there is no reason for it

GMagician, 2019-01-27:
Ok...I'm trying to understand code but there are redundant calls in my opinion.
When `PARK_HEAD_ON_PAUSE` is enabled, before enqueuing M25 P, `lcd_advanced_pause_show_message` is called, but also called because of P and finally pause_print is called with lcd.. argument false but if it was true...same call. Not to mention that P is not a defined argument (so is it an "hidden" one?)
I think that code may be reduced a little bit. I'll post a PR with such cleanup and ready to change message as state above.

InsanityAutomation, 2019-01-28:
Message was put in because resume was broken with no LCD displays when parked. That fixed it, but changing the message to be more generic landed on my to-do list. I planned on getting it this week along with the rest of the pause / resume streamer changes we've been going over.

closed 2019-01-29 by GMagician