MarlinFirmware/Marlin #13029: [1.1.9] Z axis continuously moves up during leveling

scgtrp, 2019-01-27:
### Description

I am running Marlin on a custom CNC mill. I have endstops on both ends of the X and Y axes, an endstop at the bottom of the Z axis (connected to Z_MIN_PIN) and a Z "probe" (connected to Z_MAX_PIN). The "probe" consists of a pair of wires; one is connected to the engraving bit and the other is touching my workpiece (the surface of a blank PCB), such that when they touch each other the input pin is grounded. I am trying to use Marlin's bed leveling on a rectangle within the PCB to make the engraving depth across the PCB area consistent.

### Steps to Reproduce

- home all axes
- manually move to the top left corner of the rectangle, with the engraving bit just above the PCB
- G92 X0 Y0 Z0
- G29 V2 T L0 R75 B0 F-50 S300

**Expected behavior:** The usual leveling thing happens.

**Actual behavior:** When I run G29, the mill moves to the bottom right corner of the rectangle, then moves left from there. At each point where it should stop and move down to probe, it instead moves *up* ~1cm, moving in a staircase pattern in the XZ plane until I stop it or it hits the top of the Z axis.

#### Additional Information

* I have verified with M119 that all endstops are working correctly, and that the probe reports "triggered" when it is in contact with the PCB and "open" when it is not.
* I have also tried the `bugfix-1.1.x` branch, with no change in behavior.
* [Configuration.h + Configuration_adv.h](https://github.com/MarlinFirmware/Marlin/files/2800576/configuration.zip)

crystalhand, 2019-01-29:
I would be willing to bet it is an endstop issue.  You see the auto raising happening because the endstop triggers right after it starts to probe.  so it raises the 1mm that you have set in between probe locations.  I had the same thing a while back and it took me forever to sort out.   You may need to take a look at the probe logic in the config file to make sure you everything is set properly.  It could be you have reversed logic or NO/NC set wrong or maybe even the pinout is slightly off.  I bet if you reversed the zprobe logic it would never trigger and just go straight down.

scgtrp, 2019-01-30:
@crystalhand Am I right in assuming the probe should be "triggered" when touching the bed/workpiece? That seems to be what's currently happening.

AnHardt, 2019-01-30:
Equally important is: 
The probe shall not show triggered when not touching the bed/workpiece.
Additionally no unused endstop shall appear in M119.

scgtrp, 2019-01-31:
@AnHardt Both of those are also the case; I have five endstops and one probe listed by M119 and all of them behave as expected.

marcio-ao, 2019-02-01:
@scgtrp: Try turning off the extruder motor (M18), heaters and fans prior to probing. This will rule out electrical interference from those devices. We had a situation once where our probes were failing in this manner when the extruder motor was energized.

scgtrp, 2019-02-01:
@marcio-ao It's a mill; there is no extruder motor or heater, and the only fan is above the RAMPS board cooling the stepper drivers. The spindle is turned off.

I suppose it's possible one of the other motors (or maybe that fan?) is interfering. I'll poke it with a multimeter/oscilloscope tonight and let you know what I find.

marcio-ao, 2019-02-01:
@scgtrp: Oh, sorry. I didn't read carefully enough. Might be easier to disconnect everything that isn't necessary for probing from the board to rule out interference. All I can say is that from my experience, interference is sometimes the cause of the issue you are seeing.

marcio-ao, 2019-02-01:
@scgtrp: You could also fiddle with the ENDSTOP_NOISE_THRESHOLD option in "Configuration.h". We've seen that make a difference too.

scgtrp, 2019-02-02:
So I connected an oscilloscope to it, and yep, it's definitely interference.

I was getting some from having the steppers energized, but most of it came from the spindle's power supply, which was emitting a *lot* of noise and making the input wildly flail between 0-6V. Unplugging that brought the noise down just enough for it to behave normally.

I'll close this now, as I'm now convinced this is entirely a hardware issue. Thanks for the help.

closed 2019-02-02 by scgtrp