MarlinFirmware/Marlin #13031: [bugfix-1.1.x] Select Mk2 multiplexer and dual Z Motor not Work

madnessracer, 2019-01-27:
Hello
I use a Trigorilla board (Ramps 1.4) with two Z-steppers and two limit Z-switches (min Z and max Z).
Works without problems.

Unfortunately, with this configuration, in the config.h option "MK2_MULTIPLEXER" 
can only be activated with the hint "MK2_MULTIPLEXER requires 2 or more EXTRUDERS."

If you change the number of extruders to 4 then you will get the message "No E stepper plug left for Z2"

But no more step driver is needed.

revilor, 2019-01-28:
Have you also activated SINGLENOZZLE ?

madnessracer, 2019-01-28:
I tried it with #define SINGLENOZZLE and without #define SINGLENOZZLE.
Both give the same result.

boelle, 2019-02-21:
@madnessracer problem solved?

madnessracer, 2019-02-23:
I hope that when I compile I get no more errors, but I could not test it yet

boelle, 2019-02-23:
remember to test with bugfix 2.0

boelle, 2019-03-13:
@thinkyhead think we can assume OP dead on this one

closed 2019-05-16 by thinkyhead