MarlinFirmware/Marlin #13033: Backlash Compensation

cinealfa, 2019-01-28:
### Description

BAd stuttering on arcs 
### Steps to Reproduce

With BC turned on motors miss steps stutter goes away with it turned off.

**Expected behavior:** [What you expect to happen]
Dunno what should it do?

**Actual behavior:** [What actually happens]
Whole machine shakes violently and motors miss steps
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2801077/Archive.zip)
The config file contains everything I am currently using except the backlash is off.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AnHardt, 2019-01-28:
Stuttering where on the arcs? When an axis changes direction? 4 times in a circle?

Likely you are overcompensating a lot. As soon as the mass (bed, x--carriage) is moving, during the compensation, you are overcompensating. You have to minimize the not-moving - not more.

RudyBenner, 2019-01-28:
Compiling .pioenvs\LPC1768\src\src\lcd\dogm\status_screen_DOGM.cpp.o
In file included from Marlin\src\lcd\dogm\status_screen_DOGM.cpp:32:0:
Marlin\src\lcd\dogm\dogm_Statusscreen.h:46:12: fatal error: ../../../_Statusscreen.h: No such file or directory
#include "../../../_Statusscreen.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\LPC1768\src\src\lcd\dogm\status_screen_DOGM.cpp.o] Error 1

______________________
Tried several times.

rudy

marcio-ao, 2019-02-01:
@cinealfa: I just tried the example from the G2 page, `G2 X125 Y32 I10.5 J10.5` on our printers, which use backlash compensation and I saw no unexpected vibrations. Could you post the output of `M425` so we can see how your backlash values are configured?

cinealfa, 2019-02-01:
After much fooling around, I think I've solved the problem. It seems to be related to the GCC issue with the step pulses when I upped the step pulse to 4 the problem went away.

closed 2019-02-01 by cinealfa

marcio-ao, 2019-02-01:
@cinealfa: What's this GCC issue? Do you have a ticket number?