MarlinFirmware/Marlin #13037: Ender 3 buzzer issues/bltouch

kingontheroad, 2019-01-28:
I would like to have a bl-touchon my Ender 3 pro.

To configure Marlin, I use the teaching tech tutorial on youtube.

To save space, I have the speaker / buzzer uncomment.
// #define speaker and in the pins_MELZI_CREALITY.H tab
I added the #undef BEEPER_PIN line.

When arduino has finished uploading, the buzzer is running out of control. An irritating hum sound.

Who knows how I can solve this?

nick-hunter, 2019-01-28:
The buzzer is normally connected to pin 27 on the Ender 3. When installing the BLTouch you are supposed to wire that up to the BLTouch, but it seems like your pin 27 is still connected to the buzzer. If you have everything installed correctly, you shouldn't hear the buzzer at all.

If you look around online you can get a "Pin 27 Adapter Board" which will allow you to connect the BLTouch without cutting any wires. You can also just cut the wire going to the buzzer and connect it to the BLTouch, but I would highly recommend finding a good tutorial before cutting anything.

klaus4040, 2019-02-01:
Can confirm this is it, Tried the new firmware to check correct flash and didn't install the adapter yet. Makes sense once you know the buzzer pin.

boelle, 2019-02-21:
@kingontheroad problem solved?

closed 2019-02-23 by thinkyhead