MarlinFirmware/Marlin #13040: Marlin 2.0 32 bit Linear Advance error

Milu1980, 2019-01-28:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Milu1980, 2019-01-28:
Hi,
Linaer Advance not working on Marlin 2.0 with 32 bit board.
if i set it to 0 and i want to print a test, the Extruder engine do Nothing... like disabled...
can somebody help me please ?

on 1.1.9 with 8bit board workin good.

thx.

InsanityAutomation, 2019-01-28:
This is a duplicate of #12870 - can you close this and continue the discussion there?

SupremeFPV, 2019-01-28:
See #12983 for workaround.

Milu1980, 2019-01-28:
Sorry, i am new...
What can i do now ?
Is was fixed ?
Can i get a new fixed version ?
Thx.

marcio-ao, 2019-01-31:
> See #12983 for workaround.

I can confirm that setting MINIMUM_STEPPER_PULSE to 1 on Archim 2.0 solves the problem.

It seems like the trouble lies in the following code in "Conditional_post.h":

#ifndef MINIMUM_STEPPER_PULSE
  #if HAS_DRIVER(TB6560)
    #define MINIMUM_STEPPER_PULSE 30
  #elif HAS_DRIVER(TB6600)
    #define MINIMUM_STEPPER_PULSE 3
  #elif HAS_DRIVER(DRV8825)
    #define MINIMUM_STEPPER_PULSE 2
  #elif HAS_DRIVER(A4988) || HAS_DRIVER(LV8729) || HAS_DRIVER(A5984)
    #define MINIMUM_STEPPER_PULSE 1
  #elif TRINAMICS
    #define MINIMUM_STEPPER_PULSE 0
  #else
    #define MINIMUM_STEPPER_PULSE 2
  #endif
#endif
This apparently defaults to 0 when using TRINAMICS. My guess is that this is fine (maybe even required) on the slower AVR chips, but on the faster 32-bit chips, the pulse width becomes too small. Perhaps this should be updated so on faster chips a default of 1 is used for TRINAMICS?

@teemuatlut: @thinkyhead: Thoughts on implementing this idea?

Milu1980, 2019-01-31:
Hi,

I have 2208 inside...

marcio-ao, 2019-01-31:
@Milu1980: Edit "Configuration_adv.h" in the "Marlin" directory and change this line:

`//#define MINIMUM_STEPPER_PULSE 2`

To:

`#define MINIMUM_STEPPER_PULSE 1`

Recompile, flash and see if it helps.

teemuatlut, 2019-01-31:
The datasheet gives us a parameter for `STEP input high time`.
It is determined by `t_FILTSD` and `t_clk + 20`, which ever is higher.
`t_FILTSD` ranges from 36 to 85 (min, max) with 65 given as a typical value.
`t_clk` is 1/12,65MHz = 79ns, therefore it is higher at around 100ns or 0.1ms.
Step signal low time is limited in the same manner.

I don't know linear advance logic well enough to say why the problem would be present only with it enabled.
If using a setting of `1` is enough to remedy the issue, I have no objections to changing the parsed default value.

marcio-ao, 2019-01-31:
> therefore it is higher at around 100ns or 0.1ms.

You mean 0.1us rather than 0.1ms? Anyhow, 1us certainly is greater than 0.1us, so setting MINIMUM_STEPPER_PULSE to 1 would certainly be sufficient.

It seems the Due runs at 84Mhz, which means that at an approximation, every instruction could run every 10ns. Certainly it is possible that if Marlin is executing fewer that 7 or so instructions between toggling the pins, it would be too fast for the Trinamic drivers. In [this thread](https://github.com/MarlinFirmware/Marlin/issues/12983#issuecomment-457087577), @HackingGulliver said he was seeing shorter pulses on the E stepper vs the XY steppers, but he did not say how long they were.

SupremeFPV, 2019-01-31:
I can get some oscilloscope data of the stepper output if it will help?

marcio-ao, 2019-01-31:
@SupremeFPV: If that is something you can do easily, it would be great to verify that the step signal lengths were within the acceptable ranges. Maybe try it for regular extrusion on E as well as for babystepping on X and Y.

SupremeFPV, 2019-01-31:
I'll see what I can do over the weekend. I have to bring my printer to work then.
ill do E with/without lin_adv. I havent used babystep on x/y before, is this the same as on Z?

marcio-ao, 2019-01-31:
Yes. Although measuring it on Z is probably sufficient. It is all the same code.

Milu1980, 2019-02-03:
Hi, yes its working but bad...
I set it 10 lines and do this:

Milu1980, 2019-02-03:
[https://m.facebook.com/story.php?story_fbid=2457112424316157&id=100000522704728](url)

SupremeFPV, 2019-02-03:
here we go. 
[https://imgur.com/a/dfQXYT3](Here)
I was not able to reproduce any problems with babystepping, but the step pulse when lin_adv is on and pulse length is 0 is extremely small (8us) and looks nothing like a square wave at all. Is this something that might be a issue with rise/fall time on the MCU?

Milu1980, 2019-02-03:
With "K0" can be set the correct line...
If i set the pattern to K 0.2-0.3 with 0.01 step, in 8 bit board do 10
lines, in 32bit just 2-3 and thats was :( it break just the print...

marcio-ao, 2019-02-06:
@SupremeFPV: Good work! I think this confirms why we've been having good luck with MINIMUM_STEPPER_PULSE set to 1.

I guess the question now is whether and how Marlin should be patched. Either we make a choice for all 32-bit boards or perhaps we could add MINIMUM_STEPPER_PULSE to the "pins.h" files corresponding to boards where this issue crops up.

deamoncrack, 2019-02-13:
Hello !

It does'nt work for me...

LIN_ADVANCE K 0 is ok, K 0.1 make the extuder stops. 

I use Malin bugfix 2.0 on MKS SGEN (Sbase settings with TMC2208_STANDALONE drivers ).

mhmdrefaat60, 2019-02-21:
layers shift when the printer climbs layer Can help I use tb6600
[image]

boelle, 2019-02-21:
@Milu1980 problem solved?

Milu1980, 2019-02-22:
Not really...
Its working, but the pattern looking not the samme like with the 8bit
board..
I cant upload the pic.

boelle, 2019-02-22:
just copy and paste the pic

but you can not do it from email

boelle, 2019-03-06:
remember to upload pic

just copy and paste it in here... crtl+c then here ctrl+v

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

juchong, 2019-07-11:
For anyone looking, I can confirm that setting MINIMUM_STEPPER_PULSE to 1 fixed extruder stepping on my SKR V1.3 board running Trinamic TMC2130 drivers after enabling Linear Advance.

thinkyhead, 2019-07-11:
> For anyone looking, I can confirm that setting MINIMUM_STEPPER_PULSE to 1 fixed extruder stepping on my SKR V1.3 board running Trinamic TMC2130 drivers after enabling Linear Advance.

@teemuatlut — Do you think this should apply to all "`TRINAMICS`" stepper drivers? We now have this:
```cpp
#ifndef MINIMUM_STEPPER_PULSE
  #if HAS_DRIVER(TB6560)
    #define MINIMUM_STEPPER_PULSE 30
  #elif HAS_DRIVER(TB6600)
    #define MINIMUM_STEPPER_PULSE 3
  #elif HAS_DRIVER(DRV8825)
    #define MINIMUM_STEPPER_PULSE 2
  #elif HAS_DRIVER(A4988) || HAS_DRIVER(A5984)
    #define MINIMUM_STEPPER_PULSE 1
  #elif HAS_DRIVER(LV8729)
    #define MINIMUM_STEPPER_PULSE 0
  #elif TRINAMICS
    #define MINIMUM_STEPPER_PULSE 0
  #else
    #define MINIMUM_STEPPER_PULSE 2
  #endif
#endif
```
…And apparently zero is just too small for these drivers on 32-bit boards. We're currently specifying these in µs, so maybe, if needed, we can graduate to ns and set this pulse length to ~500….

thinkyhead, 2019-07-11:
@juchong — Are you using `TMC2130_STANDALONE` or `TMC2130`? (I'm assuming the latter.)

teemuatlut, 2019-07-11:
I think dedge is a more elegant solution that doesn't increase the time spent in ISR.
But we can do it if the user has Lin Adv enabled, dedge disabled and at least one TMC.
Bonus points if it gets documented somewhere.

juchong, 2019-07-11:
@thinkyhead `TMC2130` - I'm connecting to the drivers via SPI. Out of curiosity, is the limitation the timer resolution or the actual interface hardware? The "non-square pulse" shown in the imgur album linked above looks suspiciously like what happens when a GPIO is set and reset in concurrent CPU instructions. Specifying the minimum pulse in uS would likely help, and since 2.0 is targeted at 32-bit processors, there really isn't a need to keep compatibility with legacy boards, right?

teemuatlut, 2019-07-11:
The #1 priority of Marlin v2 is to keep compatibility with 8bit boards. They're the vast majority of the user base and will be for a long time.

marcio-ao, 2019-07-11:
@thinkyhead: Our printers use TMC2130 and we needed to set MINIMUM_STEPPER_PULSE to 1.

teemuatlut, 2019-07-11:
Lin adv on or off?

marcio-ao, 2019-07-11:
@teemuatlut: With LIN_ADV, we needed to set MINIMUM_STEPPER_PULSE to 1 with the TMC2130 in order for the extruder motor to run on the Archim. With LIN_ADV off, the default of zero worked fine.

On the 8-bit AVR, MINIMUM_STEPPER_PULSE of zero works both with LIN_ADV on and LIN_ADV off.

teemuatlut, 2019-07-11:
Well that's nothing new. The conditions however are starting to stack up for the issue to occur, but as said, I'm fine with increasing the default if these conditions all are true in the config.

daiben2, 2019-07-16:
I'm haveing the same issue Ender 3 skr 1.3 tmc 2208 standalone drivers for all motors.
If linear advance uncommented the extruder stops with the value of 1 in the minimum pulse it extrudes but under printing it's not. Same with value 2

EvilGremlin, 2019-07-16:
Minimum pulse have nothing to do with this issue. Just don't use 2208 on extruder with stealthchop enabled.

daiben2, 2019-07-16:
How can I disable it or what to do?

EvilGremlin, 2019-07-16:
Only through UART.

juchong, 2019-07-16:
I just re-tuned the extruder after setting MINIMUM_STEPPER_PULSE to 1. Haven't had issues since.

SiskoBen, 2019-08-15:
Hi I ran into this as well. I have an SKR v1.3 with TMC2130 drivers in spi mode. I found this thread which  helped to resolve the issue (MINIMUM_STEPPER_PULSE to 1).

What i do not get is how a MINIMUM_STEPPER_PULSE at 0 would even be possible, apart from how difficult it would be to create such a zero length pulse ;-) the stepper would never move as it would not be able to differentiate a zero length pulse from no pulse :-P.

This is clearly a bug MINIMUM_STEPPER_PULSE should never be set to 0. at the very lowest value it should be something like "BEST EFFORT SHORTEST PULSE".

AnHardt, 2019-08-15:
That's tried to express by **MINIMUM**_STEPPER_PULSE.
On the AVRs 0 used to be no problem because the processors where not able to switch the pins fast enough to hurt.

Black6spdZ, 2019-08-17:
so what is the consensus here? any TMC driver used with extruder set min step pulse to 1 OR disable stealthcop?

ritor1sefa, 2019-09-17:
I test it
board: scr 1.1 pro
drivers: tmc5160
uncomment pulse and set 1 => nothing change; "#define MINIMUM_STEPPER_PULSE 1"
comment stealthchop AND comment pulse  => Working!
//#define STEALTHCHOP_E
//#define MINIMUM_STEPPER_PULSE 1

enjoy

boelle, 2019-10-12:
@Milu1980 tried what @ritor1sefa suggested?

jharmn, 2019-10-14:
Confirming fix in my setup: SKR 1.3 + TMC2208 (delta configuration)
Enabling LIN_ADVANCE disabled my extruder stepper
Changing MINIMUM_STEPPER_PULSE to 1 re-enabled it.

boelle, 2019-10-14:
oki will close this one as its fixed

closed 2019-10-14 by boelle

gannon145, 2019-11-20:
SKR mini DIP 1.0,  2208s, Stealthchop disabled on E stepper.

2.0.x firmware from Bigtree I got Lin_Adv working

On recent pull of 2.0.x build when I turn on Lin_Adv I don't get any E movement.

MINIMUM_STEPPER_PULSE to 1  did NOT work for me either.

yoc1983, 2020-02-05:
also here: skr 1.3, TMC 2208, marlin 2.0.3
whatever LIN_ADVANCE= on/off 
MINIMUM_STEPPER_PULSE=off/0/1
nothing work
stepper give slow and not correct movement

thinkyhead, 2020-02-07:
@yoc1983 — Please open a new issue and fill in the template. Test the latest `bugfix-2.0.x` first.

@gannon145 — Please open a new issue and fill in the template. Test the latest `bugfix-2.0.x` first.

djessup, 2020-03-18:
FYI Same issue with no practically no extrusion on stock Creality Ender 3 1.1.5 board, TMC2208 drivers set as TMC2208_STANDALONE in Configuration.h

Manually extruding via Prontiface or similar works fine, but not during printing with linear advance enabled. After disabling it it extrudes fine for the first few layers then starts having issues - not sure if it's skipping steps or something else, going to do some further testing to see if I can isolate the issue. 

If anyone knows if it's possible to run this board+drivers in UART mode I'd appreciate any pointers.

[image]
All steps/mm are set correctly. Something is going awry during the print.

InsanityAutomation, 2020-03-18:
This is a known and well documented issue with LA and TMC drivers in stealthchop mode. We have a line on a method to resolve it but it basically means a rewrite of how LA injects steps, as the interpolation function of stealthchop cant handle the sudden change in the pulse train.

thinkyhead, 2020-03-18:
> the interpolation function of stealthchop cant handle the sudden change in the pulse train

It'll be good to see this on an oscilloscope. The E stepper should be following its own trapezoid acceleration / deceleration trajectory, and pulses should be as regular as with the other axes. Is there something related to a too-high E Jerk that could be exacerbating this? I seem to recall that E Jerk needs to be in a limited range to work well with LA.

InsanityAutomation, 2020-03-18:
@sjasonsmith had a logic analyzer trace showing LA injected pulses very close to the normal step train.

sjasonsmith, 2020-03-18:
The normal part of the move and LA inject steps independent of one another. This leads to an effective double-step when they happen to align within the same ISR, which the drivers don’t like. I had some workarounds that improved things, but there were always more edge cases that would mess up, and I had to set it aside.

djessup, 2020-04-17:
FWIW I ended up replacing the stock Creality board with an SKR 1.4 and TMC2209's. This combo is fine in StealthChop w/ LA enabled.

One thing I have noticed is with UBL enabled there are micropauses when the nozzle passes over/within ~1mm of the UBL grid points. This results in slight over-extrusion at these areas and mild surface artefacts. I saw earlier reports about something similar on grid boundaries, which appears to have been fixed in more recent versions but there is still something going on at the grid corners. Any chance this is in any way related or should I be filing another ticket?