MarlinFirmware/Marlin #13041: [2.0.x] Distinct E Factors without defining additional factors throws Sanity Check failures

XBrav, 2019-01-29:
With Distinct E-Factors enabled, the additional extruder values need to be defined as a part of the array, otherwise the below errors are thrown. This is contrary to the text which states "With this option each E stepper can have its own factors for the following movement settings. If fewer factors are given than the
 total number of extruders, the last value applies to the rest."

To recreate, compile the attached Config / Config_Adv.h for a MEGA2560. To remove the error, manually define the additional E Factors.

`Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz 8KB RAM (248KB Flash)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.2.3
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|-- <Adafruit NeoPixel> 1.1.3
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <c1921b4>
|   |-- <SPI> 1.0
|-- <SailfishLCD> #c8ac22f
|-- <SailfishRGB_LED> #2426fa2
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster> #3a18be5
|-- <Wire> 1.0
|-- <SPI> 1.0
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\HAL.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\HAL_spi_AVR.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\MarlinSerial.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\persistent_store_eeprom.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\servo_AVR.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\HAL_AVR\watchdog_AVR.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\I2cEeprom.cpp.o
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\SpiEeprom.cpp.o
In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\HAL.cpp:29:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1928:1: error: non-constant condition for static assertion
static_assert(sanity_arr_2[0] > 0 && sanity_arr_2[1] > 0 && sanity_arr_2[2] > 0
^
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1928:1: error: array subscript out of bound
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1936:1: error: non-constant condition for static assertion
static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0
^
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1936:1: error: array subscript out of bound
Compiling .pioenvs\megaatmega2560\src\src\HAL\shared\backtrace\backtrace.cpp.o
In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\HAL_spi_AVR.cpp:40:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1928:1: error: non-constant condition for static assertion
static_assert(sanity_arr_2[0] > 0 && sanity_arr_2[1] > 0 && sanity_arr_2[2] > 0
^
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1928:1: error: array subscript out of bound
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1936:1: error: non-constant condition for static assertion
static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0
^
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1936:1: error: array subscript out of bound
In file included from Marlin\src\HAL\HAL_AVR\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\HAL_AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:1928:1: error: non-constant condition for static assertion
[... 55 lines cut ...]
*** [.pioenvs\megaatmega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\shared\I2cEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1928:1: error: non-constant condition for static assertion
static_assert(sanity_arr_2[0] > 0 && sanity_arr_2[1] > 0 && sanity_arr_2[2] > 0
^
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1928:1: error: array subscript out of bound
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1936:1: error: non-constant condition for static assertion
static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0
^
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1936:1: error: array subscript out of bound
*** [.pioenvs\megaatmega2560\src\src\HAL\shared\I2cEeprom.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:41:0,
from Marlin\src\HAL\shared\SpiEeprom.cpp:28:
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1928:1: error: non-constant condition for static assertion
static_assert(sanity_arr_2[0] > 0 && sanity_arr_2[1] > 0 && sanity_arr_2[2] > 0
^
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1928:1: error: array subscript out of bound
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1936:1: error: non-constant condition for static assertion
static_assert(sanity_arr_3[0] > 0 && sanity_arr_3[1] > 0 && sanity_arr_3[2] > 0
^
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1936:1: error: array subscript out of bound
*** [.pioenvs\megaatmega2560\src\src\HAL\shared\SpiEeprom.cpp.o] Error 1`

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2805823/Config.zip)

boelle, 2019-02-21:
@XBrav problem solved?

XBrav, 2019-03-03:
Apologies @boelle, I've been out of town for a bit. I will look into it this weekend.

closed 2019-03-03 by thinkyhead