MarlinFirmware/Marlin #13042: [bugfix 2.0.x] Homing doesn't work

Svatevit, 2019-01-29:
### Description

In latest versions of 2.0.x homing doesn't work on my HC Evo. Z just don't lift itself, sometimes it doesn't even go down. I just hear noise from motors and nothing happens, than I get Homing Error. Z Axis is on A4988 with optical endstop, E is on A4988 and X/Y are on TMC2130 with SPI and Sensorless Homing. The board is ancient Gen7 v1.5. What is weird, when I move bed with a LCD it works fine, both ways. I don't have any auto leveling.

I have a 2.0.x from August 14th 2018 and this one works fine. I tried new version last week and it doesn't. 

### Steps to Reproduce

1. Choose Homing from menu

**Expected behavior:** Z goes down, X/Y goes to 0, Z goes to 0

**Actual behavior:** Z goes down, and nothing. Or, Z goes down, X/Y goes to 0 and nothing. Or just nothing at all.

Svatevit, 2019-01-30:
Ok, I noticed weird behaviour. After M501 everything works as expected. After M502 also. But when I do M500 and reset printer it gets some really odd steps/mm settings (way too high) for X/Y.

Svatevit, 2019-01-30:
If I do M502, it works.
If I do M500, M501 it works,
If I reset X/Y can't move. If I do M501 everything starts to work fine.

DavidThijs, 2019-02-02:
I have sometimes an issue when the machine was powered on while the X wasn't homed. When I do a G28 , my X goes not to the middle but only to 1/4. I that is followed by a G29 bilinear probe, X stays where it was when homing finished and then probes 3 times for each Y position. So X seems to be not moving. Powering the machine off, moving the carriage to it's home and then do the same again does end with the correct result.

boelle, 2019-02-21:
@Svatevit problem solved?

Svatevit, 2019-02-21:
Nope. I tried to load data from eeprom after test_connection is called but then test fails. On weekend I'm gonna test it with my new Due3dom LE. And I'm gonna change code a little to load setting once again after test_connection. I'm starting to think the problem is with tmc library, not with Marlin, but I'm not sure yet (not enough free time to test it all).

Svatevit, 2019-02-23:
Funny thing, on Due it works fine. I don't know what's the problem with ATmega, perhaps 20MHz clock makes something funny.

closed 2019-02-23 by Svatevit

boelle, 2019-02-23:
16/20 and less memory can funny things happen

spec if you try to do things faster than the hardware can do