MarlinFirmware/Marlin #13043: Q: RE-ARM communication speed

Ketchu13, 2019-01-29:
Hello,
I apologize for asking this question ... but I would like to know if the value of the baud rate defined in the Configuration.h file has an impact on the use of a RE-ARM.
It seems to me that RE-ARM automatically detects the USB host's baud rate, but I'm not sure I understand it.
is this baudrate value only used by the UART?

gloomyandy, 2019-01-29:
If you are connecting to the on chip USB port then the baudrate in Configuration.h is not used. The host will send details of the connection (PC side baudrate, stop bits etc.) but basically pretty much everything is ignored except for the DTE state which is in effect used to indicate if the connection is active or not.

Ketchu13, 2019-01-29:
ok, thanks.

If I understand correctly, if the host uses automatic baud rate detection (as Octoprint can do) and starts detection at 115200, the baud rate will be set to 115200 ?

gloomyandy, 2019-01-29:
It depends how it does automatic baud rate detection, but probably yes. Does it really matter though? If you want the host to use a high baud rate just set the rate on the host.

Ketchu13, 2019-01-29:
Thanks again, it was just to understand and know how it works.
I have the same problem described here => https://github.com/MarlinFirmware/Marlin/issues/9157#issue-288257666 but with the Re-Arm instead of the Mega2560, so I'm trying to understand what's wrong ..But this is not the subject of this post.

Regards

closed 2019-01-29 by Ketchu13

gloomyandy, 2019-01-29:
No problem I'm sure I've seen mention that some M43 features are not fully implemented on some of the 32bit builds, so you could be running into that. You may be better of just opening an issue for the problem you are seeing.