MarlinFirmware/Marlin #13045: [1.1.9] UBL probing failed in the same area

gazit6, 2019-01-29:
### Description

I have updated both of my printers to Marlin 1.1.9. I have used the same firmware for both of them(the only difference is the bed size in the X direction. In one printer UBL works without any issues and the other one always sends 'Probing Failed' in the same area of the bed. I am using an inductive sensor for bed leveling. 
I went back to 1.1.8 and the UBL process finished without an issue. 

I have tried the following: 

- Used the same firmware that works already with the small be size - nothing changed.

- Tried with hotend heaters and bed heaters ON and OFF - nothing changed.

- Cleared EEPROM and uploaded new firmware - nothing changed.

- Made sure nothing can disrupt the probing sequence - nothing changed 

**Expected behavior:** The printer should finish probing the bed. 

**Actual behavior:** Recieve probing failed in the front left corner area.

Any suggestions? 

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2807764/Config.zip)

atomicdad, 2019-02-05:
I'm investigating an issue with UBL that could be related. Do the probe offsets differ any between your machines? I recently changed the mount location for my probe. It is now X+ Y-, whereas it was X-, Y-. UBL fails from the get go trying to probe the open space in front of the bed. I'm 100% positive my probe offsets are correct.
In ubl_G29.cpp I noticed that the probe offsets are being subtracted from the reference position. I would think that they should be added to it. I made the change and my probe actually started on the bed instead of immediately failing off the front end. I didn't have a lot of time to tinker this evening and had to power down mid probe, as I ran into one of my bed clips with the nozzle. So far it appeared to probe the bed as it should. I'll give it a shot tomorrow and make sure the clips are out of the way.

atomicdad, 2019-02-05:
I'll probably end up opening a separate issue, as these could be unrelated. But if someone more knowledgable with the UBL code could take a look at how the probe offsets are being applied, I would appreciate it. If it is incorrect, it has been for a while.

boelle, 2019-02-21:
@gazit6 problem solved?

Roxy-3D, 2019-02-21:
> In ubl_G29.cpp I noticed that the probe offsets are being subtracted from the reference position. I would think that they should be added to it. I made the change and my probe actually started on the bed instead of immediately failing off the front end.

This was changed 2 weeks.   Is your code base older than that?   If so...  Please try the latest code.   This is what was changed:   https://github.com/MarlinFirmware/Marlin/commit/ab6908a513e9ffde6caf41b3f65b6b9a31e8b355#diff-33f8d18244cc3a9886d07b3a1692a9dc

closed 2019-05-16 by thinkyhead