MarlinFirmware/Marlin #13047: MKS SBase with TFT, serial error

J-PN, 2019-01-29:
I'm having the issue with enabling both serial ports working - this used to work with 5-Jan build. 

#define SERIAL_PORT -1
//#define SERIAL_PORT_2 0

/**works ok with Melzi display 
*/

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0

/**
gives error

Compiling .pioenvs\LPC1768\src\src\module\printcounter.cpp.o
In file included from Marlin\src\module\../inc/MarlinConfig.h:49:0,
from Marlin\src\module\configuration_store.h:24,
from Marlin\src\module\configuration_store.cpp:47:
Marlin\src\module\configuration_store.cpp: In function 'void say_M603()':
Marlin\src\module\configuration_store.cpp:2346:52: error: 'port' was not declared in this scope
void say_M603(PORTARG_SOLO) { SERIAL_ECHOPGM_P(port, "  M603 "); }
^
Marlin\src\module\../inc/../core/serial.h:82:63: note: in definition of macro 'SERIAL_ECHOPGM_P'
#define SERIAL_ECHOPGM_P(p,x)             (serialprintPGM_P(p,PSTR(x)))
^
Marlin\src\module\configuration_store.cpp:2346:52: note: suggested alternative: 'qsort'
void say_M603(PORTARG_SOLO) { SERIAL_ECHOPGM_P(port, "  M603 "); }
^

Marlin\src\module\../inc/../core/serial.h:82:63: note: in definition of macro 'SERIAL_ECHOPGM_P'
#define SERIAL_ECHOPGM_P(p,x)             (serialprintPGM_P(p,PSTR(x)))
^
Compiling .pioenvs\LPC1768\src\src\module\probe.cpp.o
*** [.pioenvs\LPC1768\src\src\module\configuration_store.cpp.o] Error 1
 [ERROR] Took 55.06 seconds
*/

gloomyandy, 2019-01-29:
Do you have EEPROM_CHITCHAT enabled? If not can you try enabling it and see if the error goes away?

J-PN, 2019-01-29:
Tnx, that did the trick

closed 2019-01-29 by J-PN

gloomyandy, 2019-01-29:
Actually I think it is a bug that it does not work if you don't have it enabled. My suggestion was just a test (though it is also a work around), to see if I was correct. Looks like some recent changes have broken things if you have your original config.