MarlinFirmware/Marlin #13048: [1.1.8] G29 results successfully stored to EEPROM, but no compensation actually done

fishcu, 2019-01-29:
### Description
I would like to avoid having `G29` before every print -- instead I would like to load stored EEPROM values and use those.
I tried different approaches and I can't get it to work. It only works if I do a `G29` JUST before every print.

Please help me debug the issue. Very happy to run commands to test.

### Steps to Reproduce

1. Do ABL, store to EEPROM, load from EEPROM, verify output: `G29`, `M500`, `M501`, `M420 V`.
All good, will show the estimated leveling parameters.
```
0.006 | 0.087 | 0.1 | 0.074 | 0.025
-0.088 | -0.022 | -0.038 | -0.056 | -0.105
-0.05 | -0.003 | -0.001 | -0.05 | -0.084
-0.017 | 0.035 | 0.04 | 0.03 | -0.017
-0.013 | 0.036 | 0.044 | 0.038 | 0
```
2. My start gcode (abridged):
```
[...]
;---------------------------------------—
; Preheating nozzle
M104 S{material_print_temperature}
M117 Homing and leveling
G28 ; Home
; G29 V3; automatic bed leveling <---- commented out for testing. Works if I uncomment!
;---------------------------------------—
[...]
;---------------------------------------—
M117 Purging nozzle
T{initial_extruder_nr}; set tool to initial one
M420 S1 ; resume Auto-leveling after extruder change
G92 E0
G1 F300 E-0.5
G1 X210 Y170 Z0.4 F8000
G1 F400 E8
G1 Y50 E23 F1000
G1 E22 F1000 ; retract a bit
G1 F600 Z1 ; pull away
G1 X50 F8000
```
3. Print will start, but no compensation on Z axis done. 

What I also tried so far:
- I have tried moving `M420 S1` to different locations, with no change.
- I have confirmed that there are no `G28` commands after the `M420` command.
- I have confirmed that there are no further tool changes (`T0` or `T1`) in the script.

#### Additional Information

Marlin version 1.1.8 from my repo, which is a fork of Geeetech's repo, which is derived from official Marlin. `Configuration` and `Configuration_adv` headers which have been flashed to the printer can be found here:
https://github.com/fishcu/Prusa_I3_3Dprinter/blob/master/A10M_marlin1.1.8_3Dtouch/Marlin/Configuration.h
https://github.com/fishcu/Prusa_I3_3Dprinter/blob/master/A10M_marlin1.1.8_3Dtouch/Marlin/Configuration_adv.h

Printer is a Geeetech A10M (Creality ender-3 clone with multi-color & mixing abilities). Uses GT2560 V3 board.

GMagician, 2019-01-29:
I think that here are not managed forked versions and I also think that 1.1.8 has been superseeded by 1.1.9 and not further fix will be done on it.
Better should be using 2.0.x bugfix and see if problem is still present there.
But that's what I think not law

fishcu, 2019-01-29:
Sure. My fork doesn't have any mods besides simple calibration (PID, bl touch probe offsets, etc.), and Geeetech's repo has actually had commits by thinkyhead. So I'm still hoping I can get some help, as it's not far from an official version.

reloxx13, 2019-02-02:
im on 2.0.x-bugfix, no problems with m420 s1.