MarlinFirmware/Marlin #13051: WIFI SD Card file directory not updating on Marlin BugFix-2.0.x

bcjams, 2019-01-29:
I am running Marlin Bugfix-2.0.x on a ReArm Ramps 1.4 using a Full Graphic RepRap Graphics card with built in SD reader.  I have a Toshiba FlashAir 16GB Wifi SD Card installed.   When I upload files from PC via WIFI I can verify that SD card has the new file, but the Marlin "Print from SD" menu does not list new file.  Resetting the ReArm controller, or alternately removing and reinserting the SD card will 'refresh' the files and they can then be selected via menu

I had no issues using this same SD Card when running an older controller with Marlin 1.8.5

[Documents.zip](https://github.com/MarlinFirmware/Marlin/files/2809903/Documents.zip)

p3p, 2019-01-29:
The Re-ARM has an onboard sd card slot that is more reliable (no long cable for data corruption) and does not interfere with the display so it is selected by default, to make Marlin use the sd card on the display you need to change a setting in the `pins_RAMPS_RE_ARM.h` file.

```
#define LPC_SD_LCD              // Marlin uses the SD drive attached to the LCD
//#define LPC_SD_ONBOARD        // Marlin uses the SD drive on the control board
```

The display will show some small errors while accessing the card when using this sd card slot because of how the hardware is designed, so generally if you can use the on-board slot we recommend that.

gloomyandy, 2019-01-29:
Did you try just refreshing the file list? If you go to the "print from SD" menu the first entry in the list of files is "Refresh" this should cause Marlin to re-read the directory contents and may make the file appear.

bcjams, 2019-01-30:
To clarify the issue, I have everything configured properly to read the SD card.  The problem is that Marlin does not refresh the files unless I reboot, or remove and reinsert the SD.   If I insert an SD, it will see the files on the SD card, and I am able to print with no issue.  The problem is how do I refresh the SD card without having to reboot or reinsert card. 

As far as a 'Refresh' menu select, I do not see that option on my screen under the "Print from SD" menu.

Bob-the-Kuhn, 2019-01-30:
I think the problem might be the use of a directory cache by Marlin 2.0.  

The SDSORT_CACHE_NAMES option is turned on automatically when using the SD card on the RepRap Full Graphics card with the re-ARM board.  This was part of a work around to keep the LCD from glitching when doing SD card directory accesses.

gloomyandy, 2019-01-30:
Hmm so it seems the "Refresh" option is only present if you do not have the SD_DETECT pin defined. I suppose un-defining that would be an option (though you would then have to either manually mount the SD card using "Init SD" or find some way to issue a G21 on boot (Does Marlin have a user defined boot sequence?). Or you could edit the menu code and force the refresh option. 

How robust are these WiFi SD cards? If Marlin is not seeing the directory change then is there a chance of it corrupting the file system if it writes to the SD card? I know that when we added the ability to share an SD card between Marlin and a PC (using USB), we opted to be cautious and not allow both to access the SD card at the same time (So we require Marlin to mount the SD card to gain access and when you do that we stop sharing the card via USB). Are we being to cautious in this case? I have a feeling the Smoothieware allows both systems to access the card.

Bob-the-Kuhn, 2019-01-30:
These WIFI SD cards use the full size format.  They can't fit into the Re-ARM's microSD slot.

bcjams, 2019-01-31:
I confirmed Bob Kuhns theory by modifying Conditionals.post.h 
1610  #define SDSORT_CACHE_NAMES false

I get the expected results by seeing the correct directory structure, with the only negative impact of a horizontal line midway through the display.
[image]

boelle, 2019-02-21:
@bcjams problem solved?

closed 2019-03-15 by p3p