MarlinFirmware/Marlin #13059: TMC2130 on dual Z (parallel on ramps) has overtemperature warning even when cool!

tobecwb, 2019-01-31:
### Description

I got a CR10-S5 with dual Z steppers (pancake stepper motor).
It's common that this printer loose sync on z motors when the stepper is turned off.
So, I put a belt between the two steppers drivers to sync, but this require more torque to raise or lower.

I replaced the motherboard with a Arduino Mega 2560 + Ramps and TMC2130 on all axis (including E0).
The default current of 800 for Z axis has a lots of missing steps, so, I increased the current until all was ok.

The current that I'm using is 1300mah (remember, two motors on parallel).
After raise the Z, I always got the following message:

> Z driver overtemperature warning!

And Marlin begin dropping the current.
But checking the temperature on TMC2130 driver and stepper motor, all appears to be ok!
less than 30C... the ambient temperature is 22C

### Steps to Reproduce

1. Wire two steppers motors on parallel
2. Use a single TMC2130 driver to control the two steppers motors
3. Send a g-code to run the two stepper motors

**Expected behavior:** 
No warning about temperature when driver and stepper motor was only a little above ambient temperature.

**Actual behavior:**
Marlin report that the driver got a overtemperature.

**Additional Information**

    Send: M122
    Recv: 		X	Y	Z	E0
    Recv: Enabled		false	false	false	false
    Recv: Set current	800	800	1150	800
    Recv: RMS current	795	795	1104	795
    Recv: MAX current	1121	1121	1557	1121
    Recv: Run current	25/31	25/31	19/31	25/31
    Recv: Hold current	12/31	12/31	9/31	12/31
    Recv: CS actual		12/31	12/31	9/31	12/31
    Recv: PWM scale		70	43	70	5
    Recv: vsense		1=.18	1=.18	0=.325	1=.18
    Recv: stealthChop	true	true	true	true
    Recv: msteps		16	16	16	16
    Recv: tstep		1048575	1048575	1048575	1048575
    Recv: pwm
    Recv: threshold		98	98	658	283
    Recv: [mm/s]		100.85	100.85	3.00	30.04
    Recv: OT prewarn	false	false	false	false
    Recv: OT prewarn has
    Recv: been triggered	false	false	true	false
    Recv: off time		5	5	5	5
    Recv: blank time	24	24	24	24
    Recv: hysteresis
    Recv: -end		2	2	2	2
    Recv: -start		3	3	3	3
    Recv: Stallguard thrs	0	0	0	0
    Recv: DRVSTATUS	X	Y	Z	E0
    Recv: stallguard				X
    Recv: sg_result		0	0	0	0
    Recv: fsactive
    Recv: stst		X	X	X	X
    Recv: olb
    Recv: ola
    Recv: s2gb
    Recv: s2ga
    Recv: otpw
    Recv: ot
    Recv: Driver registers:	X = 0x80:0C:00:00
    Recv: 	Y = 0x80:0C:00:00
    Recv: 	Z = 0x81:09:00:00
    Recv: 	E0 = 0x80:0C:00:00
    Recv: 
    Recv: 
    Recv: ok

tobecwb, 2019-02-01:
> You are talking about different cases.
> One is talking about 2 motors in parallel to 1 driver.
> The other about 2 drivers in parallel, each connected to 1 motor.

sorry about the confusion, but english it's not my primary language!

but like I said on **Steps to reproduce**:

> Use a single TMC2130 driver to control the two steppers motors

I was talking about two stepper motors on a single driver.
I really don't know how the driver works on this case, but  i see that even with 1000mah, the motors are very weak (I can stop with only two fingers on shaft!)... if I wire only one motor on the stepper driver, the motor are more strong.

Ketchu13, 2019-02-01:
Like teemuatlut, I think you should reduce your current for your dual Z steppers motors with only one driver.

With a CR-10 + ramps+mega2560 + TMC2130 + dual stepper motors (original Z stepper motors) in parallel on only one stepper driver, is fine with 900mA setted.

AnHardt, 2019-02-01:
Your concept is not that brilliant.
Pancake motors have noticeable less torque than normal ones. To make them equally strong you have to increase their current.
Connecting two motors, in parallel, to one driver doubles the needed current thru the driver. Because that is limited you don't get enough current in the motor.
Connecting the motors in series effectively doubles their common resistance - if not able to double effective voltage (PWM), cutting current to half.

The drivers temperature sensor is directly on the silicon of the drivers chip. It is able to react very fast. At least much faster than you can measure the increased temperature at the heat-sink. For that the heat has to creep from the silicone to the chips housing, true the board, into the heat-sink and has to distribute there in. I guess the speed difference is factor ~1000.