MarlinFirmware/Marlin #13060: Filament runout sensor-disable

hussainsail2002, 2019-02-01:
Description of the bug

I have been trying to use the M412 S0 command to disable the filament runout sensor. I added M412 S0 to the start Gcode. But it still triggers the filament run out pause sequence.

I am using marlin 2.0

 Steps to Reproduce

1. You need to use marlin 2.0 and have advanced pause and filament run out activated, connect your sensor to the board and test if the run out sensor works
2. To disable the runout sensor use M412 S0 at the start Gcode
3. Run the filament out side the sensor and check if the print continues without pausing

**Expected behavior:** 

I expected the Printer to have continued printing without activating advanced pause.

**Actual behavior:** 

So this is the Start G code I use:
M412 S0;
G28;
G29;
I am using Simplify 3d

Once the printer finishes heating up it pauses, I resume it. Then it homes and does a 9 point bed leveling and once its done with that it pauses again if I resume it again it will just pause again unless I have the run out switch turned on.

RudyBenner, 2019-02-01:
why is it pausing?

hussainsail2002, 2019-02-05:
> why is it pausing?

I am assuming its pausing because of the filament run out sensor being activated.

> Making changes around this code this week, so ill give it a once over. May behave properly with #12984 and/or #13039

Awesome !, would love to test it out once done !!.
Please do let me know.

InsanityAutomation, 2019-02-05:
Quite literally debugging this second lol

InsanityAutomation, 2019-02-18:
Can you verify this issue no longer exists? Id suggest a snapshot with PR #13193 included though as it fixes one last minor issue with pause / resume separate from this.

hussainsail2002, 2019-02-20:
@InsanityAutomation Hi !!

Thanks for the updating the firmware,

The issue still persists, but not completely.

I print from the SD card so I added M412 S0 to the start gcode as I did before.
Start gcode is as follows:

M412 S0;
G28 ; home all axes
G29 ;

The machine homes and does a nine point bed leveling, after completing the bed leveling it pauses because of the run-out sensor.The machine prompts me to insert filament and then resume. I resume it **without** inserting the filament through the runout sensor and then it works fine !!

I am not sure if I am doing anything wrong or maybe the Bltouch that I use is creating issues,
is there a way to remove the pausing completely ?

boelle, 2019-02-21:
@hussainsail2002 problem solved?

hussainsail2002, 2019-02-22:
@boelle 

The problem has not been completely solved yet.

Its definitely better than before, the problem is that it still pauses after completing the G29 sequence.
I don't think that should happen.

boelle, 2019-02-22:
using bugfix 2.0?

hussainsail2002, 2019-02-22:
> using bugfix 2.0?

Yes.

boelle, 2019-02-22:
had to ask, a lot think that bugfix 2.0 is only for 32 bit

and its here that all updates happen

InsanityAutomation, 2019-02-28:
I think the issue is that its entering the printing state before the sensor is disabled. Since the SD print changes states immediately unlike host printing that goes off of the print job timer getting called my a heat stabilize command. Being in the start gcode wasnt a scenario considered during design of the feature, it was more of a problem removal item than an every print use like that. Ill have to think a little about how to keep it from tripping immediately without potentially letting someone start a dry print....

hussainsail2002, 2019-03-02:
> I think the issue is that its entering the printing state before the sensor is disabled. 

Yes I think that you are correct. Because once the print starts I can  pull out the filament from sensor and disable it , and nothing happens to the print. If I disabled run out before the print starts then it pauses.

thinkyhead, 2019-03-03:
Filament runout is checked continually, and not just on E motion (for example), so it will get tripped anytime there is an active print job (sd card file started or print job timer running). If the filament sensor is not wanted, it should be turned off before starting the print. OFF may be the better default state. Then you can turn it ON in your first-layer G-code snippet when desired, and it won't get in the way otherwise.

boelle, 2019-03-13:
@hussainsail2002 still having an issue?

hussainsail2002, 2019-03-19:
Hi,

Yes turning the filament runout off before starting the print works and I am doing this via the LCD interface instead of using the M412 command.

Thanks @InsanityAutomation and @thinkyhead 

Closing the issue.

closed 2019-03-19 by hussainsail2002