MarlinFirmware/Marlin #13068: [1.1.9] TMC2208 Randomly disabled stealthchop and wont reenter...even with firmware reflashes?

freighttt, 2019-02-03:
i installed 2208's on my mks gen v1.4 about a month ago and yesterday it suddenly decided stealthchop was going to take a vacation. , while at the same time cooking the stepper motors.

i have not physically modified the drivers in any way. am i correct in understanding that they dont need jumpers/soldering on the mks gen 1.4? either way, they've run without issue until now.
code is not my strong suit, but i am pretty tech savvy - ive run tried to following.

Adjusting current via m906
enabling hybrid threshold 
reflashing the original firmware build as used when they were first installed
tmc_adv // void en_spreadCycle(bool) and bool stealth() with 1 and 0
sliced stl with super low speeds to see if it was a spreadcycle issue

m122 returns this:

		X	Y	Z	E0
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	1436	1436	1436	1436
Recv: MAX current	2025	2025	2025	2025
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual		0/31	0/31	0/31	0/31
Recv: PWM scale		0	0	0	0
Recv: vsense		0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false
Recv: msteps		256	256	256	256
Recv: tstep		0	0	0	0
Recv: pwm
Recv: threshold		1581	1581	10284	4302
Recv: [mm/s]		100.02	100.02	3.00	29.73
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		0	0	0	0
Recv: blank time	16	16	16	16
Recv: hysteresis
Recv: -end		-3	-3	-3	-3
Recv: -start		1	1	1	1
Recv: Stallguard thrs
Recv: DRVSTATUS	X	Y	Z	E0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:	X = 0x00:00:00:00
Recv: 	Y = 0x00:00:00:00
Recv: 	Z = 0x00:00:00:00
Recv: 	E0 = 0x00:00:00:00

about to give this tmc_adv setup a swing, but i dont have much hope. any ideas? am i stupid and missing the obvious "enable stealthchop" gcode command?

jvdbrg, 2019-02-03:
If I am correct, than the register values are not valid witch indicates that there is no uart connection between yout drivers and your board.

teemuatlut, 2019-02-03:
Good place to start http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

freighttt, 2019-02-05:
> Good place to start http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

ive run through all that - tried current and rolled back libraries, cables are all good etc. im going to try flashing marlin 2 and see what happens...

> If I am correct, than the register values are not valid witch indicates that there is no uart connection between yout drivers and your board.

i did see that my drivers are returning invalid values...but im lost as to how to fix it. like i mentioned, everything was fine for weeks, i hadnt changed or adjusted anything. its not clear whats changed, unless all 4 drivers did something crazy like fry at the same time?

boelle, 2019-03-06:
@freighttt still working on this?

freighttt, 2019-03-06:
@boelle 
I ended up wiring them up for uart(which unleashed a whole new issue, of course haha), and they appear to be communicating fine. Given marlin 2 is upon us, this can probably be closed. I suspect it was a fluke of some type, probably from the tmc side of things and not an issue rooted in marlin.

closed 2019-03-06 by freighttt

boelle, 2019-03-06:
marlin 2 is allready here :-D

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

freighttt, 2019-03-06:
Haha, you know what I mean! @boelle