MarlinFirmware/Marlin #13069: Resume print stuck

budy238, 2019-02-03:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

budy238, 2019-02-03:
[New Compressed (zipped) Folder.zip](https://github.com/MarlinFirmware/Marlin/files/2825307/New.Compressed.zipped.Folder.zip)

boelle, 2019-03-06:
@budy238 is this still a problem?

and describe the problem in details

try bugfix 2.0

boelle, 2019-03-13:
@thinkyhead think we can close, OP did not describe and has lost interest

closed 2019-05-17 by thinkyhead

budy238, 2019-05-17:
I apologize for not responding I appreciate the diligent service