MarlinFirmware/Marlin #13073: Q: LCD Resume print menu in standby

rmoravcik, 2019-02-03:
I just uploaded bugfix-2.0.x (https://github.com/MarlinFirmware/Marlin/commit/4d1093b3868a37ee0def13dbe2785f3ce98fcf6d) to my printer and I discovered that "Resume print" menu option is always visible even when printer is not doing any print job.

I've seen some LCD menu pause/resume fixes by @InsanityAutomation merged to bugfix-2.0.x recently.

So, my question is: Is this an regresion of those fixes or an intention to have it visible?

FanDjango, 2019-02-03:
I have seen the same and have also been wondering

InsanityAutomation, 2019-02-03:
It's intentional if you have host action commands enabled. the printer has no idea if you have pause the job from a remote host so that will send a resume command back to the host.

FanDjango, 2019-02-03:
I am not sure I have "Host Action Commands" enabled. I believe that I don't. What are the exact #defines in question?

FanDjango, 2019-02-03:
Okidoki - Gotcha. 

But I do not have any of them defined.

InsanityAutomation, 2019-02-03:
Just took a quick peak and it looks like the conditions around the menu got dropped. Ill add em back in as part of the PR mentioned above.

rmoravcik, 2019-02-04:
It's fixed with commit https://github.com/MarlinFirmware/Marlin/commit/0a7ea7063f82bc914e5cbfb55f0db400d0c3c936. Thank you.

closed 2019-02-04 by rmoravcik