MarlinFirmware/Marlin #13077: Tevo Tornado Gold BLTouch Setup Marlin 2.0

blackdogws, 2019-02-04:
I really need some help, please!!! I'm using Marlin 2.0 setup for stock drivers and a BLTouch. I have a Tornado Gold with no mods other than the BLTouch. When I auto home the printer goes to X193Y160Z10. Don't no why this is not the center of the bed. Plus the hot end pushes into the bed and the BLTouch errors out. I set the BLTouch to 8.3 when you can just fit a sheet of paper between the hot end and bed. I have set the Z probe offset to 2.5. I have no idea what to do next, I have tried several of the firmware files and end up with different issues. The firmware I have now seem to be the closest to working. Anybody have any ideas or a configure file that work for a BLTouch and a stock Tornado Gold.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2829814/Archive.zip)

brmcc, 2019-02-07:
This is my first go at making a Marlin firmware config and I'm also trying to setup a sensor on the new gold Tevo Tornado (MKS GEN L v1.0 board) and have had some of the same problems.  I don't have much experience but have learned a few things.

I had already made my own config file pretty much from scratch, but I tried yours and although there are a lot of differences, one thing remains. They both home OK with INVERT_Z_DIR set to false.  Well, as well as they can with the stepper motor running backwards. There are no errors though and it still manages to hit the endstop.  However, on ours, the Z and extruder need to be inverted to go in the right direction.

This is why your hotend is crashing into the bed.  Instead of raising it like it should, it's smashing it down.  If you invert the Z, it never homes on the Z axis though.  It never even tries.  It's like it's doing more than just changing the Z direction when INVERT_Z_DIR is enabled.  It's like something is changing on how the endstops are interpreted.

I wish I had a solution but I can at least confirm there is some problems.  I'm going to try just physically swap my Z motor wires around, hopefully today, and see what that does.

blackdogws, 2019-02-07:
I got all of my issues solved. The biggest was the right fan cooling shroud was hitting the bed and not allowing the hot end to be set correctly. Once I identified that and I set the absolute zero of the Z offset everything worked great.

crazy72, 2019-02-24:
Hi blackdogws, do you have a link to the firmware 2.0 for stock version and bltouch ?
Thx

blackdogws, 2019-02-26:
Look on the tornado user group on Facebook under files. I think it was on the second page. 

Shawn
402-350-0102
Please leave a message if I don't answer.

boelle, 2019-03-06:
@blackdogws still working on this?

boelle, 2019-03-13:
@thinkyhead and this one is dead too it seems

closed 2019-05-17 by thinkyhead