MarlinFirmware/Marlin #13078: Fade height 1.1.9 , Marlin crash during a print

creatron, 2019-02-05:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Roxy-3D, 2019-02-05:
You can't change the fade height (by very much) during a print, or you are going to cause problems.   The reason is each layer's height is gets less and less z-height correction as the nozzle gets closer to the fade height.

boelle, 2019-02-21:
@creatron problem solved?

creatron, 2019-02-22:
The problem still exist, it* IS* possible to change the fade-height during
a print, one just need to enter the baby steps menu, for me the z-offset
change is not suppose be be available during baby steps.

One can verify it just use the *auto home*, quickly go to *Tune* menu,
then *baby
step*, and 'whala' the fade-height is available.    The whole idea of bady
steps is to use it during print,

Kind Regards
Gerrie

-- 
Gerrie de Jager

(Quite people have the loudest minds.
Whether you think you can or you think you can't -- your'ed right.
Don't confuse having less with being less, having more with being more.
Who you are is far more important than what you have.)

Phone +27(0)11 024 3832
Phone +27(0)11 026 5631
Fax   086 552 2890
Cell  072 198 1678

*www.creatron.co.za <http://www.creatron.co.za>*

This e-mail, its contents and any file attachments transmitted with it are
intended solely for the addressee(s) and may contain confidential
proprietary information. Access by any other party without the express
written permission of the sender is unauthorised. If you have received this
e-mail in error you may not copy, distribute or use the contents,
attachments or information in any way. Please destroy it and contact the
sender.

boelle, 2019-02-22:
and you are using bugfix  2.0?

thinkyhead, 2019-02-22:
> it* IS* possible to change the fade-height during a print

We should disable that capability, as it definitely does cause trouble. I'll do that pretty soon.

boelle, 2019-03-13:
@thinkyhead maybe close this one too?

closed 2019-05-17 by thinkyhead