MarlinFirmware/Marlin #13079: clarification needed on probe offset

ruggb, 2019-02-05:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

reloxx13, 2019-02-05:
example:
if ur probe says "found the bed here, so here is 0"
now ur z offset says "okay, but ur nozzle is -2mm away from this point"
and ur printer accepts with "gj guys, ill go to probes 0 and go further down by -2 to have the nozzle at the bed, this is my print position"

boelle, 2019-03-06:
@ruggb still working on this?

ruggb, 2019-03-06:
I think I have it resolved;
a NOZZLE activated probe needs a positive Z offset. It triggers AFTER hitting the bed.
A separate probe (BL Touch) needs a negative Z offset, it trigger BEFORE the nozzle gets to the bed.
Maybe you could say that and it would be VERY clear.

closed 2019-03-06 by ruggb

boelle, 2019-03-06:
it does say so in the config

#define X_PROBE_OFFSET_FROM_EXTRUDER 23  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 9  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.215   // Z offset: -below +above  [the nozzle]

Z offset: -below +above  [the nozzle]

so it's VERY clear to me, and i'm not a super smart person :-D

ruggb, 2019-03-06:
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.215 // Z offset: -below +above [the nozzle]
"-" indicated bed gets closer to nozzle
"below" indicates bed is farther from nozzle.
maybe not so clear, huh?

boelle, 2019-03-06:
its clear to me that - means it triggers below nozzle
+ means above the nozzle

ruggb, 2019-03-06:
THAT is the confusion - you are talking about Zoffset then you explain it by where the nozzle triggers.
The Zoffset is opposite of where the nozzle triggers. saying - and below to me are mutually exclusive. It is either one or the other. But since you wrote it it is perfectly clear to you.

boelle, 2019-03-06:
no i explain it by where the probe triggers

and it has been so since at least 2013 where i got my first printer, so its nothing new

boelle, 2019-03-06:
but yes Z offset is oposite, but i have not seen any that use Z offset

boelle, 2019-03-06:
and i did not write anything.... but i guess you mean you as collective again