MarlinFirmware/Marlin #13081: Probe Offsets

atomicdad, 2019-02-05:
### Description

I'm curious if the probe offsets as defined in Configuration.h are being applied correctly in the UBL code. 

Specifically lines 1244 & 1245 of ubl_G29.cpp (4fa1c52) in Bugfix-2.0.x and is also present in just about any other version.  The issue (if it exists) has been around for some time. I haven't searched to see if this issue exists anywhere else.

Function:
`mesh_index_pair unified_bed_leveling::find_closest_mesh_point_of_type(const MeshPointType type, const float &rx, const float &ry, const bool probe_as_reference, uint16_t bits[16])`

Lines in question:
```
// Get our reference position. Either the nozzle or probe location.
const float px = rx - (probe_as_reference == USE_PROBE_AS_REFERENCE ? X_PROBE_OFFSET_FROM_EXTRUDER : 0),
            py = ry - (probe_as_reference == USE_PROBE_AS_REFERENCE ? Y_PROBE_OFFSET_FROM_EXTRUDER : 0);
```
			
Shouldn't the probe offsets be added to the nozzle position? I noticed that after physically moving my probe to a position in the X+, Y- quadrant it would attempt to probe off the front end of the bed and continually fail. A fellow user of the same printer noticed the same issue when trying to use UBL with a probe in the same quadrant. He eventually gave up and reprinted a probe holder for the X-, Y- quadrant, and it worked. I changed the lines above to use the sum of the values and it appeared to work as intended.

Can anyone with any insight into the UBL code take a look at this? Doing some digging in the Configuration.h commit history it looks like probe offsets used to be calculated as (hotend - probe) and then changed to (probe - hotend) somewhere at the end of 2014 or early 2015.

Roxy-3D, 2019-02-06:
I think what you are suggesting is the code should be:
```
// Get our reference position. Either the nozzle or probe location.
const float px = rx + (probe_as_reference == USE_PROBE_AS_REFERENCE ? X_PROBE_OFFSET_FROM_EXTRUDER : 0),
            py = ry + (probe_as_reference == USE_PROBE_AS_REFERENCE ? Y_PROBE_OFFSET_FROM_EXTRUDER : 0);
```
Is that correct?   If so, can you give that a quick test, and I'll zap the code to conform to your findings.

atomicdad, 2019-02-06:
Yes, that is correct.

Roxy-3D, 2019-02-06:
Have you tested the change to verify it actually fixes the problem you are seeing?

Roxy-3D, 2019-02-06:
OK...   Let's leave this Issue open for a little while.   Let's see if we can get a few other people to verify the behavior is better now.   (I zapped both v1.1 Bugfix and v2.0)

gloomyandy, 2019-02-06:
As I said on discord, I do think the original code is wrong. But I'm struggling to see how it can cause a probe operation off the bed (which I think was the original problem). As far as I can tell the bug may sometimes cause the wrong grid point to be selected for probing (it may not be the nearest), but that point is then checked to ensure it can be reached by the probe (at least that is my reading of the code) and so will be skipped if it is not possible to probe it. It would be really good to have an example test case for this.

atomicdad, 2019-02-06:
gloomyandy, I had the same line of thought as well. I don't get it. The offsets in that function don't seem to be used for anything other than the comparison. I will revert the code back and try again when I can. I'll report back with the results. I'd like to help track down the cause.

I did a full project search for PROBE_OFFSET_FROM_EXTRUDER_X and it is used in quite a few places. It might be worth looking into to see if any other usage is in error as well. I will attempt to do so.

As indicated in my original post, I think this issue might come from the change in the way the probe offset was described several years ago. It appears to originally have been similar to the way Home Offsets are now; as coordinates to get back to the endstops. In this case the coordinates to get back to the nozzle. And then it changed to the coordinates from the nozzle to the probe. I'm wondering if when that was changed the functions using it were never updated in some cases.

gloomyandy, 2019-02-06:
@atomicdad do you have the details of the setup that you have had problems with? Is your co-ordinate system larger than than the bed? I use UBL a lot and have not seen any issues with this, but I do agree that the code looks odd, so I suspect your change is correct. I just can't figure out how it can cause an off bed probe. As you say it may be that the same problem exists somewhere else and is causing the problem. It would be really useful to have a test case if you can come up with one!

gloomyandy, 2019-02-06:
@Roxy-3D I've just been running a few simple tests with the offset changed from + to - and it doesn't seem much different to me! Certainly no worse than before (I've never totally understood the pattern used for a G29 P1!). As I said above I think that it makes more sense with things the new way (as the probe offsets are added earlier on in the call stack so adding them again can't really be correct).

atomicdad, 2019-02-07:
Ok, had time this evening to run some tests. I was able to reproduce the off bed probing with the firmware copy I had been using. And once again swapping the subtraction with addition made it start probing right next to home on the front left of my bed. 

So I went ahead and loaded Roxy's commit from last night and was unable to reproduce the bug regardless of sign (+ or -). It did start probing in the center of the bed, however. Never seen that before. 

I need to look through the UBL code to figure out how the mesh should be created. It seems like my probe was all over the place at random. I'm not sure it matters in the end as the mesh output looked correct, but it was interesting.

I'll probably load up 1.1.x tomorrow or this weekend and see if I can reproduce it there. Otherwise whatever was causing it appears to have been fixed elsewhere in 2.0 and was somehow helped along by correcting the function I first posted.

atomicdad, 2019-02-07:
Hold it! Found the culprit. 

Try the pre fix UBL with that.
```
#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS 0
#define MANUAL_Z_HOME_POS 0
```

Roxy-3D, 2019-02-07:
> I've just been running a few simple tests with the offset changed from + to - and it doesn't seem much different to me! Certainly no worse than before (I've never totally understood the pattern used for a G29 P1!).

@gloomyandy  Thank you for the help!   The pattern for G29 P1 (and for that matter G26) starts where ever the nozzle is and expands outward.   This was done for two reasons.   First, because it was expected it would take a long time to fully generate an accurate mesh.   Allowing the user to generate a mesh just for the portion of the bed that they needed was thought to be useful.   This concern turned out to be less important than initially guessed.   

The other reason this was done was to protect the user's machine.   Where ever the nozzle starts probing, it is assumed adjacent locations (mesh points) will vary, but not by a huge amount.    So if the first probed point is OK, the surrounding points are probably not going to be different by a lot.     By expanding outward from known 'safe' points on the mesh, it gives the user time to abort the probing operation if the nozzle is getting too close to the bed.

Roxy-3D, 2019-02-07:
I don't understand how the Home Position factors into this?????

atomicdad, 2019-02-07:
I'm not sure, but if I comment those out and let Marlin choose home at the center of the bed it works fine.

Roxy-3D, 2019-02-07:
That kind of suggests there are coordinates being generated for the G29 P1 that are off the grid in some cases....

atomicdad, 2019-02-07:
It appears to be unrelated to MIN_PROBE_EDGE. When set to 0, the result is the same.

Roxy-3D, 2019-02-07:
@atomicdad   What is the exact behavior when you see the failure?  What does the machine do?  Does the G29 P1 terminate or keep going?   Are any error messages output to PronterFace (or Repetier) ?

atomicdad, 2019-02-07:
It just keeps attempting to probe in the x and y endstop positions (both are lit). 
Message is "Error:Probing Failed"

atomicdad, 2019-02-07:
And just to be clear this is using the "Step-By-Step UBL -> 1 Build Cold Mesh" option in the interface.

Roxy-3D, 2019-02-07:
OK.  Understood.      I may never have seen this problem because I always home the machine at the center of the bed.    That makes sense for a number of reasons.   But still....   it should work if the machine is homed at (0,0)    

Can you attach a remote console such as PronterFace or Repetier?   The reason I'm asking is we could add a few debug lines to print out what is happening at each step.   And after a few iterations, we should know what is wrong and what to do to fix it.

Also...   if we do that, it would probably make sense to turn on the #define DEBUG_LEVELING_FEATURE capability because we would see all the steps the probing is doing.   In fact, we should probably start with that and use that information to add a few lines of debug information if we need it.    

Is that an option?

atomicdad, 2019-02-07:
Would be happy to. The output is fairly large. Should I use pastebin or upload a text file or something?

atomicdad, 2019-02-07:
21:09:50.670 : echo:enqueueing "G28"
21:09:50.683 : echo:enqueueing "G29 P1"
21:09:50.684 : echo:G28
21:09:50.684 : >>> G28
21:09:50.686 : Machine Type: Cartesian
21:09:50.686 : Probe: FIX_MOUNTED_PROBE
21:09:50.688 : Probe Offset X:40 Y:-40 Z:-2.00 (Right-Front & Below Nozzle)
21:09:50.689 : Auto Bed Leveling: UBL
21:09:50.689 : (disabled)
21:09:50.692 : current_position=(-10.00, 70.00, 5.00) : bracket_probe_move
21:09:50.694 : > endstops.enable(true)
21:09:50.698 : current_position=(0.00, 0.00, 5.00) : sync_plan_position
21:09:50.702 : >>> do_blocking_move_to(-502.50, -502.50, 5.00)
21:09:50.703 : <<< do_blocking_move_to
21:09:50.809 : >>> homeaxis(X)
21:09:50.809 : Home 1 Fast:
21:09:50.812 : >>> do_homing_move(X, -502.50, [50.00])
21:09:50.921 : <<< do_homing_move(X)
21:09:50.922 : Move Away:
21:09:50.923 : >>> do_homing_move(X, 5.00, [50.00])
21:09:51.168 : <<< do_homing_move(X)
21:09:51.169 : Home 2 Slow:
21:09:51.171 : >>> do_homing_move(X, -10.00, 25.00)
21:09:51.492 : <<< do_homing_move(X)
21:09:51.493 : >>> set_axis_is_at_home(X)
21:09:51.493 : For X axis:
21:09:51.494 : home_offset = 0.00
21:09:51.495 : position_shift = 0.00
21:09:51.495 : > home_offset[X] = 0.00
21:09:51.498 : current_position=(0.00, 0.00, 5.00) :
21:09:51.499 : <<< set_axis_is_at_home(X)
21:09:51.501 : current_position=(0.00, 0.00, 5.00) : sync_plan_position
21:09:51.504 : current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
21:09:51.504 : <<< homeaxis(X)
21:09:51.506 : >>> homeaxis(Y)
21:09:51.506 : Home 1 Fast:
21:09:51.508 : >>> do_homing_move(Y, -502.50, [50.00])
21:09:51.617 : <<< do_homing_move(Y)
21:09:51.618 : Move Away:
21:09:51.619 : >>> do_homing_move(Y, 5.00, [50.00])
21:09:51.864 : <<< do_homing_move(Y)
21:09:51.865 : Home 2 Slow:
21:09:51.867 : >>> do_homing_move(Y, -10.00, 25.00)
21:09:52.189 : <<< do_homing_move(Y)
21:09:52.189 : >>> set_axis_is_at_home(Y)
21:09:52.189 : For Y axis:
21:09:52.191 : home_offset = 0.00
21:09:52.192 : position_shift = 0.00
21:09:52.192 : > home_offset[Y] = 0.00
21:09:52.195 : current_position=(0.00, 0.00, 5.00) :
21:09:52.196 : <<< set_axis_is_at_home(Y)
21:09:52.197 : current_position=(0.00, 0.00, 5.00) : sync_plan_position
21:09:52.201 : current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
21:09:52.201 : <<< homeaxis(Y)
21:09:52.203 : >>> homeaxis(Z)
21:09:52.203 : Home 1 Fast:
21:09:52.205 : >>> do_homing_move(Z, -598.50, [4.00])
21:09:53.605 : <<< do_homing_move(Z)
21:09:53.606 : Move Away:
21:09:53.607 : >>> do_homing_move(Z, 2.00, [4.00])
21:09:54.270 : <<< do_homing_move(Z)
21:09:54.271 : Home 2 Slow:
21:09:54.273 : >>> do_homing_move(Z, -4.00, 1.00)
21:09:56.377 : <<< do_homing_move(Z)
21:09:56.378 : >>> set_axis_is_at_home(Z)
21:09:56.378 : For Z axis:
21:09:56.380 : home_offset = 0.00
21:09:56.381 : position_shift = 0.00
21:09:56.382 : *** Z HOMED TO ENDSTOP (Z_MIN_PROBE_ENDSTOP) ***
21:09:56.384 : > home_offset[Z] = 0.00
21:09:56.385 : current_position=(0.00, 0.00, 0.00) :
21:09:56.386 : <<< set_axis_is_at_home(Z)
21:09:56.389 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:09:56.391 : current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
21:09:56.393 : <<< homeaxis(Z)
21:09:56.394 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:09:56.396 : current_position=(0.00, 0.00, 0.00) : bracket_probe_move
21:09:56.400 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
21:09:56.400 : <<< G28
21:09:56.400 : echo:G29 P1
21:09:56.411 : Mesh invalidated. Probing mesh.
21:09:56.413 : current_position=(0.00, 0.00, 0.00) : set_probe_deployed
21:09:56.415 : deploy: 1
21:09:56.415 : do_probe_raise(20.00)
21:09:56.417 : >>> do_blocking_move_to(0.00, 0.00, 22.00)
21:09:56.420 : <<< do_blocking_move_to
21:10:02.090 : >>> do_blocking_move_to(0.00, 0.00, 22.00)
21:10:02.091 : <<< do_blocking_move_to
21:10:02.103 : Mesh only partially populated.
21:10:02.106 : current_position=(0.00, 0.00, 22.00) : set_probe_deployed
21:10:02.107 : deploy: 0
21:10:02.108 : >>> do_blocking_move_to(0.00, 0.00, 22.00)
21:10:02.110 : <<< do_blocking_move_to
21:10:02.675 : X:0.00 Y:0.00 Z:22.00 E:0.00 Count X:0 Y:0 Z:8800

atomicdad, 2019-02-07:
21:15:27.445 : <<< probe_pt
21:15:27.471 : >>> probe_pt(30.00, 30.00, raise, 0, probe_relative)
21:15:27.474 : current_position=(-10.00, 70.00, 5.00) :
21:15:27.475 : >>> do_blocking_move_to(-10.00, 70.00, 5.00)
21:15:27.477 : <<< do_blocking_move_to
21:15:27.479 : current_position=(-10.00, 70.00, 5.00) : set_probe_deployed
21:15:27.479 : deploy: 1
21:15:27.480 : do_probe_raise(20.00)
21:15:27.483 : >>> do_blocking_move_to(-10.00, 70.00, 22.00)
21:15:27.485 : <<< do_blocking_move_to
21:15:31.902 : >>> do_blocking_move_to(-10.00, 70.00, 22.00)
21:15:31.903 : <<< do_blocking_move_to
21:15:31.907 : current_position=(-10.00, 70.00, 22.00) : >>> run_z_probe
21:15:31.909 : current_position=(-10.00, 70.00, 22.00) : >>> do_probe_move
21:15:31.912 : >>> do_blocking_move_to(-10.00, 70.00, 0.00)
21:15:31.915 : <<< do_blocking_move_to
21:15:43.055 : current_position=(-10.00, 70.00, 0.00) : sync_plan_position
21:15:43.057 : current_position=(-10.00, 70.00, 0.00) : <<< do_probe_move
21:15:43.059 : SLOW Probe fail!
21:15:43.061 : current_position=(-10.00, 70.00, 0.00) : <<< run_z_probe
21:15:43.063 : >>> do_blocking_move_to(-10.00, 70.00, 5.00)
21:15:43.066 : <<< do_blocking_move_to
21:15:44.477 : current_position=(-10.00, 70.00, 5.00) : set_probe_deployed
21:15:44.478 : deploy: 0
21:15:44.480 : >>> do_blocking_move_to(-10.00, 70.00, 5.00)
21:15:44.481 : <<< do_blocking_move_to
21:15:44.483 : Error:Probing failed
21:15:44.483 : <<< probe_pt
21:15:44.510 : >>> probe_pt(30.00, 30.00, raise, 0, probe_relative)
21:15:44.513 : current_position=(-10.00, 70.00, 5.00) :
21:15:44.514 : >>> do_blocking_move_to(-10.00, 70.00, 5.00)
21:15:44.516 : <<< do_blocking_move_to
21:15:44.518 : current_position=(-10.00, 70.00, 5.00) : set_probe_deployed
21:15:44.518 : deploy: 1
21:15:44.520 : do_probe_raise(20.00)
21:15:44.522 : >>> do_blocking_move_to(-10.00, 70.00, 22.00)
21:15:44.524 : <<< do_blocking_move_to
21:15:48.942 : >>> do_blocking_move_to(-10.00, 70.00, 22.00)
21:15:48.943 : <<< do_blocking_move_to
21:15:48.947 : current_position=(-10.00, 70.00, 22.00) : >>> run_z_probe
21:15:48.949 : current_position=(-10.00, 70.00, 22.00) : >>> do_probe_move
21:15:48.952 : >>> do_blocking_move_to(-10.00, 70.00, 0.00)
21:15:48.954 : <<< do_blocking_move_to
21:16:00.095 : current_position=(-10.00, 70.00, 0.00) : sync_plan_position
21:16:00.097 : current_position=(-10.00, 70.00, 0.00) : <<< do_probe_move
21:16:00.099 : SLOW Probe fail!
21:16:00.101 : current_position=(-10.00, 70.00, 0.00) : <<< run_z_probe
21:16:00.103 : >>> do_blocking_move_to(-10.00, 70.00, 5.00)
21:16:00.106 : <<< do_blocking_move_to
21:16:01.518 : current_position=(-10.00, 70.00, 5.00) : set_probe_deployed
21:16:01.519 : deploy: 0
21:16:01.520 : >>> do_blocking_move_to(-10.00, 70.00, 5.00)
21:16:01.522 : <<< do_blocking_move_to
21:16:01.524 : Error:Probing failed

Roxy-3D, 2019-02-07:
> Should I use pastebin or upload a text file or something?

No.  This is fine...  And we can edit out stuff later that we don't need if this thread drags on for any length of time...

But it looks like it is trying to probe at (-10, 70) which is off the bed.    We must have a boundary or range check missing or not catching the case where the home position is (0,0)

gloomyandy, 2019-02-07:
Those positions look as if they should be OK and that it should be probing at 30,30 which is on the bed (though the nozzle will be off the bed at the X endstop). But it all looks OK. Any chance of a video of what happens?

atomicdad, 2019-02-07:
It doesn't move from the x and y endstop positions. Z goes up and it starts probing. Unfortunately, I'm out of time for the evening, but I can grab you the G29 W real quick and try to get a video tomorrow. Sorry!

atomicdad, 2019-02-07:
21:32:58.296 : N18 G29 W*92
21:32:58.299 : echo:N18 G29 W*92
Nothing?

Roxy-3D, 2019-02-07:
You may have the G29 W turned off....

atomicdad, 2019-02-07:
How do I enable it?

Roxy-3D, 2019-02-07:
I'm looking for the exact option name right now...

atomicdad, 2019-02-07:
```
21:41:15.812 : Unified Bed Leveling System v1.01 inactive.
21:41:15.862 : No Mesh Loaded.
21:41:15.914 : planner.z_fade_height : 0.0000
21:41:15.917 : # of samples: 100
21:41:15.919 : Mean Mesh Height: 0.000000
21:41:15.920 : Standard Deviation: 0.000000
21:41:15.921 : zprobe_zoffset: -2.0000000
21:41:15.924 : MESH_MIN_X  (MAX((((325) / 2) - (325) / 2) + 30, -10))=30
21:41:15.976 : MESH_MIN_Y  (MAX((((325) / 2) - (325) / 2) + 30, -10))=30
21:41:16.029 : MESH_MAX_X  (MIN((((325) / 2) + (325) / 2) - (30), 325))=294
21:41:16.082 : MESH_MAX_Y  (MIN((((325) / 2) + (325) / 2) - (30), 325))=294
21:41:16.133 : GRID_MAX_POINTS_X  10
21:41:16.184 : GRID_MAX_POINTS_Y  10
21:41:16.234 : MESH_X_DIST  29.33
21:41:16.236 : MESH_Y_DIST  29.33
21:41:16.543 : X-Axis Mesh Points at: 30.000  59.333  88.667  118.000  147.333  176.667  206.000  235.333  264.667  294.000
21:41:16.799 : Y-Axis Mesh Points at: 30.000  59.333  88.667  118.000  147.333  176.667  206.000  235.333  264.667  294.000
21:41:16.849 : ubl_state_at_invocation :0
21:41:16.851 : ubl_state_recursion_chk :0
21:41:16.901 : Meshes go from 0x02C8 to 0x0F7F
21:41:16.953 : sizeof(ubl) :  1
21:41:16.954 : z_value[][] size: 400
21:41:16.980 : EEPROM free for UBL: 0x0CB7
21:41:17.032 : EEPROM can hold 8 meshes.
21:41:17.058 : Unified Bed Leveling sanity checks passed.
```

Roxy-3D, 2019-02-07:
Hmmmm.....   UBL thinks the corner of your mesh is at (30,30)   
21:41:15.924 : MESH_MIN_X  (MAX((((325) / 2) - (325) / 2) + 30, -10))=30
21:41:15.976 : MESH_MIN_Y  (MAX((((325) / 2) - (325) / 2) + 30, -10))=30

atomicdad, 2019-02-07:
I do have MIN_PROBE_EDGE set to 30. That's it no?

Roxy-3D, 2019-02-07:
And your Configuration.h says:
```
#define X_BED_SIZE 325
#define Y_BED_SIZE 325
#define X_MIN_POS -10
#define Y_MIN_POS -10
```
So it doesn't seem to me it should think your corner is at (30,30).    Do you have an inset specified?
But that 30mm for a minimum probe edge might be factoring into this.   If it is...  we should have some Sanity Checks to detect the conflict.

atomicdad, 2019-02-07:
I just set MESH_INSET to 0.

```
21:48:57.732 : Unified Bed Leveling System v1.01 inactive.
21:48:57.783 : No Mesh Loaded.
21:48:57.834 : planner.z_fade_height : 0.0000
21:48:57.837 : # of samples: 100
21:48:57.839 : Mean Mesh Height: 0.000000
21:48:57.840 : Standard Deviation: 0.000000
21:48:57.841 : zprobe_zoffset: -2.0000000
21:48:57.844 : MESH_MIN_X  (MAX((((325) / 2) - (325) / 2) + 0, -10))=0
21:48:57.896 : MESH_MIN_Y  (MAX((((325) / 2) - (325) / 2) + 0, -10))=0
21:48:57.949 : MESH_MAX_X  (MIN((((325) / 2) + (325) / 2) - (0), 325))=324
21:48:58.002 : MESH_MAX_Y  (MIN((((325) / 2) + (325) / 2) - (0), 325))=324
21:48:58.053 : GRID_MAX_POINTS_X  10
21:48:58.104 : GRID_MAX_POINTS_Y  10
21:48:58.154 : MESH_X_DIST  36.00
21:48:58.156 : MESH_Y_DIST  36.00
21:48:58.462 : X-Axis Mesh Points at: 0.000  36.000  72.000  108.000  144.000  180.000  216.000  252.000  288.000  324.000
21:48:58.717 : Y-Axis Mesh Points at: 0.000  36.000  72.000  108.000  144.000  180.000  216.000  252.000  288.000  324.000
21:48:58.767 : ubl_state_at_invocation :0
21:48:58.770 : ubl_state_recursion_chk :0
21:48:58.820 : Meshes go from 0x02C8 to 0x0F7F
21:48:58.871 : sizeof(ubl) :  1
21:48:58.873 : z_value[][] size: 400
21:48:58.899 : EEPROM free for UBL: 0x0CB7
21:48:58.950 : EEPROM can hold 8 meshes.
21:48:58.977 : Unified Bed Leveling sanity checks passed.

```

Roxy-3D, 2019-02-07:
OK?   And what does the M111 S255 show with regard to a G29 P1 ?

atomicdad, 2019-02-07:
```
21:54:18.563 : echo:Home XYZ first
21:54:18.563 : >>> G28
21:54:18.565 : Machine Type: Cartesian
21:54:18.566 : Probe: FIX_MOUNTED_PROBE
21:54:18.569 : Probe Offset X:40 Y:-40 Z:-2.00 (Right-Front & Below Nozzle)
21:54:18.570 : Auto Bed Leveling: UBL
21:54:18.570 : (disabled)
21:54:18.572 : current_position=(0.00, 0.00, 0.00) : bracket_probe_move
21:54:18.574 : > endstops.enable(true)
21:54:18.576 : Raise Z (before homing) to 5.00
21:54:18.579 : >>> do_blocking_move_to(0.00, 0.00, 5.00)
21:54:18.581 : <<< do_blocking_move_to
21:54:19.994 : current_position=(0.00, 0.00, 5.00) : sync_plan_position
21:54:19.997 : >>> do_blocking_move_to(-502.50, -502.50, 5.00)
21:54:20.000 : <<< do_blocking_move_to
21:54:20.106 : >>> homeaxis(X)
21:54:20.106 : Home 1 Fast:
21:54:20.109 : >>> do_homing_move(X, -502.50, [50.00])
21:54:20.229 : <<< do_homing_move(X)
21:54:20.230 : Move Away:
21:54:20.232 : >>> do_homing_move(X, 5.00, [50.00])
21:54:20.476 : <<< do_homing_move(X)
21:54:20.478 : Home 2 Slow:
21:54:20.479 : >>> do_homing_move(X, -10.00, 25.00)
21:54:20.801 : <<< do_homing_move(X)
21:54:20.802 : >>> set_axis_is_at_home(X)
21:54:20.802 : For X axis:
21:54:20.804 : home_offset = 0.00
21:54:20.805 : position_shift = 0.00
21:54:20.805 : > home_offset[X] = 0.00
21:54:20.807 : current_position=(0.00, 0.00, 5.00) :
21:54:20.809 : <<< set_axis_is_at_home(X)
21:54:20.810 : current_position=(0.00, 0.00, 5.00) : sync_plan_position
21:54:20.814 : current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
21:54:20.814 : <<< homeaxis(X)
21:54:20.815 : >>> homeaxis(Y)
21:54:20.815 : Home 1 Fast:
21:54:20.818 : >>> do_homing_move(Y, -502.50, [50.00])
21:54:20.927 : <<< do_homing_move(Y)
21:54:20.927 : Move Away:
21:54:20.929 : >>> do_homing_move(Y, 5.00, [50.00])
21:54:21.174 : <<< do_homing_move(Y)
21:54:21.175 : Home 2 Slow:
21:54:21.177 : >>> do_homing_move(Y, -10.00, 25.00)
21:54:21.498 : <<< do_homing_move(Y)
21:54:21.500 : >>> set_axis_is_at_home(Y)
21:54:21.500 : For Y axis:
21:54:21.501 : home_offset = 0.00
21:54:21.502 : position_shift = 0.00
21:54:21.502 : > home_offset[Y] = 0.00
21:54:21.505 : current_position=(0.00, 0.00, 5.00) :
21:54:21.506 : <<< set_axis_is_at_home(Y)
21:54:21.507 : current_position=(0.00, 0.00, 5.00) : sync_plan_position
21:54:21.511 : current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
21:54:21.511 : <<< homeaxis(Y)
21:54:21.513 : >>> homeaxis(Z)
21:54:21.513 : Home 1 Fast:
21:54:21.515 : >>> do_homing_move(Z, -598.50, [4.00])
21:54:24.164 : <<< do_homing_move(Z)
21:54:24.165 : Move Away:
21:54:24.167 : >>> do_homing_move(Z, 2.00, [4.00])
21:54:24.829 : <<< do_homing_move(Z)
21:54:24.831 : Home 2 Slow:
21:54:24.832 : >>> do_homing_move(Z, -4.00, 1.00)
21:54:26.934 : <<< do_homing_move(Z)
21:54:26.935 : >>> set_axis_is_at_home(Z)
21:54:26.935 : For Z axis:
21:54:26.937 : home_offset = 0.00
21:54:26.938 : position_shift = 0.00
21:54:26.939 : *** Z HOMED TO ENDSTOP (Z_MIN_PROBE_ENDSTOP) ***
21:54:26.941 : > home_offset[Z] = 0.00
21:54:26.942 : current_position=(0.00, 0.00, 0.00) :
21:54:26.943 : <<< set_axis_is_at_home(Z)
21:54:26.946 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:54:26.948 : current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
21:54:26.949 : <<< homeaxis(Z)
21:54:26.951 : current_position=(0.00, 0.00, 0.00) : sync_plan_position
21:54:26.953 : current_position=(0.00, 0.00, 0.00) : bracket_probe_move
21:54:26.956 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
21:54:26.956 : <<< G28
21:54:26.957 : Default storage slot 0 selected.
21:54:26.958 : Mesh invalidated. Probing mesh.
21:54:26.961 : current_position=(0.00, 0.00, 0.00) : set_probe_deployed
21:54:26.961 : deploy: 1
21:54:26.962 : do_probe_raise(20.00)
21:54:26.965 : >>> do_blocking_move_to(0.00, 0.00, 22.00)
21:54:26.966 : <<< do_blocking_move_to
21:54:32.636 : >>> do_blocking_move_to(0.00, 0.00, 22.00)
21:54:32.638 : <<< do_blocking_move_to
21:54:32.662 : >>> probe_pt(36.00, 0.00, raise, 0, probe_relative)
21:54:32.663 : current_position=(0.00, 0.00, 22.00) :
21:54:32.666 : >>> do_blocking_move_to(-4.00, 40.00, 22.00)
21:54:32.669 : <<< do_blocking_move_to
21:54:32.777 : current_position=(-4.00, 40.00, 22.00) : set_probe_deployed
21:54:32.778 : deploy: 1
21:54:32.781 : current_position=(-4.00, 40.00, 22.00) : >>> run_z_probe
21:54:32.782 : current_position=(-4.00, 40.00, 22.00) : >>> do_probe_move
21:54:32.785 : >>> do_blocking_move_to(-4.00, 40.00, 0.00)
21:54:32.787 : <<< do_blocking_move_to
21:54:43.926 : current_position=(-4.00, 40.00, 0.00) : sync_plan_position
21:54:43.929 : current_position=(-4.00, 40.00, 0.00) : <<< do_probe_move
21:54:43.930 : SLOW Probe fail!
21:54:43.932 : current_position=(-4.00, 40.00, 0.00) : <<< run_z_probe
21:54:43.935 : >>> do_blocking_move_to(-4.00, 40.00, 5.00)
21:54:43.937 : <<< do_blocking_move_to
21:54:45.349 : current_position=(-4.00, 40.00, 5.00) : set_probe_deployed
21:54:45.351 : deploy: 0
21:54:45.352 : >>> do_blocking_move_to(-4.00, 40.00, 5.00)
21:54:45.353 : <<< do_blocking_move_to
21:54:45.353 : Error:Probing failed
```

Roxy-3D, 2019-02-07:
Can we set your Home Position to (200,200) just to get you going?      I think I understand the problem well enough that when I get home (and have a printer available) I can find and fix the problem.    But I don't want to leave you hanging until then.

Roxy-3D, 2019-02-07:
OK!  Great!    THANKS FOR THE HELP!

Let's leave this thread open until we get successful resolution.

atomicdad, 2019-02-07:
Sounds great. Thank YOU! :)

atomicdad, 2019-02-09:
Here are some of the first probes plotted graphically using both a '-' and '+' in the original function we changed at the beginning of this Issue.
[image]

gloomyandy, 2019-02-09:
Yep that pretty much follows what I was seeing (though I use Z_SAFE_HOMING which means my probing starts in the centre of the bed).

filipgoc, 2019-03-12:
I have an issue that seems to be related: it's kind of the opposite. 

My probe is to the right off the nozzle X_PROBE_OFFSET_FROM_EXTRUDER 22
So if I want to probe as much left on the bed as possible, (and I can, I have the space to go off-bed) MESH_MIN_X for me should be to be about -10. 

With the old code (two minuses), UBL did not allow me any value for MESH_MIN_X lower than + X_PROBE_OFFSET_FROM_EXTRUDER. Exactly. Maybe it's a coincidence, but it seems strange.
It simply does not probe the first column if I try to go lower than that - even though the probe is clearly above the bed area. 

With the updated code (two pluses), UBL does allow to go a bit more left (not exactly the offset anymore), but not all they way still. Now I don't see any pattern. All insets are set to 0 including the probe edge... 

I'm happy to provide more testing data/ logs if this seems useful.

boelle, 2019-06-22:
missing label

closed 2019-07-11 by thinkyhead