MarlinFirmware/Marlin #13083: TMC2130 Skipping Steps

dcwalmsley, 2019-02-05:
Hi Guys, I'm back with a problem that I cannot figure out how to correct.

Background:
HyperCube (Core XY) 
NEMA17 motors
Einsy Rambo (24v)
Prusa Direct-Drive Extruder (E3D nozzle)
Prusa MMU2

Currently have Marlin bugfix 2.0 from a few days ago flashed to my printer.  I'll post files at end of this post.
My current issue is that my Y-Axis has been shifting at a random heights on multiple prints of an Elephant.  Not seeing this with X-Axis nor Z-Axis.

I'm also using TMCStepper (v0.3.1) by teemuatlut library.  I did have it set to v.0.3.0 a day ago and decided to give the latest library a go to see if it was the issue.  It was not. 

So I'm stuck with prints failing due to Y-Axis shifts.   Note:  I also have the @section tmc_smart section #define Y_CURRENT set to 1100 in hopes this helped, but again it did not.

So attached are the two famous files you guys ask for and I would appreciate it if someone could look them over and tell me it a goof on my part and where to fix it or point out that it could be a code issue.

Thanks for the support.

R,
Doug
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2834054/Configuration.zip)

marcio-ao, 2019-02-06:
Is that R_SENSE correct? We use the Einsy Retros and the R_SENSE is 0.12 rather than 0.11. But then again, the Einsy Retro is a variant of the Einsy Rambo, so don't take my word on it. Check the actual schematic for your board.

marcio-ao, 2019-02-06:
Also, I don't recommend using STEALTH on XY. We only do it for Z and E and our printers are plenty silent with the following TMC_ADV:

        #define TMC_ADV \
            stepperX.toff(1); \
            stepperX.hstrt(4); \
            stepperX.hend(0);\
            stepperX.tbl(1); \
            stepperY.toff(1); \
            stepperY.hstrt(4); \
            stepperY.hend(0);\
            stepperY.tbl(1);

marcio-ao, 2019-02-06:
BTW, our motor currents are 920, 920, 960 for X, Y and Z, respectively.

teemuatlut, 2019-02-06:
Einsy Rambo uses 0.220Ohm sense resistors.
This means current is limited to about 940mA.
Because of the misconfigured R_SENSE value, if my math is correct, you're running the driver at around 560mA.
Also consider trying lower acceleration (maybe around half) and jerk values at first and then bit by bit go higher.

> Also, I don't recommend using STEALTH on XY. We only do it for Z and E and our printers are plenty silent with the following TMC_ADV:

No need to set `tbl(1)` as that is already how it is configured by default. Unless of course you've changed it in your fork =)
Hysteresis values can be set in `CHOPPER_TIMING`. Following your values it would be `{1, -3, 5}`.
I'm curious, have you measured the chopper frequency with your values?
For steppers running in stealthChop, the values don't have any effect, other than off time which merely enables a driver, so there's no harm in setting it "globally".

marcio-ao, 2019-02-06:
> I'm curious, have you measured the chopper frequency with your values?

Nah, they were tuned "by ear". I won't even make a claim they were the best values, they simply worked well during our testing phase and now we are sticking with them for our Mini 2 printers.

> For steppers running in stealthChop, the values don't have any effect, other than off time which merely enables a driver, so there's no harm in setting it "globally".

Interesting. For our newer TAZ prototypes, we are just using the Marlin default values.

marcio-ao, 2019-02-06:
> Einsy Rambo uses 0.220Ohm sense resistors... This means current is limited to about 940mA.

Ah, yes, it's coming back to me now. We requested that Johnny make the Einsy Retro with 0.12 sense resistors so it could reach the currents the Mini Rambo was capable of. Hence the "Retro" -- it's basically a better drop in replacement than for the Mini Rambo than Prusa's Einsy Rambo, but it does forgo features like the Raspberry Pi connector.

dcwalmsley, 2019-02-06:
So I need to adjust R_Sense to 0.12?

teemuatlut, 2019-02-06:
> Einsy Rambo uses 0.220Ohm sense resistors.

As in
```cpp
#define R_SENSE 0.22
```

dcwalmsley, 2019-02-06:
> Einsy Rambo uses 0.220Ohm sense resistors.
> As in
> #define R_SENSE 0.22

Ah, OK. Let reconfigure my firmware and give that a try. Thanks.

dcwalmsley, 2019-02-08:
Mixed results.
On my Hypercube using Einsy board, changing R_Sense to .22 and adjust mA of steppers worked.

However, I have a D-Bot printer using Azteeg X3 Pro with TMC2130 on X,Y,Z, & Z2 and DRV8825 on extruders and running .22 R-Sense produced all kinds of driver overheating warnings.  The software adjusted the drivers down to 300 mA and this was so low that I had massive layer shifting.  So, I'm attempting to find a setting that will produce consistent and reliable results.  Suggestions?

teemuatlut, 2019-02-08:
Azteeg X3 Pro has removable drivers.
The idea is to *describe Marlin your hardware*.

Not every TMC2130 has 0.22Ohm resistors. Or 0.11 for that matter.
Match the setting to what you actually have.

dcwalmsley, 2019-02-08:
How would I know what my TMC has for resistor Ohms?  I'm trying to locate the order to see who I purchased them from but at this time I'm guessing.

teemuatlut, 2019-02-08:
There should be two larger SMD resistors near the driver IC.
[image]
In this image here, you can see the two resistors having written `R110` on both of them.

dcwalmsley, 2019-02-08:
Gotcha.  Found out the brand of TMC2130, I purchased them from Filastruder and they use the Silentstepstick TMC2130-LA driver.  According to multiple images on the on the web it appears that the resister is a R11.  So I will go back and set R-Sense to .11 if I understand that correctly.

dcwalmsley, 2019-02-08:
Took a couple pictures of the Silemtstepstick board.
[image]
[image]

As you can see it lists the resistor as R11.  Put meter on it and it registered .4 Ohms.  I also measured an different set of TMC2130 boards from BigTreeTech and theirs show "R110" which also measured at .4 Ohms.  So this leads me to believe that SilentStepstick resistors are not showing the Zero on their chips labeling.

So what is the recommended R-Sense and Current setting I should start out with?

vilsed, 2019-02-11:
I would trust the resistor rating more than unknown resistance of your multimeter leads, mainly which you are measuring :) "R110" and "R11" is exactly the same thing. If it said "R011", then the actual resistance would be around 0.011Ω. Although I doubt that such resistors actually exist. Set you R_sense at 0.11 and current at about 600-700mA I'd say and go from there. The R_sense should not be changed after that, unless you replace the hardware.

AnHardt, 2019-02-11:
Measuring resistances 'in circit' almost always gives a to low value because the current finds an alternate way thru the circit - parallel resistors/diodes/ICs/Power supply/...

vilsed, 2019-02-11:
@AnHardt 0.4Ω is more than 0.11Ω :) In TMC2130 case the R_sense resistor goes straight from the drivers' end stage H-bridge to ground. There's really no alternative path for the current to go, at least not with some comparable resistance and while the driver is powered down (all H-bridge FET's open). But yes, your point holds true on some occasions.

AnHardt, 2019-02-11:
About the accuracy of Multimeters at low resistencys you talked before, i thought. 0.1Ohm is just a few centimeters of wire, or bit of oxide, or ... .
I just wanted to state: "In general measuring resistors 'in circuit' is not a good idea."
A FET is also just two diodes plus a gate. 
Polarity of measurement also may make a difference.

dcwalmsley, 2019-02-11:
Thanks for the suggestions folks.  I have R-Sense set to .11 and Current to 650mA.  

Issue I'm running into is that the drivers are overheating and the enabling of MONITOR_DRIVER_STATUS to steps down the milliamps to an unsustainable current level (350-400 mA) and my Y-axis is under powered causing layer shifts and print failures. 

I've mounted two 50mm cooling fans over top of the Azteeg board to direct cool air onto drivers/heat sink in an effort to reduce overheating issues.  I do not desire to disable monitor driver status as I feel this is a needed safety measure but I have added additional fans to help maintain the higher milliamp need to run my NEMA 17 motors.   More to follow on this.

vilsed, 2019-02-12:
You've done a right thing - without any active cooling, TMC's tend to heat up quite substantially. Other thing that might help (it did for me, although as a side-effect) would be to increase the Vmot supply voltage to 24V, if its not already (I'm not familiar with Azteeg X3). Higher voltages always mean lower currents for the same final power, hence lower heat losses. As a side bonus - heaters heat up faster, since you can pump more current into them :) If you can't replace the power supply, and if the current one is adjustable, I'd suggest cranking it up as far as it will go (12V supplies can go to ~18V, as I've seen a couple examples of).

Another way could be playing with chopper settings, I think. @teemuatlut could tell you more about this.

dcwalmsley, 2019-02-12:
@vilsed,  I have my printers running on 24vdc 30Amp power supplies.  Azteeg X3 Pro is capable of handling this level of power. 

I wish to thank everyone who provided input to this ticket.  At this time, I plan to close this ticket as I have now have stable drivers running with help of the cooling fans.  I'm running 650mA with R_Sense at .11 and no layer shifts/skipping thus far.

Again thank you for the help.

closed 2019-02-12 by dcwalmsley