MarlinFirmware/Marlin #13085: [Documentation] Help me grok MultiU24X32toH16(), please

i-make-robots, 2019-02-06:
I'm trying to understand the ISR and the acceleration/deceleration routines.  Acceleration says

````acc_step_rate = STEP_MULTIPLY(acceleration_time, current_block->acceleration_rate) + current_block->initial_rate;````

looking further, 

````#define STEP_MULTIPLY(A,B) MultiU24X32toH16(A, B)````

still further,

```
// intRes = longIn1 * longIn2 >> 24
// uses:
// A[tmp] to store 0
// B[tmp] to store bits 16-23 of the 48bit result. The top bit is used to round the two byte result.
// note that the lower two bytes and the upper byte of the 48bit result are not calculated.
// this can cause the result to be out by one as the lower bytes may cause carries into the upper ones.
// B A are bits 24-39 and are the returned value
// C B A is longIn1
// D C B A is longIn2
//
static FORCE_INLINE uint16_t MultiU24X32toH16(uint32_t longIn1, uint32_t longIn2)
```

What I don't get is the purpose of the >>24.  This is somehow tied to the 2Mhz timer on the 16Mhz clock, but it's not clicking for me how it works.  Any light you can shed would be greatly appreciated.

AnHardt, 2019-02-09:
In general:
If multiplying two numbers with n and m binary digits the result will have not more than n+m digits.
If the target variable has only o digits you have to right-shift n+m-o times to get the o most relevant digits and have to scale the unit.
Knowing you are trowing n-m-o digits away can help (if done right) to **speed up** the calculation by simply not calculating the anyway unused digits (or at  least a part of them).

boelle, 2019-03-06:
@i-make-robots still working on this?

i-make-robots, 2019-03-06:
I don't get how it works, but I'm using it effectively.

boelle, 2019-03-06:
then maybe click close below?

closed 2019-03-06 by i-make-robots