MarlinFirmware/Marlin #13088: Advanced Pause not compiling with Serial Port 2 on MKS Sbase

B3D-P, 2019-02-06:
### Description

Similar to #9948  
Advanced pause on MKS SBase using LPC1768 generates an error when serial port 2 is defined. 
When serial port 2 is commented out it will compile with no errors but then you will not be able to use the TFT Display.
The error is: 
Marlin\src\module\configuration_store.cpp:2346:52: error: 'port' was not declared in this scope
Using Atom with PlatformIO with bugfix within the week
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2834828/Configuration_adv.zip)

### Steps to Reproduce

1. [First Step] Compile with MKS Sbase and uncomment advanced pause
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen] 
compile without error

**Actual behavior:** [What actually happens]
Marlin\src\module\configuration_store.cpp:2346:52: error: 'port' was not declared in this scope

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2019-02-22:
I'm working on a reform to the serial print code that should fix this. I'll let you know when the patch is in.

thinkyhead, 2019-02-24:
The patch has been applied. As a result the compile error should be fixed.

B3D-P, 2019-02-24:
Great Thanks!

closed 2019-02-24 by B3D-P