MarlinFirmware/Marlin #13089: Z homing with Bltouch not working and gives a STOP error

Krimik, 2019-02-06:
### Description
I have installed a genuine Bltouch on my Anycibic i3 Mega according to [this](https://cdn.thingiverse.com/renders/88/ab/35/72/87/ec7b446903558be2ad5d1b41d37230cd_preview_featured.jpg) wiring diagram.

When testing XYZ homing, XY are doing fine, but Z homing behaves weirdly. Z axis goes up by 5-10mm, Bltouch probes once and then log gives an error.

### Steps to Reproduce
1) Switch on the printer
2) Do Z homing

#### Additional Information
[Here](https://vimeo.com/315615320) is a video of me switching it on and doing Z home.

Here is the error from the log
> 11:24:44.428 : N16 G28 Z0*110
> 11:24:53.247 : Error:STOP called because of BLTouch error - restart with M999
> 11:24:53.247 : Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
> 11:24:53.590 : X:0.00 Y:0.00 Z:15.00 E:0.00 Count X:0 Y:0 Z:6000
> 11:25:07.459 : N17 M999*28
> 11:25:07.459 : N1 M110*34

M119 log:
> 11:50:56.538 : Reporting endstop status
> 11:50:56.538 : x_min: TRIGGERED
> 11:50:56.538 : y_min: TRIGGERED
> 11:50:56.538 : z_min: TRIGGERED
> 11:50:56.538 : z2_min: open
 
Configs: [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2835661/Marlin.zip)

Also, this behaves identically if I disconnect black&white connector of Bltouch from green header PCB. Would this connection being bad be consistent with this behavior?

If not, probably I've done something wrong in config file...

Thanks!

abpat2203, 2019-02-08:
I have the exact same issue. I have an Ender 3. Following are the steps I have tried:

1. Opened up the control box to check my Z-stepper wiring is still in place. It is hotglued and will not budge.
2. Flashed back from Marlin to TH3D, which was working earlier. That does not work either now.
3. Replaced the genuine BLTouch with a clone Geetech 3D touch that I had lying around. Same error.
4. Changed #define Z_MIN_PROBE_ENDSTOP_INVERTING false
5. Inverted the black and white wires
6. Installed 1.1.x (bugfix) of Marlin

maar1201gh, 2019-02-09:
I have the exact same issue

abpat2203, 2019-02-09:
Turned out to be a wiring issue. The black and white wires had to be plugged in again and hotglued. It worked for some time but then the issue came back. 

I have temporarily removed the BLTouch. Will need to get Molex connectors to replace the Dupont connectors to get a better stable connection everytime.

maar1201gh, 2019-02-10:
After 1 week trying to discover why BLTOUCH was working well in recent past and now failed, i decided to leave bltouch RIP.
I'm using tempered glass as hot bed with glass mirror over hot bed.
To resolve my issue, i put aluminium tape  between 2 glasses and change to an indutive sensor. Now all is working again. in any case, Thanks for you help
regards

boelle, 2019-03-13:
@thinkyhead think we can assume OP dead here too :-)

Krimik, 2019-03-13:
Still alive. I gave up on BLtouch since than and moved back to manual mesh leveling. Ill give 2.0 a try though!

boelle, 2019-03-13:
that would be a good choice as its where things are updated and fixed most

just look at commit history and you will see what i mean

shitcreek, 2019-05-15:
@Krimik any update? Is 2.0 working for you?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-11 by thinkyhead

YaMoef, 2019-12-24:
@Krimik where you able to solve the problem with the new Marlin software? If not do you have a board with an extra connector for your bl touch? I mean with this if you had to connect your bltouch to the Z endstop or to an extra connector for you BL Touch. Do you use the stock board? Or did you upgrade it? If the connector or seperate I might have the problem and maybe I'm able to solve. I have currently the same ussue and I'm trying to solve it.

Krimik, 2019-12-24:
I kinda gave up bltouch and started doing manual mesh leveling instead. :)

YaMoef, 2019-12-24:
Oh, ok if I find a solution I'll explain what I did and you can still choose if you want to try it or not.

YaMoef, 2019-12-24:
Got the solution. So the problem was that my board (SKR Pro V1.1 with the CR-10s) has a seperate connection for the BL Touch but I left the Z min endstop disconnected so for some or other reason it was still using that endstop which caused it to be always triggered and my bltouch working properly. But when I did a Z axis home it always gave me the same error as it gave to you (@Krimik) so what I did is changing the pin for the Z endstop to the same pin as I connected my Bl touch on. Also my Z_MIN_PROBE_ENDSTOP_INVERTING is set to false. This problem could also be solved I think with inverting the Z min endstop maybe? Hope this helps for other who encounter this problem.

drkline, 2019-12-26:
YaMoef...  I'm not following you when you said that you "changing the pin for the Z endstop to the same pin as I connected my Bl touch on".  Could you explain a bit more in depth please?

YaMoef, 2019-12-26:
So on my motherboard (SKR Pro V1.1) I have Bltouch connector and a Z endstop connector. Both of the connector are completely isolated from eachother (except for the 5V and GND pin). So at first it wasn't working for me, so I thought back on my creality board how I did it there since it was working over there. And what I did different to the application now is that I connected on my creality board I cutted the wires of my endstop and directly connected my bltouch to it. Where in this application with the SKR Pro I did it to another connecter where the switch pin of the bltouch is not the same as the endstop pin. So what I did is in the software change my endstop pin to the same pin as my Bltouch switch pin is. What I did is also connected the bltouch to Z endstop like I did on my Creality board only this is done by software, not by hardware as done on the creality board. Watch the picture  (https://imgur.com/a/6aUcSSQ) there you can see difference between the connection of the bltouch on my SKR and on my Creality board where it was working pefectly. I hope this explains it eneugh if not, don't hessitate for more askin more explenation.

Solxz, 2020-01-07:
I'm having this exact same issue with my CR-10S Pro.  When I got to level, the probe extends and retracts, then the gantry moves up, the probe extends and retracts again, and the gantry moves up again.