MarlinFirmware/Marlin #13090: [2.0 bugfix] TMC 2130 homing problem

mcmuff91, 2019-02-06:
Hello everyone,

atm I'am changing the board of my delta printer for ceramics from Ramps 1.4 with DRV8825 to Ramps FD with TMC2130.

Facts of the  system:
- TMC2130 in standalone mode , spreadcycle mode ( higher holding torque)
- Ramps FD driven by 12V
- mechanical endstops

Issue:

When I send G28 to the printer , all axes are moving up synchron till one (maximum two) endstop is pressed/triggered.  
After this, the triggered axes are moving down and doing the normal bump.  The carraige(s) that were not hitting an endstop wont move.

After this all axes are moving down to safe home position.

Problem: The effector is not centered. Sometimes i need to send G28 more than 1 time until the axes starts to move. 

What i tried:
- checking endstops (measuring)
- going from stealthchop to spreadcycle mode
- adjusting vref of the steppers
- going down with homing speed
- reduce acceleration

Thanks for your time :)

AnHardt, 2019-02-06:
If not already done try to activate endstop denoising.

mcmuff91, 2019-02-07:
I tried. No changes

vilsed, 2019-02-11:
I was having a similar issue, but only occasionally. Software endstop denoising, as well as capacitors for electrical denoising did not work. What I found out was that homing at 200mm/s is not a good idea :) Now homing at 50mm/s and rebumping at 10mm/s works reliably, had no issues.

mcmuff91, 2019-02-11:
Located the issue. The jumper to activate the internal pullups was not described in the manual. Added a jumper and now everything is working

closed 2019-02-11 by mcmuff91